luxonis/depthai-ros #282: [Question] How to load a NN model from Model Zoo into depth_ai_ros driver ?

cwillboyd, 2023-04-12:
I am trying to load an NN model (fast_depth_480x640) that I downloaded from the [model zoo](https://zoo.luxonis.com/). I have the .bin and .xml file on my local machine. What is the process of getting this to run using the camera.launch file in depthai_ros_driver ? 

I believe I need to change a few lines in the camera.yaml file related to `camera_i_nn_type: spatial`  and `nn_i_nn_config_path: depthai_ros_driver/mobilenet`. 

Is it correct that I need to add a .json file in depthai_ros_driver/config/nn ? How do I convert the XML file from the model zoo into .json format? 

Do I need to specify the model_path as well ? I see that there are some .blob files located in depthai-ros/depthai_examples/resources ... do I need to add a .blob file for my model here ? How can I convert the .bin and .xml files from the model zoo into .blob format ? 

Lastly what should I set for `camera_i_nn_type` if I am using a nerual depth model based on RGB, should I change that value to rgb or keep it as spatial ? 

Thank You

Serafadam, 2023-04-20:
Hi, details on model conversion are available [here](https://docs.luxonis.com/en/latest/pages/model_conversion/), though currently to publish custom output from a NN you would either need to edit source code of depthai_ros_driver to include custom output or use depthai_bridge with custom callback function.

kgallowa, 2023-07-11:
Working on the same type of problem (i.e. wanting to use a custom NN model with depthai-ros) and had the same question. @cwillboyd, did you end up figuring this out?

Serafadam, 2023-08-17:
Hi @cwillboyd @kgallowa, corry for the late reply, in the newer versions of `depthai_ros_driver` you can load custom pipelines using ROS plugilib mechanism, that way you can create a custom NN node (similarly to how segmentation is set up) and load it into the pipeline.

wendwosenbb, 2024-05-29:
@Serafadam 
is there away in the plugin mechanism to parse custom inference output?  specifically, how to parse pre non_max_suppression results so that  the host side handle the nms.