luxonis/depthai-ros #283: Enable the left and right images when running the stereo_inertial_node

LanWu076, 2023-04-13:
When I run the roslaunch depthai_examples stereo_inertial_node.launch, I can have depth images and point cloud. Is there a way to enable the left and right cameras streams as well?

Serafadam, 2023-04-20:
Hi, to enable left and right streams in stereo_inertial_node you would have to edit the source code, you can publish LR streams with depth_aligned argument, though that disables rgb pointcloud. Having both stereo, rgb, left and right streams is available in depthai_ros_driver, when respective parameters are set.

HanDaSeul, 2023-04-21:
@Serafadam how? Is it 'stereo_inertial_node.py'  in the launch folder, right? I just modified 'depth_aligned' to False, that makes stream left and right and disables rgb pointcloud as you said, but with rgb disabled. True is defult and that makes only rgb streams, not left and right streams.

Serafadam, 2023-04-21:
Hi I meant [this launch file](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/launch/camera.launch.py) and [these parameters](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/camera.yaml)

LanWu076, 2023-04-24:
Hi @Serafadam,

Thank you for the reply. Is there a way to modify the source code to have depth, rgb, left and right, and point cloud in depthai_ros_driver?

I'd like to keep the point cloud but also enable the left and right streams.

Thanks