luxonis/depthai-ros #285: Noetic version: point clouds are different when running depthai_bridge and depthai_ros_driver

LanWu076, 2023-04-14:
I have the same issue as in #223. It seems the similar issue remaining in the latest noetic branch (not the 271_neotic). The point cloud is okay when running the depthai_example (based on depthai_bridge) but extremely noisy when running the depthai_ros_driver. 

depthai_example (side view) it is okay
[image]

depthai_ros_driver(side view) it is noisy
[image]

depthai_example (top view) it is okay
[image]

depthai_ros_driver(top view) it is noisy, even travel through the wall
[image]
 
I tried to enable the filtering based on the default config file (camera.yaml). It does not affect the performance.

ndaley7, 2023-08-08:
> Hi, thanks for reporting, newest Noetic release will add more parameters to depth filtering, which should make both pointclouds the same

Hello!  Its a bit difficult to tell on the release page.  Were the additional parameters added and if so what were they specifically?

Serafadam, 2023-08-11:
Hi, yes, a lot of parameters have been updated and pointclouds should be roughly the same, if you experience any issues please let us know.