luxonis/depthai-ros #286: [BUG] IMU orientation message does not consider Magnetometer data

ladrians, 2023-04-17:
Noticed the IMU orientation vector is not calculated.

To reproduce it is just to check the `/oak/imu` topic where the orientation is `0 0 0 1`.

I am using a `OAK D PRO W` device with a BNO085 IMU.
	
based on the [docs](https://docs.luxonis.com/projects/api/en/latest/components/messages/imu_data/#imudata) I naively did a modification on the `stereo_inertial_publisher.cpp` sample adding

```
imu->enableIMUSensor(dai::IMUSensor::MAGNETOMETER_RAW, 10);
```

But there are no visible results, finally I could check the orientation is still empty, I am using [noetic](https://github.com/luxonis/depthai-ros/blob/noetic/depthai_bridge/src/ImuConverter.cpp#L177) but the same happens with [humble](https://github.com/luxonis/depthai-ros/blob/humble/depthai_bridge/src/ImuConverter.cpp#L181).

```
    interpMsg.orientation.x = 0.0;
    interpMsg.orientation.y = 0.0;
    interpMsg.orientation.z = 0.0;
    interpMsg.orientation.w = 1.0;
```

Would like to know the plans to integrate a orientation value, or to publish a standalone magnetometer topic?
Currently we are using a 9 axis external IMU and using the madgwick filter node to get and absolute reference for the IMU output.
We want to use the integrated IMU from the camera to keep configuration as simple as possible, the expected output for us is to get an absolute orientation for outdoor navigation.

Is there a sample code I could integrate to check if the information can be published?

Related information I could find, [here](https://github.com/luxonis/depthai-ros/pull/90) and [here](https://github.com/luxonis/depthai-ros/issues/168).

**system log**
 - [oak2_log_system_information.zip](https://github.com/luxonis/depthai-ros/files/11249124/oak2_log_system_information.zip)
 - NVidia Jetson Orin using Jetpack 5.1.1 Ubuntu 20.04.6 LTS
 - ROS noetic 1.16.0
 - `depthai-ros` built from source
 - `depthai/depthai-core` library installed from rosdep
 - apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver

```sh
Package: ros-noetic-depthai
Version: 2.20.2-1focal.20230202.200823
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 21.3 MB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 4.5), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-noetic-catkin
Homepage: https://www.luxonis.com/
Download-Size: 10.2 MB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Version: 2.6.1-2focal.20230225.012112
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 18.4 kB
Depends: ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-depthai-examples, ros-noetic-depthai-ros-driver, ros-noetic-depthai-ros-msgs
Download-Size: 2,124 B
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: The depthai-ros package

Package: ros-noetic-depthai-bridge
Version: 2.6.1-2focal.20230216.065040
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 69.5 MB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 4.2), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgcodecs4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libboost-dev, libopencv-dev, ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-ros-msgs, ros-noetic-image-transport, ros-noetic-robot-state-publisher, ros-noetic-ros-environment, ros-noetic-roscpp, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-stereo-msgs, ros-noetic-vision-msgs, ros-noetic-xacro
Download-Size: 10.5 MB
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: The depthai_bridge package

Package: ros-noetic-depthai-ros-msgs
Version: 2.6.1-2focal.20230216.064856
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 475 kB
Depends: ros-noetic-geometry-msgs, ros-noetic-message-runtime, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-vision-msgs
Download-Size: 39.4 kB
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: Package to keep interface independent of the driver

Package: ros-noetic-depthai-ros-driver
Version: 2.6.1-2focal.20230225.010515
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 1,490 kB
Depends: libc6 (>= 2.17), libconsole-bridge0.4, libgcc-s1 (>= 4.2), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 5.2), ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-diagnostic-msgs, ros-noetic-dynamic-reconfigure, ros-noetic-image-pipeline, ros-noetic-image-transport, ros-noetic-image-transport-plugins, ros-noetic-message-filters, ros-noetic-nodelet, ros-noetic-roscpp, ros-noetic-rospy, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-std-srvs, ros-noetic-vision-msgs
Download-Size: 277 kB
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main arm64 Packages
Description: Depthai ROS Monolithic node.
```

Serafadam, 2023-04-24:
Hi, currently rotation vector is not enabled in the driver, will add option to use it's data this week.

ladrians, 2023-04-24:
ok cool will be waiting for that.
Can you detail if the magnetometer will be available as a separate Rostopic?

Serafadam, 2023-05-04:
Hi @ladrians , you can find details [here](https://github.com/luxonis/depthai-ros/pull/305), in short there will be an option to choose different message types when publishing, either with a custom message that contains both imu and magnetic field message or to have those messages split into two different topics

ladrians, 2023-05-07:
lgtm! is there a change to port it to noetic branch?

ladrians, 2023-06-03:
@Serafadam is there a change to backport it to the `noetic` branch?

Serafadam, 2023-06-05:
Hi, sorry for the delay, I was away for some time, yes, the newest Noetic branch also has those changes

Serafadam, 2023-09-14:
Closing since the issue was resolved.

closed 2023-09-14 by Serafadam