luxonis/depthai-ros #287: camera.launch: IMU Data is not being published

everskies, 2023-04-18:
**Describe the bug**
I've been trying to get the camera.launch.py example to publish IMU data together with the stereo data and RGB image. The `/oak/imu/data` topic gets created:

```
ros2 topic list
/joint_states
/oak/imu/data
/oak/rgb/camera_info
/oak/rgb/image_raw
/oak/rgb/image_raw/compressed
/oak/rgb/image_raw/compressedDepth
/oak/rgb/image_raw/theora
/oak/stereo/camera_info
/oak/stereo/image_raw
/oak/stereo/image_raw/compressed
/oak/stereo/image_raw/compressedDepth
/oak/stereo/image_raw/theora
/parameter_events
/robot_description
/rosout
/tf
/tf_static
```

However, no data seems to arrive:

```
ros2 topic hz /oak/imu/data
WARNING: topic [/oak/imu/data] does not appear to be published yet
```

I've tried both the current humble release and one built from scratch.

**Minimal Reproducible Example**
`ros2 launch depthai_ros_driver camera.launch.py`

```
[INFO] [launch]: All log files can be found below /home/sleziak/.ros/log/2023-04-18-22-47-30-727616-bubble-2551
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [2562]
[INFO] [component_container-2]: process started with pid [2564]
[component_container-2] [INFO] [1681850852.374532197] [oak_container]: Load Library: /home/sleziak/ros/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[robot_state_publisher-1] [INFO] [1681850852.540030735] [oak_state_publisher]: got segment oak
[robot_state_publisher-1] [INFO] [1681850852.540405935] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-1] [INFO] [1681850852.540491657] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-1] [INFO] [1681850852.540537342] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-1] [INFO] [1681850852.540575471] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1681850852.540611785] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-1] [INFO] [1681850852.540646452] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-1] [INFO] [1681850852.540679822] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-1] [INFO] [1681850852.540713359] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-1] [INFO] [1681850852.540746488] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-2] [INFO] [1681850852.691349716] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1681850852.691494308] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1681850852.745608206] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [INFO] [1681850855.725776186] [oak]: Camera with MXID: 18443010F1213B1300 and Name: 1.1.2 connected!
[component_container-2] [INFO] [1681850855.727304153] [oak]: USB SPEED: SUPER
[component_container-2] [INFO] [1681850855.750146502] [oak]: Device type: OAK-D-PRO-AF
[component_container-2] [INFO] [1681850855.757216437] [oak]: Pipeline type: RGBD
[component_container-2] [INFO] [1681850855.801349463] [oak]: NN Family: mobilenet
[component_container-2] [INFO] [1681850856.071826360] [oak]: NN input size: 300 x 300. Resizing input image in case of different dimensions.
[component_container-2] [INFO] [1681850856.277071831] [oak]: Finished setting up pipeline.
[component_container-2] [INFO] [1681850857.683827872] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
```

**Expected behavior**
There should be data being published to the `/oak/imu/data` topic

<details>
[... 394 lines cut ...]
Description: The depthai_bridge package

Package: ros-humble-depthai-ros-msgs
Version: 2.6.4-1jammy.20230302.210403
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 1,310 kB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 3.3.1), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), ros-humble-fastcdr, ros-humble-builtin-interfaces, ros-humble-geometry-msgs, ros-humble-rclcpp, ros-humble-rosidl-default-generators, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 106 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages
Description: Package to keep interface independent of the driver

Package: ros-humble-depthai-ros-driver
Version: 2.6.4-1jammy.20230303.005120
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 844 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libopencv-stitching4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), ros-humble-ament-cmake-auto, ros-humble-camera-calibration, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-diagnostic-msgs, ros-humble-image-pipeline, ros-humble-image-transport, ros-humble-image-transport-plugins, ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 213 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main arm64 Packages
Description: Depthai ROS Monolithic node.
```
</details>

everskies, 2023-04-18:
Update:

Turns out that the data is being published, but only once every 10 or so launches of `camera.launch.py`.

HanDaSeul, 2023-05-02:
Did you check what's the reason? There's nothing problem with ROS2 but ROS noetic is. It doesn't publish imu at all when I run camera.launch. But stereo_inertial.launch in the dai_examples publish imu.

Serafadam, 2023-05-04:
Hi @everskies, from your debug information it seems that your ROS packages are a little behind the official ones, could you upgrade and check again? Also, which camera model is that?
@HanDaSeul same here, could you post your camera model and packages version?

HanDaSeul, 2023-05-04:
@Serafadam Repied here. https://github.com/luxonis/depthai-ros/issues/304#issuecomment-1533244754