luxonis/depthai-ros #288: [BUG] {RTAB Map not recieving data and publishing topic error on Luxonis OAKD (ROS2 Foxy)}

Aloha-Churchill, 2023-04-20:
**Describe the bug**
I am currently using a Raspberry Pi 4 with the Luxonis OAK-D camera. My goal is to be able to run RGBD SLAM from a remote development machine in the same network. Currently, I am experimenting with RTAB-Map since it has support for the OAK-D. However, I am having two primary issues. First, when I run the command

`ros2 launch depthai_examples stereo_inertial_node.launch.py depth_aligned:=false` I am seeing the following topics

```
^Cubuntu@ubuntu:~/test_ws$ ros2 topic list
/color/camera_info
/color/image
/image/compressed
/image/compressedDepth
/imu
/joint_states
/left/camera_info
/left/image_rect
/left/image_rect/compressed
/left/image_rect/compressedDepth
/parameter_events
/right/camera_info
/right/image_rect
/right/image_rect/compressed
/right/image_rect/compressedDepth
/robot_description
/rosout
/stereo/camera_info
/stereo/converted_depth
/stereo/depth
/stereo/depth/compressed
/stereo/depth/compressedDepth
/stereo/points
/tf
/tf_static
```

which seems correct. However, on my remote development machine I can echo out the /imu topic but not more intensive topics such as /color/image which then does not allow for launching rtabmap on the remote development machine. In case it is helpful I think there were two possible causes to this. First, I converted to using a splitter cable and powering the camera separately and this helped the /imu topic show up (from the raspberry pi). The network I am using has high bandwidth, and while I still think it may be a network issue, I'm not completely sure. Now, I can echo out these topics on the raspberry pi. However, when I tried to launch rtabmap from this launch file that I created

```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([
        Node(
            package='rtabmap_ros',
            executable='rtabmap',
            name='rtabmap',
            parameters=[{
                'delete_db_on_start': True,
                'subscribe_stereo': True,
                'frame_id': 'oak-d_frame',
                'approx_sync': True,
                'wait_imu_to_init': True,
                'Reg/Strategy': '0', 
            }],
            remappings=[
                ('/left/image_rect', '/left/image_rect'),
                ('/right/image_rect', '/right/image_rect'),
                ('/left/camera_info', '/left/camera_info'),
                ('/right/camera_info', '/right/camera_info'),
                ('/imu/data', '/imu'),
                ('/odom', '/odom'),
            ],
            output='screen'
        )
    ])
```
I am seeing 

```
[rtabmap-1] [WARN] [1681951004.072285317] [rtabmap]: rtabmap: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10). 
[rtabmap-1] rtabmap subscribed to (approx sync):
[rtabmap-1]    /odom \
[rtabmap-1]    /left/image_rect \
[rtabmap-1]    /right/image_rect \
[rtabmap-1]    /left/camera_info \
[rtabmap-1]    /right/camera_info
```

I did a `ros2 topic hz` for the left and right images and am getting this rate 

```
ubuntu@ubuntu:~$ ros2 topic hz /left/image_rect
average rate: 9.287
	min: 0.038s max: 0.182s std dev: 0.03285s window: 11
```
for both. I was wondering if the rate was too low and if that was causing a problem. If this is a case, is there a way to debug this issue? 

Thank you so much for your time and help! Please let me know if there is any additional information I can provide.

Cheers,
Aloha

Serafadam, 2023-04-21:
Hi, I guess this is similar to https://github.com/luxonis/depthai-ros/issues/266, I'd suggest trying out solutions proposed there and see if it helps

Aloha-Churchill, 2023-04-29:
With the new depathai foxy update, RTAB-Map now working. Thanks for the help!

closed 2023-04-29 by Aloha-Churchill