luxonis/depthai-ros #290: [Question]  ROS1(noetic) depthai_examples launch file with OAK1 camera.

Duckyu, 2023-04-20:
Hi, I have a question about using depthai_examples.

I'm comparing the performance of the OAK1 camera with other camera setups. 
I want to test the OAK1 camera with examples from the depthai_examples package for ROS1 noetic.

However, I can not find out how to set the parameter of the launch file compatible with the OAK1 camera.
If someone has already tried this task, may I ask you for some tips? (Specifically, I'm interested in "[depthai_examples/launch/yolov4_publisher.launch](https://github.com/luxonis/depthai-ros/blob/noetic/depthai_examples/launch/yolov4_publisher.launch)")

Thank you

Serafadam, 2023-04-25:
Hi, to test out Yolo with OAK1, you can either use `depthai_ros_driver` or create a custom example (we have PR set up for that in Humble, it will be ported to Noetic in next release https://github.com/luxonis/depthai-ros/pull/262)

Duckyu, 2023-04-26:
Thanks a lot for your comment. I'll check that conversation.

closed 2023-04-26 by Duckyu