luxonis/depthai-ros #299: [Question] Unable to driver FFC-4P via ROS package

Peize-Liu, 2023-04-26:
I successfully build depthai-ros, however i am ubale to use FFC-4P via ROS package.
I tested `roslaunch depthai_ros_driver camera.launch` but get 
[image]
I modified some configuration in example_multicam.luanch, got error
[image]
[image]

I aslo tried `roslaunch depthai_examples mobile_publisher.launch`
But It only published one camera image

Serafadam, 2023-04-26:
Hi, could you post more logs according to [bug template? ](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)
On launching part, FFC cameras are currently supported [with CamArray option in depthai-ros-driver](https://github.com/luxonis/depthai-ros/issues/255)

Peize-Liu, 2023-04-26:
My environment info using provided python script
[log_system_information.txt](https://github.com/luxonis/depthai-ros/files/11330958/log_system_information.txt)

depethai
 sdk
`Package: ros-noetic-depthai
Version: 2.21.2-1focal.20230406.144634
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 21.7 MB
Depends: libc6 (>= 2.17), libgcc-s1 (>= 4.2), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-noetic-catkin
Homepage: https://www.luxonis.com/
Download-Size: 10.3 MB
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Package: ros-noetic-depthai-ros
Version: 2.6.4-1focal.20230406.152421
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 18.4 kB
Depends: ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-depthai-examples, ros-noetic-depthai-ros-driver, ros-noetic-depthai-ros-msgs
Download-Size: 2,316 B
APT-Manual-Installed: yes
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
Description: The depthai-ros package

Package: ros-noetic-depthai-bridge
Version: 2.6.4-1focal.20230406.150135
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 69.5 MB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.0), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgcodecs4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 9), libboost-dev, libopencv-dev, ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-ros-msgs, ros-noetic-image-transport, ros-noetic-robot-state-publisher, ros-noetic-ros-environment, ros-noetic-roscpp, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-stereo-msgs, ros-noetic-vision-msgs, ros-noetic-xacro
Download-Size: 10.5 MB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
Description: The depthai_bridge package

Package: ros-noetic-depthai-ros-msgs
Version: 2.6.4-1focal.20230306.191627
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 475 kB
Depends: ros-noetic-geometry-msgs, ros-noetic-message-runtime, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-vision-msgs
Download-Size: 39.4 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
Description: Package to keep interface independent of the driver

Package: ros-noetic-depthai-ros-driver
Version: 2.6.4-1focal.20230406.151056
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 1,649 kB
Depends: libc6 (>= 2.14), libconsole-bridge0.4, libgcc-s1 (>= 3.0), libopencv-core4.2 (>= 4.2.0+dfsg), libopencv-imgproc4.2 (>= 4.2.0+dfsg), libstdc++6 (>= 5.2), ros-noetic-camera-info-manager, ros-noetic-cv-bridge, ros-noetic-depthai, ros-noetic-depthai-bridge, ros-noetic-diagnostic-msgs, ros-noetic-dynamic-reconfigure, ros-noetic-image-pipeline, ros-noetic-image-transport, ros-noetic-image-transport-plugins, ros-noetic-message-filters, ros-noetic-nodelet, ros-noetic-roscpp, ros-noetic-rospy, ros-noetic-sensor-msgs, ros-noetic-std-msgs, ros-noetic-std-srvs, ros-noetic-vision-msgs
Download-Size: 322 kB
APT-Manual-Installed: no
APT-Sources: http://packages.ros.org/ros/ubuntu focal/main amd64 Packages
Description: Depthai ROS Monolithic node.
`

` libdepthai-opencv.so (libc6,x86-64) => /usr/local/lib/libdepthai-opencv.so
   libdepthai-core.so (libc6,x86-64) => /usr/local/lib/libdepthai-core.so`

[roslaunch-Khalil-WS-257752.log](https://github.com/luxonis/depthai-ros/files/11330943/roslaunch-Khalil-WS-257752.log)

Serafadam, 2023-05-05:
Hi, thanks for the debug info, could you try loading and testing the latest version available? (v2.7.1)

Peize-Liu, 2023-05-05:
> Hi, thanks for the debug info, could you try loading and testing the latest version available? (v2.7.1)

I still got same error in latest version

Serafadam, 2023-05-08:
This is when running `camera.launch.py` with default config file, or did you provide a new one?

Peize-Liu, 2023-05-08:
> This is when running `camera.launch.py` with default config file, or did you provide a new one?

This error occurs when using default launch file. I aslo tried change camera.yaml  "camera_i_pipeline_type" from RGB to CAMARRAY , still the same erro however.

Peize-Liu, 2023-05-09:
I find the problem. My local environment did not Source  ROS local setup script  automatically. So the driver cannot launch nodelet properly.
source local ros setup script and then launch package would solve the problem.

closed 2023-05-09 by Peize-Liu

Peize-Liu, 2023-05-09:
After test on my arm and x86 platform, this problem is till exsist. But this error can be worked around by using package Installed from ros binaries