luxonis/depthai-ros #300: OAK D Pro POE no images with ROS2 and Noetic

Victorsoeby, 2023-04-27:
I have an OAK-D camera that ran with USB, which i had working with this repo on the noetic branch. I was able to run the dockerized setup and have all the topics appear on my local machine and within the container without any issues. 
Now i have set up the OAK-D Pro POE camera, and was looking forward to use the wide angle sensor. However, when i try to run it through docker in the same setup i am getting errors no matter which of the launchfiles in depthai_ros_driver i try to run. This is the camera.launch.py for instance: 
`
WARNING: Some networks were defined but are not used by any service: orin_docker_compose_default
Starting buddy_depthai_ros_1 ... done
Attaching to buddy_depthai_ros_1
depthai_ros_1  | [INFO] [launch]: All log files can be found below /root/.ros/log/2023-04-27-13-45-44-097686-victor-Legion-5-Pro-16ACH6H-1
depthai_ros_1  | [INFO] [launch]: Default logging verbosity is set to INFO
depthai_ros_1  | [INFO] [robot_state_publisher-1]: process started with pid [71]
depthai_ros_1  | [INFO] [component_container-2]: process started with pid [73]
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266573731] [camera_state_publisher]: got segment camera
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266636729] [camera_state_publisher]: got segment camera_imu_frame
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266647764] [camera_state_publisher]: got segment camera_left_camera_frame
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266655446] [camera_state_publisher]: got segment camera_left_camera_optical_frame
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266663409] [camera_state_publisher]: got segment camera_model_origin
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266670532] [camera_state_publisher]: got segment camera_rgb_camera_frame
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266678215] [camera_state_publisher]: got segment camera_rgb_camera_optical_frame
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266686387] [camera_state_publisher]: got segment camera_right_camera_frame
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266694838] [camera_state_publisher]: got segment camera_right_camera_optical_frame
depthai_ros_1  | [robot_state_publisher-1] [INFO] [1682603144.266703428] [camera_state_publisher]: got segment oak-d-base-frame
depthai_ros_1  | [component_container-2] [INFO] [1682603144.497938202] [camera_container]: Load Library: /ws/install/lib/libdepthai_ros_driver.so
depthai_ros_1  | [component_container-2] [INFO] [1682603144.539042731] [camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
depthai_ros_1  | [component_container-2] [INFO] [1682603144.539089875] [camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
depthai_ros_1  | [component_container-2] [INFO] [1682603144.543884918] [camera]: No ip/mxid specified, connecting to the next available device.
depthai_ros_1  | [component_container-2] [INFO] [1682603151.107728797] [camera]: Camera with MXID: 184430109159A30F00 and Name: 10.3.3.66 connected!
depthai_ros_1  | [component_container-2] [INFO] [1682603151.107849904] [camera]: PoE camera detected. Consider enabling low bandwidth for specific image topics (see readme).
depthai_ros_1  | [component_container-2] [INFO] [1682603151.130946569] [camera]: Device type: OAK-D-PRO-W-POE-97
depthai_ros_1  | [component_container-2] [INFO] [1682603151.137299241] [camera]: Pipeline type: RGBD
depthai_ros_1  | [component_container-2] [INFO] [1682603151.160099983] [camera]: NN Family: mobilenet
depthai_ros_1  | [component_container-2] [INFO] [1682603151.207725435] [camera]: NN input size: 300 x 300. Resizing input image in case of different dimensions.
depthai_ros_1  | [component_container-2] [INFO] [1682603151.247151856] [camera]: Finished setting up pipeline.
depthai_ros_1  | [component_container-2] [184430109159A30F00] [10.3.3.66] [5.745] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
depthai_ros_1  | [component_container-2] [ERROR] [1682603155.099552911] [camera_container]: Component constructor threw an exception: ColorCamera(0) - 'video' width or height (1280, 720) bigger than maximum at current sensor resolution (854, 534)
depthai_ros_1  | [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'camera' of type 'depthai_ros_driver::Camera' in container '/camera_container': Component constructor threw an exception: ColorCamera(0) - 'video' width or height (1280, 720) bigger than maximum at current sensor resolution (854, 534)
`
My docker-compose file looks like this: 

`
services:
  depthai_ros:
    build: 
      context: ros_depthai_oak
      dockerfile: Dockerfile
    privileged: true
    cap_add:
      - ALL
    volumes:
      - /dev/:/dev/
      - /tmp/.X11-unix:/tmp/.X11-unix
    devices:
      - "/dev/video1:/dev/video1"
    environment:
      - ROS_DOMAIN_ID=$_ROS_DOMAIN_ID
      - DISPLAY
    network_mode: host
    command: bash -c "ros2 launch depthai_ros_driver camera.launch.py"
    logging:
      driver: $_DOCKER_LOGGING_DRIVER
      options:
        max-file: $_DOCKER_LOGGING_MAX_FILE
        max-size: $_DOCKER_LOGGING_MAX_SIZE
`
I have tried using the depthai_python repo to see if it is my setup or a hw fault in my camera, but this is working just fine. I also looked into if the container can actually see the camera through the POE setup, but from the errormessage it appears the camera is accessible. Has anyone tried the same? I find it quite confusing that it says the sensor resolution is (845, 534)

Serafadam, 2023-04-27:
Hi, it seems parameter related as they are set up for cameras with mono sensors, I'll change them in the next PR, in the meantime, could you check if changing `left_i_set_isp_scale: false` & `right_i_set_isp_scale: false` resolve that issue?

Victorsoeby, 2023-04-28:
Hi @Serafadam, thanks for the quick reply. This solved some but not all of my issues. I can now successfully launch several of the launch files, and can have both the RGB image and depth image stream. It seems, however, that the depth align does not work correctly, as i am getting this from the camera.launch.py file: 

```
root@victor-Legion-5-Pro-16ACH6H:/# ros2 launch depthai_ros_driver camera.launch.py 
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-04-28-09-57-33-757235-victor-Legion-5-Pro-16ACH6H-178
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [192]
[INFO] [component_container-2]: process started with pid [194]
[robot_state_publisher-1] [INFO] [1682675853.922572649] [oak_state_publisher]: got segment oak
[robot_state_publisher-1] [INFO] [1682675853.922658059] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-1] [INFO] [1682675853.922682363] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-1] [INFO] [1682675853.922702685] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-1] [INFO] [1682675853.922721262] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1682675853.922740048] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-1] [INFO] [1682675853.922757786] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-1] [INFO] [1682675853.922775804] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-1] [INFO] [1682675853.922793054] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-1] [INFO] [1682675853.922810164] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-2] [INFO] [1682675854.153322066] [oak_container]: Load Library: /ws/install/lib/libdepthai_ros_driver.so
[component_container-2] [INFO] [1682675854.189981076] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1682675854.190084504] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1682675854.195066242] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [INFO] [1682675860.937841715] [oak]: Camera with MXID: 184430109159A30F00 and Name: 10.3.3.66 connected!
[component_container-2] [INFO] [1682675860.937984881] [oak]: PoE camera detected. Consider enabling low bandwidth for specific image topics (see readme).
[component_container-2] [INFO] [1682675860.960688140] [oak]: Device type: OAK-D-PRO-W-POE-97
[component_container-2] [INFO] [1682675860.965839795] [oak]: Pipeline type: RGBD
[component_container-2] [INFO] [1682675860.981345111] [oak]: NN Family: mobilenet
[component_container-2] [INFO] [1682675861.029812792] [oak]: NN input size: 300 x 300. Resizing input image in case of different dimensions.
[component_container-2] [INFO] [1682675861.068007685] [oak]: Finished setting up pipeline.
[component_container-2] [INFO] [1682675862.255802479] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-2] [184430109159A30F00] [10.3.3.66] [7.224] [StereoDepth(4)] [error] Stereo alignment error: 1, trying to recover.
[component_container-2] [184430109159A30F00] [10.3.3.66] [12.142] [StereoDepth(4)] [error] Stereo rectification error: 1, trying to recover.
[component_container-2] [184430109159A30F00] [10.3.3.66] [49.335] [StereoDepth(4)] [error] Stereo alignment error: 2, trying to recover.
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)

```
As far as i know i should have an aligned depth topic avaliable, right? 
These are the ones i can see when doing ros2 topic list
```
root@victor-Legion-5-Pro-16ACH6H:/# ros2 topic list
/btAction
/joint_states
/oak/imu/data
/oak/nn/spatial_detections
/oak/rgb/camera_info
/oak/rgb/image_raw
/oak/rgb/image_raw/compressed
/oak/rgb/image_raw/compressedDepth
/oak/rgb/image_raw/theora
/oak/stereo/camera_info
/oak/stereo/image_raw
/oak/stereo/image_raw/compressed
/oak/stereo/image_raw/compressedDepth
/oak/stereo/image_raw/theora
/parameter_events
/robot_description
/rosout
/tf
/tf_static
```
By the way i changed branch to humble for other reasons. This also has the left/right_i_set_isp_scale: false issue. 
Any thoughts on how to proceed here?

closed 2023-05-08 by Victorsoeby