luxonis/depthai-ros #302: [BUG] {connection and nodes issue}

JJJ2777, 2023-05-02:
The connection between my PC an turlebot4 doesn't work correctly.
I have: ubuntu 20.04, Ros2 Galactic and the Turlebot4 lite with the create 3 and on the turlebot4 is a raspberry pi 4

I have the simple discovery for the Networking, so my PC, Turtlebot4 and my raspberry are all in the same WI-FI.

When I don`t launch something manual these are my lists:

### **on the PC:**

**ros2 topic list**
/battery_state
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/imu
/interface_buttons
/ip
/joint_states
/joy
/mouse
/parameter_events
/robot_description
/rosout
/scan
/stereo/camera_info
/stereo/depth
/tf
/tf_static
/wheel_status

**ros2 node list**

### **on my Turtlebot4:**

**ros2 topic list**
/battery_state
/cmd_audio
/cmd_lightring
/cmd_vel
/color/preview/camera_info
/color/preview/image
/diagnostics
/diagnostics_agg
/diagnostics_toplevel_state
/dock
/hazard_detection
/imu
/interface_buttons
/ip
/ir_intensity
/ir_opcode
/joint_states
/joy
/kidnap_status
/mobility_monitor/transition_event
/mouse
/odom
/parameter_events
/robot_description
/robot_state/transition_event
/rosout
/scan
/slip_status
/static_transform/transition_event
/stereo/camera_info
/stereo/depth
/stop_status
/tf
/tf_static
/wheel_status
/wheel_ticks
/wheel_vels

**ros2 node list**
/_internal/composite_hazard
/_internal/kinematics_engine
/_internal/mobility
/_internal/stasis
/analyzers
/mobility_monitor
/motion_control
/rgb_stereo_node
/robot_state
/robot_state_publisher
/rplidar_composition
/static_transform
/system_health
/turtlebot4_diagnostics
/turtlebot4_node
/ui_mgr

My problem: I can’t see or use the nodes from my Turtlebot4. I downloaded everything with their dependencies successful.  But I can’t drive my Turtlebot4 with my PC  and Keyboard Teleoperation or Command Velocity only directly with the Raspberry. For example when I want to dock my Turtlebot4 with my PC with:

`ros2 action send_goal /dock irobot_create_msgs/action/DockServo {}`

it only says: _Waiting for an action server to become available..._
all the time but nothing happens, it only worked directly from my Turtlebot with the Raspberry

Also I can sometimes see diagnostic data from my Turtlebot4 on my PC when i manually launch on my Turtlebot4:

`ros2 launch turtlebot4_diagnostics diagnostics.launch.py`

And then I launch on my PC:

`ros2 launch turtlebot4_viz view_diagnostics.launch.py`

Then I get this node list on the PC:

**ros2 node list**
/analyzers
/robot_monitor_log_node
/rqt_gui_py_node_20110
/timeline_log_node
/timeline_subscriber
/turtlebot4_diagnostics

**And this is my diagnostic window:**
[image]

**When I also Launch:**

`ros2 launch turtlebot4_bringup oakd.launch.py`
`ros2 launch turtlebot4_bringup rplidar.launch.py`

**Then I get this Node list on my PC:** 

ros2 node list
/rgb_stereo_node
/robot_monitor_log_node
/rplidar_composition
/rqt_gui_py_node_20110
/timeline_log_node
/timeline_subscriber

(sometimes there are the same nodes twice and I get a warning message about it, because duplication of names can lead to problems)

and my diagnostic Window looks like this:
[image]

A few days ago when I launched it manually and the camera worked but now it doesn’t record any events anymore.

And the IMU has the Problem that the frequency is to high but I also don’t know how to change that.

JJJ2777, 2023-05-08:
Yeah, I think I put it in the wrong stack.
I can't change it, because we are using Galactic at my university. Does ROS2 only work with Humble?

Serafadam, 2023-05-08:
Hi, since Galactic is EOL, I'd recommend asking on [Turtlebot repository](https://github.com/turtlebot/turtlebot4) what would be the recommended steps here, when it comes to the driver itself I would either build it from source (you can use Foxy branch for that) or try using one of Docker images to run it separately and see if that works out.

JJJ2777, 2023-05-30:
Thank you for your help, I fixed the problem by changing _RMW_IMPLEMENTATION=rmw_cyclonedds_cpp_ to _RMW_IMPLEMENTATION=rmw_fastrtps_cpp_.

closed 2023-05-30 by JJJ2777