luxonis/depthai-ros #303: [BUG] RGBD launch file doesn't publish any depth

jessestevens5b, 2023-05-03:
I've just loaded up a fresh Raspberry Pi 4 running ubuntu 22.04 and ROS2 humble.

I've installed the binaries for depthai_ros_drivers and have now attempted to connect and use them as per instructions with both my OAK-D-LITE and OAK-D-PRO-FF.

I run the command:
`ros2 launch depthai_ros_driver rgbd_pcl.launch.py`

And all appears to be well:
`[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-05-03-14-48-17-968054-ubuntu-5422
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [5436]
[INFO] [component_container-2]: process started with pid [5438]
[robot_state_publisher-1] [INFO] [1683089299.682779649] [oak_state_publisher]: got segment oak
[robot_state_publisher-1] [INFO] [1683089299.683104252] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-1] [INFO] [1683089299.683159344] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-1] [INFO] [1683089299.683198491] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-1] [INFO] [1683089299.683231971] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1683089299.683263841] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-1] [INFO] [1683089299.683295044] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-1] [INFO] [1683089299.683331599] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-1] [INFO] [1683089299.683362820] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-1] [INFO] [1683089299.683393042] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-2] [INFO] [1683089299.788852912] [oak_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
[component_container-2] [INFO] [1683089300.033244733] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1683089300.033390211] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/convert_metric_node' in container '/oak_container'
[component_container-2] [INFO] [1683089300.105440042] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1683089300.105598464] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-2] [INFO] [1683089300.105647000] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-2] [INFO] [1683089300.105684129] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-2] [INFO] [1683089300.105718017] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-2] [INFO] [1683089300.105750164] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-2] [INFO] [1683089300.105783793] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziRadialNode>
[component_container-2] [INFO] [1683089300.105815181] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[component_container-2] [INFO] [1683089300.105846773] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzrgb_node' in container '/oak_container'
[component_container-2] [INFO] [1683089300.144692815] [oak_container]: Load Library: /opt/ros/humble/lib/libdepthai_ros_driver.so
[component_container-2] [INFO] [1683089300.160922848] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1683089300.161100270] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1683089300.216986678] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [INFO] [1683089303.323654419] [oak]: Camera with MXID: 18443010F1C3AB0F00 and Name: 1.1.2 connected!
[component_container-2] [INFO] [1683089303.325463764] [oak]: USB SPEED: HIGH
[component_container-2] [INFO] [1683089303.348316973] [oak]: Device type: OAK-D-PRO-FF
[component_container-2] [INFO] [1683089303.352872086] [oak]: Pipeline type: RGBD
[component_container-2] [INFO] [1683089303.383479664] [oak]: Finished setting up pipeline.
[component_container-2] [INFO] [1683089303.658166276] [oak]: Camera ready!
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
`

But I don't see anything being published to /oak/points (it exists but there is nothing publishing to it when I use ros2 topic hz to check).

I then:
`export DEPTHAI_DEBUG=1`

And try running it again. More detail here but nothing that stands out really for me as far as I can see:

`[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-05-03-14-55-06-723412-ubuntu-5581
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [5595]
[INFO] [component_container-2]: process started with pid [5597]
[robot_state_publisher-1] [INFO] [1683089708.105736775] [oak_state_publisher]: got segment oak
[robot_state_publisher-1] [INFO] [1683089708.106065438] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-1] [INFO] [1683089708.106124844] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-1] [INFO] [1683089708.106162547] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-1] [INFO] [1683089708.106196140] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1683089708.106227491] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-1] [INFO] [1683089708.106261324] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-1] [INFO] [1683089708.106292490] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-1] [INFO] [1683089708.106323490] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-1] [INFO] [1683089708.106353730] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[... 221 lines cut ...]
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.429] [system] [warning] PRINT:LeonCss: dot: 0.00 mA, flood: 0.00 mA, duty = 1.00
[component_container-2] ===== IR write bus 2: 0x04 = 0x00
[component_container-2] ===== IR write bus 2: 0x01 = 0x24
[component_container-2] 200.000 mA -> 16
[component_container-2] dot: 0.00 mA, flood: 200.00 mA, duty = 0.30
[component_container-2] ===== IR write bus 2: 0x03 = 0x10
[component_container-2] ===== IR write bus 2: 0x01 = 0x25
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.429] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] finished sippPrePrepareLinesFreeRT
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.462] [system] [warning] PRINT:LeonMss: finished sippPrePrepareLinesFreeRT
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.473] [system] [warning] PRINT:LeonCss: SH2 Reset.
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.492] [IMU(9)] [info] IMU product ID:
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.492] [IMU(9)] [info] Part 10004563 : Version 3.9.9 Build 2
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.492] [IMU(9)] [info] Part 10003606 : Version 1.8.0 Build 338
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.492] [IMU(9)] [info] Part 10004135 : Version 5.5.3 Build 162
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.492] [IMU(9)] [info] Part 10004149 : Version 5.1.12 Build 183
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [1.495] [system] [warning] PRINT:LeonCss: waited for reset 0 ms 
[... cut ...]
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [2.069] [system] [info] Memory Usage - DDR: 168.07 / 340.42 MiB, CMX: 2.10 / 2.50 MiB, LeonOS Heap: 49.10 / 77.32 MiB, LeonRT Heap: 4.54 / 41.23 MiB
[component_container-2] [18443010F1C3AB0F00] [1.1.2] [2.069] [system] [info] Temperatures - Average: 45.32 °C, CSS: 46.23 °C, MSS 44.64 °C, UPA: 46.00 °C, DSS: 44.41 °C
`

It would be lovely for a bit more of a step by step for this little guy and its companion driver as it seems there's quite a lot of jiggling to do to get things running. Even just some more commenting and possible options to configure each camera as needed to get it up and running.

Serafadam, 2023-05-03:
Hi, is this occurring when listening on topics from another device? Can you echo image topics?

Serafadam, 2023-09-14:
Closing due to inactivity, please feel free to reopen it if you have further questions/comments.

closed 2023-09-14 by Serafadam

mora200217, 2025-01-03:
Hi, updates on this ?

Serafadam, 2025-01-07:
Hi, this issue shouldn't occur, if it does for you please create a new issue with Bug template filled out.