luxonis/depthai-ros #304: [BUG] {no imu messages appear}"no new message" in ROS noetic

HanDaSeul, 2023-05-03:
stereo_inertial_launch in depthai_examples works fine. But camera.launch in depthai_ros_driver has problem. I can see /oak/imu/data after the command 'rostopic list'. But any message doesn't to be appeared. Also, keep showing "no new message" after the command 'rostopic hz /oak/imu/data'. 

I'm using with RPi4 8GB, ubuntu20.04 with ROS noetic. ROS2 foxy works fine as publishing imu messages with same camera.launch in depthai_ros_driver.

Serafadam, 2023-05-03:
Hi, which version of the driver do you have installed?

HanDaSeul, 2023-05-03:
> Hi, which version of the driver do you have installed?

2.21.2 for depthai-python, 2.7.1 for depthai_ros_driver.

HanDaSeul, 2023-05-09:
> Hi @HanDaSeul, could you check if you still get the same error when using `rgbd_pcl.launch`?

@Serafadam yes, still getting the same error.

martinlucan, 2023-05-15:
@Serafadam @HanDaSeul I faced the same issue but managed to somehow find a configuration solution. The IMU does not run in camera_i_pipeline_type: RGBD, but does work in camera_i_pipeline_type: stereo. Could you confirm that this works out also for you?

HanDaSeul, 2023-05-18:
> @Serafadam @HanDaSeul I faced the same issue but managed to somehow find a configuration solution. The IMU does not run in `camera_i_pipeline_type: RGBD`, but does work in `camera_i_pipeline_type: stereo`. Could you confirm that this works out also for you?

That problem occus to me when it is set to stereo. What environemnt are you using with? ros noetic?

martinlucan, 2023-05-18:
@HanDaSeul Lowering the IMU frequency helped out for me. Set the following values in your config.yaml. Although` rostopic hz ` anywas shows message frequency of roughly 130-140 hz.
```
i_acc_freq: 100 
i_rot_freq: 100
```

Serafadam, 2023-05-19:
Hi @HanDaSeul, could you update depthai-ros to the newest version (2.7.2) and see if that changes things for you?

HanDaSeul, 2023-05-26:
> Hi @HanDaSeul, could you update depthai-ros to the newest version (2.7.2) and see if that changes things for you?

@Serafadam sorry for the late reply. Unfortunately, nothing changed. When I set left and right camera not to publish(false)  that makes imu publishes fine. Problem occurs when those were set to publish(true).

Serafadam, 2023-09-11:
Hi @HanDaSeul , sorry for the late response, the driver has been updated quite a bit since them, do you still get issues when publishing IMU data?

joaomsbernardino, 2024-04-17:
Hi @Serafadam, not the original poster but also experiencing the exact same issue on the latest noetic branch build. The only way to get imu messages was to reduce the configured imu frequency.

alexswerner, 2024-07-10:
I'm seeing the same on ROS2 humble with depthai-ros v2.7.2, and a OAK-D-PRO-AF running on a raspberry pi 4. Interestingly a driver started with a `i_pipeline_type: stereo` pipeline does produce IMU messages as does one with `rgb`. However with `rgbd` it does not work.
I did have to reduce the publishing frequency for the IMU. The whole config looks like this:
```
/oakd:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: false
      i_floodlight_brightness: 0
      i_laser_dot_brightness: 100
      i_nn_type: none
      i_pipeline_type: rgb
      i_usb_speed: SUPER_PLUS
    imu:
      i_acc_freq: 100
      i_acc_cov: 0.0
      i_batch_report_threshold: 1
      i_enable_rotation: false
      i_gyro_cov: 0.0
      i_gyro_freq: 100
      i_mag_cov: 0.0
      i_mag_freq: 100
      i_max_batch_reports: 10
      i_message_type: IMU
      i_rot_cov: -1.0
      i_rot_freq: 100
      i_sync_method: LINEAR_INTERPOLATE_ACCEL
    rgb:
      i_board_socket_id: 0
      i_fps: 10.0
      i_height: 720
      i_interleaved: false
      i_max_q_size: 10
      i_preview_size: 256
      i_enable_preview: true
      i_low_bandwidth: true
      i_low_bandwidth_codec: jpeg
      i_low_bandwidth_preview: true
      i_low_bandwidth_preview_codec: h264
      i_keep_preview_aspect_ratio: true
      i_publish_topic: true
      i_resolution: '1080'
      i_width: 1280
    use_sim_time: false
```

I don't know which IMU type my camera has in case that matters. Any pointers would be appreciated.

Serafadam, 2024-07-12:
Hi, @alexswerner, do those issues also happen after upgrading to the new driver version?

alexswerner, 2024-07-12:
Yes, they do. However, I've now slightly modified my config and things started working. Here is the config I use now:
```
/oakd:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: false
      i_floodlight_brightness: 0
      i_laser_dot_brightness: 100
      i_nn_type: none
      i_pipeline_type: rgbd
      i_pipeline_dump: true
      i_usb_speed: SUPER_PLUS
    imu:
      i_acc_freq: 100
      i_acc_cov: 0.0
      i_batch_report_threshold: 10
      i_enable_rotation: false
      i_gyro_cov: 0.0
      i_gyro_freq: 100
      i_mag_cov: 0.0
      i_mag_freq: 100
      i_max_batch_reports: 10
      i_message_type: IMU
      i_rot_cov: -1.0
      i_rot_freq: 100
      i_sync_method: LINEAR_INTERPOLATE_ACCEL
    rgb:
      i_board_socket_id: 0
      i_fps: 10.0
      i_height: 720
      i_interleaved: false
      i_max_q_size: 10
      i_preview_size: 256
      i_enable_preview: true
      i_low_bandwidth: true
      i_low_bandwidth_codec: jpeg
      i_low_bandwidth_preview: true
      i_low_bandwidth_preview_codec: h264
      i_keep_preview_aspect_ratio: true
      i_publish_topic: true
      i_resolution: '1080'
      i_width: 1280
    stereo:
      i_align_depth: true
      i_get_base_device_timestamp: true
      i_output_disparity: false
      i_bilateral_sigma: 0
      i_board_socket_id: 0
      i_depth_filter_size: 5
      i_depth_preset: HIGH_ACCURACY
      i_disparity_width: DISPARITY_96
      i_enable_companding: false
      i_enable_decimation_filter: false
      i_enable_distortion_correction: true
      i_enable_spatial_filter: false
      i_enable_speckle_filter: false
      i_enable_temporal_filter: false
      i_enable_threshold_filter: false
      i_extended_disp: false
      i_height: 720
      i_low_bandwidth: false
      i_low_bandwidth_quality: 50
      i_lr_check: true
      i_lrc_threshold: 10
      i_max_q_size: 30
      i_rectify_edge_fill_color: 0
      i_stereo_conf_threshold: 255
      i_set_input_size: false
      i_input_width: 1280
      i_input_height: 720
      i_width: 1280
    use_sim_time: false
```