luxonis/depthai-ros #307: rgb_node Does Not Exist

swiz23, 2023-05-05:
**Describe the bug**
I'm running the following command:

ros2 launch depthai_examples rgb_publisher.launch.py

but I get this error:
[ERROR] [launch]: Caught exception in launch (see debug for traceback): executable 'rgb_node' not found on the libexec directory '/opt/ros/humble/lib/depthai_examples' 

It seems that this is because the commit at https://github.com/luxonis/depthai-ros/commit/29cf4a0140d211b7b871c93198f3433b648933ae removes the rgb_node executable, but does not update the rgb_publisher launch file to use a different node.

Maybe that node in the executable should be replaced with rgb_stereo_node?

swiz23, 2023-05-08:
Are you saying that all launch files at https://github.com/luxonis/depthai-ros/tree/humble/depthai_examples/launch will be removed in the next release?

Or those launch files will still exist, but they will launch either the depthat_ros_driver or stereo_inertial_node executables?

Serafadam, 2023-05-09:
Just the `rgb_publisher` launch file.

Serafadam, 2023-05-09:
Just `rgb_publisher` launch file will be removed, other launch files stay as is.

swiz23, 2023-05-09:
Got it! Thanks. When the next release occurs, I'll close this issue.

closed 2023-09-11 by Serafadam