luxonis/depthai-ros #312: Feed recordings into XLinkIn Node

ninoweg, 2023-05-12:
I recorded color/left/right streams as mp4 as described in [Recording](https://docs.luxonis.com/projects/sdk/en/latest/features/recording/#recording) and I would now like to feed them into my c++ pipeline (similar to [stereo_intertial_publisher.cpp](https://github.com/luxonis/depthai-ros/blob/humble/depthai_examples/src/stereo_inertial_publisher.cpp)) instead of the physical camera. 

In [Replaying](https://docs.luxonis.com/projects/sdk/en/latest/features/replaying/) it states that that results in using XLinkIn, but how can I send recorded frames to XLinkIn nodes? I couldn't find any examples on how to do so.

Serafadam, 2023-05-15:
Hi, we don't have it currently in depthai_examples, but you can  check out [depthai-core mobilenet example](https://github.com/luxonis/depthai-core/blob/main/examples/MobileNet/video_mobilenet.cpp), or if you want to pass data from ROS topic, [depthai_ros_driver](https://github.com/luxonis/depthai-ros/blob/11f8784be662a09a45c0591b3d39a2d0b7c61a66/depthai_ros_driver/src/dai_nodes/sensors/sensor_wrapper.cpp#L67).

ninoweg, 2023-05-20:
Thank's for the help, I managed to get everything running inside a ROS node. I guess I overlooked [depthai-core mobilenet example](https://github.com/luxonis/depthai-core/blob/main/examples/MobileNet/video_mobilenet.cpp).  
Regarding the recordings: It only seems to be possible to feed ColorCamera.video (or also MonoCamera.out) output directly into the VideoEncoder whereas when using ColorCamera.isp or ColorCamera.preview I get an error regarding the frame type: `[VideoEncoder(8)] [warning] Arrived frame type (8) is not either NV12 or YUV400p (8-bit Gray)`. I solved this by using ImageManip and by converting frame type to NV12. Is it also possible to feed RGB frames directly into the encoder or do I need to use ImageManip?

ninoweg, 2023-05-20:
Is there maybe also an example on how to create StereoDepth from recordings? I'm recorded both left and right MonoCamera and I'm feeding that into a StereoDepth node. I don't get any error message but the node doesn't output any depth either. I used the same setup as in [depthai-core mobilenet example](https://github.com/luxonis/depthai-core/blob/main/examples/MobileNet/video_mobilenet.cpp). My guess is that this might also be an format issue

Serafadam, 2023-05-24:
Hi @ninoweg, I guess this is related to InstanceNum in message, you need to set it so that Stereo node knows which camera ID is related to which message, you can see the example in [depthai_ros_driver](https://github.com/luxonis/depthai-ros/blob/11f8784be662a09a45c0591b3d39a2d0b7c61a66/depthai_ros_driver/src/dai_nodes/sensors/sensor_wrapper.cpp#L68)

ninoweg, 2023-05-24:
That was it, thanks a lot!

closed 2023-09-11 by Serafadam