luxonis/depthai-ros #317: Using USB2 with dai ros2 driver not possible

LukGen, 2023-05-19:
Hello,

I'm trying to use the depthai ros driver with USB2 Hardware. When launching the camera I get the following error:

Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND

When using the oak-d as webcam (https://docs.luxonis.com/en/latest/pages/oak_webcam/) It only works when forcing usb2.

Is there any possibility to force USB2 within the driver similar to when using oak-d as webcam? I already tried setting the i_usb_speed to HIGH instead of SUPER_SPEED but it didn't work.

Thanks in advance!

Serafadam, 2023-05-19:
Hi, are you able to run [demo scirpt](https://docs.luxonis.com/en/latest/#demo-script)? And which distro are you using?

ujasindro, 2023-08-03:
Hello,

I am also having the same error "X_LINK_DEVICE_NOT_FOUND" with OAK -D -Pro usb camera. I am running it on jetson agx orin inside docker container built from image luxonis/depthai-ros:humble-latest

Serafadam, 2023-08-11:
Closing due to inactivity, related issue: https://github.com/luxonis/depthai-ros/issues/363

closed 2023-08-11 by Serafadam