luxonis/depthai-ros #322: [BUG] {"No Available Devices" despite having connection in Rviz}

Cur1ousKea, 2023-05-26:
Hello there,
im operating a turtlebot4 with an oakd pro camera. Ros2 and Humble. I installed every needed component for DepthAi.
But it is not able to detect the oakd camera.
When trying to execute that command:
ros2 launch depthai_ros_driver camera.launch.py
[image]
this happens.
I get a camera image with Rviz2, so it shouldnt be a camera issue.
[image]
Here i tried executing the following: ros2 launch depthai_ros_driver rtabmap.launch.py
**Describe the bug**
A clear and concise description of what the bug is.

If available launch files don't work in your case, please check if you also get errors while running:
This Demo script also says No DepthAI device found
[image]

**Expected behavior**
It should find the cam and execute the program.

**Attach system log**

 - Which OS/OS version are you using? Ubuntu 22.04.
 - Which ROS version are you using? ROS2
 - Is `depthai-ros` built from source or installed from apt? apt
 - Is `depthai/depthai-core` library installed from rosdep or manually? rosdep
 - Please include versions of following packages - The newest i installed them today from scratch
[launch.log](https://github.com/luxonis/depthai-ros/files/11570506/launch.log)
[launch.log](https://github.com/luxonis/depthai-ros/files/11570512/launch.log)

tschinfra, 2023-05-26:
I got almost the same Problem but in my case with the following launch file:
`ros2 launch depthai_ros_driver pointcloud.launch.py `

Console output:
```
[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-05-26-08-08-25-039473-ubuntu-2191
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [2205]
[INFO] [component_container-2]: process started with pid [2207]
[component_container-2] [INFO] [1685088507.010003168] [oak_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
[robot_state_publisher-1] [INFO] [1685088507.121141398] [oak_state_publisher]: got segment oak
[robot_state_publisher-1] [INFO] [1685088507.121540913] [oak_state_publisher]: got segment oak-d_frame
[robot_state_publisher-1] [INFO] [1685088507.121617523] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-1] [INFO] [1685088507.121658615] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-1] [INFO] [1685088507.121688430] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1685088507.121716522] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-1] [INFO] [1685088507.121744170] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-1] [INFO] [1685088507.121771484] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-1] [INFO] [1685088507.121799873] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-1] [INFO] [1685088507.121827502] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-2] [INFO] [1685088507.428169271] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1685088507.428322381] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/convert_metric_node' in container '/oak_container'
[component_container-2] [INFO] [1685088507.559210442] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1685088507.559357422] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-2] [INFO] [1685088507.559525328] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-2] [INFO] [1685088507.559560198] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-2] [INFO] [1685088507.559591401] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-2] [INFO] [1685088507.559618883] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-2] [INFO] [1685088507.559646771] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzi' in container '/oak_container'
[component_container-2] [INFO] [1685088507.667742978] [oak_container]: Load Library: /opt/ros/humble/lib/libdepthai_ros_driver.so
[component_container-2] [INFO] [1685088507.687540901] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1685088507.687711455] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1685088507.766448171] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [ERROR] [1685088516.102667096] [oak]: Failed to connect to device, error message: X_LINK_DEVICE_ALREADY_IN_USE
[component_container-2] [INFO] [1685088516.102904186] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [ERROR] [1685088524.408346321] [oak]: Failed to connect to device, error message: X_LINK_DEVICE_ALREADY_IN_USE
[component_container-2] [INFO] [1685088524.409672066] [oak]: No ip/mxid specified, connecting to the next available device.

```

Log file 'launch.log':
```
1685088505.0432942 [INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2023-05-26-08-08-25-039473-ubuntu-2191
1685088505.0439813 [INFO] [launch]: Default logging verbosity is set to INFO
1685088506.5084186 [INFO] [robot_state_publisher-1]: process started with pid [2205]
1685088506.5094092 [INFO] [component_container-2]: process started with pid [2207]
1685088507.1232398 [robot_state_publisher-1] [INFO] [1685088507.121141398] [oak_state_publisher]: got segment oak
1685088507.1240950 [robot_state_publisher-1] [INFO] [1685088507.121540913] [oak_state_publisher]: got segment oak-d_frame
1685088507.1246538 [robot_state_publisher-1] [INFO] [1685088507.121617523] [oak_state_publisher]: got segment oak_imu_frame
1685088507.1257315 [robot_state_publisher-1] [INFO] [1685088507.121658615] [oak_state_publisher]: got segment oak_left_camera_frame
1685088507.1263344 [robot_state_publisher-1] [INFO] [1685088507.121688430] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[... cut ...]
1685089157.0894380 [INFO] [component_container-2]: sending signal 'SIGTERM' to process[component_container-2]
1685089162.0846329 [ERROR] [component_container-2]: process[component_container-2] failed to terminate '10.0' seconds after receiving 'SIGTERM', escalating to 'SIGKILL'
1685089162.0870111 [INFO] [component_container-2]: sending signal 'SIGKILL' to process[component_container-2]
1685089162.1132691 [ERROR] [component_container-2]: process has died [pid 2207, exit code -9, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args -r __node:=oak_container -r __ns:=/']
```

Serafadam, 2023-05-26:
Hi, could you also post debug logs from depthai_demo?

Cur1ousKea, 2023-05-26:
there you go, i hope this is what u need
[output.txt](https://github.com/luxonis/depthai-ros/files/11576218/output.txt)

Cur1ousKea, 2023-05-26:
[output.txt](https://github.com/luxonis/depthai-ros/files/11576779/output.txt)
okay i changed a few things but now its even more confusing but it seems to find sth now

Cur1ousKea, 2023-08-07:
Is there an update to this topic?

Serafadam, 2023-08-10:
Hi @Cur1ousKea  @tschinfra , sorry for the delay, this issue got lost in the backlog, let's get back to it:
- Did you update the libraries for the drivers?
- Does OAK-D-Pro have an external power input, or is it connected only to RPi USB?

Cur1ousKea, 2023-08-12:
Yes the libraries are up to date. 
No the Oak-D is directly connected via USB to the RPi.
Another question. Where do i have to install it to? Shall i install it directly on the RPi or on my local machine and using the ros2 connection to the robot?
Furthermore, i installed it to the RPi and none of the given executeables are working. mostly with the same error:
[component_container-2] [ERROR] [1691840532.598751139] [oak]: No available devices
[component_container-2] [INFO] [1691840532.600808500] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [ERROR] [1691840551.352476126] [oak]: Failed to find device after booting, error message: X_LINK_DEVICE_NOT_FOUND
Its always kinda confusing because the camera shuts down once the robot is docked and throws other errors, but while moving this error persists. 
[camera model.txt](https://github.com/luxonis/depthai-ros/files/12327270/camera.model.txt)
[pointscloud error.txt](https://github.com/luxonis/depthai-ros/files/12327271/pointscloud.error.txt)

Serafadam, 2023-08-16:
> X_LINK_DEVICE_NOT_FOUND
Can you try checking [udev rules?](https://docs.luxonis.com/en/latest/pages/troubleshooting/#udev-rules-on-linux)
> No the Oak-D is directly connected via USB to the RPi.
Not sure what's the exact setup on Turtlebot, but generally with RPi we recommend using [Y-adapters](https://shop.luxonis.com/collections/accessories/products/oak-y-adapter), as sometimes the power draw can cause issues on those boards (more information [here](https://docs.luxonis.com/projects/hardware/en/latest/pages/guides/raspberrypi/#powering-oak-from-rpi)). Weird behavior when docking also gives hints that this might be the issue, though I would start with contacting the Turtlebot team if that issue is related to general setup or if there is some error on the hardware level. You could also try running different configuration (disable IR drivers/NN).
Also, is the camera working normally when connected to the PC?
> Where do i have to install it to? Shall i install it directly on the RPi or on my local machine and using the ros2 connection to the robot?
Do you mean where depthai-ros is installed/run? It should be installed on the RPi, unless there has been a change in Turtlebot setup.

Cur1ousKea, 2023-08-18:
Thank you for your response!
What worked for me was: export DEPTHAI_DEBUG=1
So i have only one problem left. 
Once i am starting the rtabmap.launch.py on the Turtlebot4 everything seems fine.
[rtabdebug.txt](https://github.com/luxonis/depthai-ros/files/12384125/rtabdebug.txt)
I cant start Rviz2 from the Turtlebot itself, so i have to start it manually from the external Linux machine. 
But that shouldnt be a problem. For Example rdbd_pcl.launch.py works with pointcloud2 without any issue.
Could you maybe give me the details how the rviz setup of the native rtabmap.launch.py is?
Or if you maybe found any issue in the log. 
Thanks!

Serafadam, 2023-08-21:
It's a bit weird that enabling the debug flags gets rid of the problem, I will share that with the team. 
Regarding rtabmap,[ the launch file currently available](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/launch/rtabmap.launch.py) is used mostly as an example for verification. I would suggest creating a separate file, as:
- Current launch file additionally runs `rtabmap_viz`, which is used to visualize SLAM results (not needed if you run rtabmap on a robot), you could probably run it on external linux machine, although this may cause lags/errors due to DDS/WiFi setup
- RTABMap parameters are set up for handheld demo running on a laptop, for RPi you would need to adjust them accordingly, more information on that on the[ RTABMap wiki ](http://wiki.ros.org/rtabmap_ros/Tutorials/Advanced%20Parameter%20Tuning)
- For RPi, you might additionally need to lower camera FPS/resolution to optimize further
- RGBD Odometry will additionally increase CPU usage, you can try checking if Odometry provided by Turtlebot's wheels is enough
- Regarding Rviz setup, I would advise to subscribe only to the most important/least resource hungry sources, as large amount of data (raw images and pointclouds) can decrease performance of the whole system. We don't have an example RTABMap Rviz config file, as this is something that depends largely on user.
- Instead of Rviz, you could try using [Foxglove](https://foxglove.dev/) with `foxglove_bridge` running on the robot as it could improve performance

Cur1ousKea, 2023-08-29:
Okay i tried around a while and wasnt able to start rtabmap_viz on my own, would you mind sharing the launching command with me? Thanks!

Serafadam, 2023-08-31:
Hi, unfortunately I don't have Turtlebot at hand, so I can't verify that, but I would personally start with removing rtabmap_viz from rtabmap.launch.py, and see if topics are working (you can also remove RGBDOdometry if it takes too much CPU, and use odom from the robot diff drive). After you launch the whole stack, you should be able to visualize topics/slam results with Rviz running on host.

Cur1ousKea, 2023-09-10:
Alright, i tried it and it works thanks!
for anyone having the same issue:
edit rtabmap.launch,py on turtlebot ~/dai_ws/src/depthai-ros/depthai_ros_driver/launch/rtabmap.launch.py
remove the line with rtabmap_viz within the file
after saving the file rebuild the repository ->  cd  ~/dai_ws ->  ./src/depthai-ros/build.sh

PC: ros2 run rtabmap_viz rtabmap_viz
Turtlebot4: ros2 launch depthai_ros_driver rtabmap.launch.py

closed 2023-09-10 by Cur1ousKea

Ragnar7982, 2024-08-12:
@Cur1ousKea Hello, I'm also using Turtlebot4 and RTAB-map, do you mind share the depthai_ros_driver rtabmap.launch.py setting?  Thank you.
OAK-D-PRO is very hard to use when wireless...

Cur1ousKea, 2024-08-12:
Sorry, it was a Bachelorthesisproject and i deleted everthing afterwards because it annoyed me that much...

Ragnar7982, 2024-08-12:
@Cur1ousKea Hi, thank you for the reply, can you tell me what was your steps about remote pc and raspberry pi4(connect to oak-d pro camera) if you remember?
And you successfully using remote pc and raspberry pi4(connect to oak-d pro camera) to run RTAB-map?
Thank you!