luxonis/depthai-ros #324: [BUG] {Title of the bug} Oak-d-lite launch file not running

youngjae41, 2023-06-02:
OS : Ubuntu 20.04
Hardware : jetson orin nano
ROS : noetic

oak-d lite launch file doesn't work 

There are 2 launch files that work

1. stereo_node.launch
2. stereo_nodelet.launch

Doesn't oak-d-lite support color image topic ?

Serafadam, 2023-06-05:
Hi, could you post some more info according to the [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)?

youngjae41, 2023-06-07:
@Serafadam 

The following error occurs when rgb_stereo.launch is operated and the point cloud and image are not output

[1844301041D2C11200] [1.2.2] [0.855] [MonoCamera(0)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[1844301041D2C11200] [1.2.2] [0.855] [MonoCamera(1)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P

Should I use stereo_inertial_node.launch?

youngjae41, 2023-06-08:
@AnujPokhrel Thank you for answer.
I also installed ros-noetic-depthai-ros and downloaded and built the code, but the same thing happens. It works fine on oak-d-pro, but not on oak-d-lite.

HomeworldL, 2023-06-12:
I also encountered the same problem. It seems that the IMU is not detected, but the oak-d-lite does not have an IMU.

[component_container-2] [ERROR] [1686574432.480990432] [oak_container]: Component constructor threw an exception: IMU(9) - IMU invalid settings!: IMU not detected. Your board doesn't have IMU or requires firmware update (Set imu.enableFirmwareUpdate(True) explicitly.). 0, 328
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: IMU(9) - IMU invalid settings!: IMU not detected. Your board doesn't have IMU or requires firmware update (Set imu.enableFirmwareUpdate(True) explicitly.). 0, 328

Faisalsouz, 2023-06-13:
** Problem description ** 
I'm testing OAK-DLITE on jetson Nano, ROS noetic, OS: focal 20.04
I have tried to build both from source and binary installation but, no success. 
I,m getting following error 

```bash
etting /run_id to 9c5c8d34-09d8-11ee-a7cf-3b6c5c5cf392
process[rosout-1]: started with pid [7200]
started core service [/rosout]
process[oak_state_publisher-2]: started with pid [7207]
process[oak_nodelet_manager-3]: started with pid [7208]
process[oak-4]: started with pid [7209]
[ INFO] [1686653644.725360716]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1686653644.728352507]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1686653644.753956767]: Initializing nodelet with 8 worker threads.
[ INFO] [1686653644.769778609]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[ INFO] [1686653644.961498816]: No ip/mxid/usb_id specified, connecting to the next available device.
[ INFO] [1686653647.674180198]: Camera with MXID: 1844301061F6B11200 and Name: 1.3 connected!
[ INFO] [1686653647.675211572]: USB SPEED: HIGH
[ INFO] [1686653647.697818723]: Device type: OAK-D-LITE
[ WARN] [1686653647.699806738]: Device has no IR drivers
[ INFO] [1686653647.701143723]: Pipeline type: RGBD
[ INFO] [1686653647.741798182]: NN Family: mobilenet
[ INFO] [1686653647.833975353]: Finished setting up pipeline.
[1844301061F6B11200] [1.3] [1.702] [MonoCamera(4)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[1844301061F6B11200] [1.3] [1.702] [MonoCamera(6)] [error] OV7251 only supports THE_480_P/THE_400_P resolutions, defaulting to THE_480_P
[FATAL] [1686653657.613480170]: Failed to load nodelet '/oak` of type `depthai_ros_driver/Camera` to manager `oak_nodelet_manager'
[oak_nodelet_manager-3] process has died [pid 7208, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager /nodelet_manager/load_nodelet:=oak/nodelet_manager/load_nodelet /nodelet_manager/unload_nodelet:=oak/nodelet_manager/unload_nodelet /nodelet_manager/list:=oak/nodelet_manager/list __name:=oak_nodelet_manager __log:=/home/souz/.ros/log/9c5c8d34-09d8-11ee-a7cf-3b6c5c5cf392/oak_nodelet_manager-3.log].
log file: /home/souz/.ros/log/9c5c8d34-09d8-11ee-a7cf-3b6c5c5cf392/oak_nodelet_manager-3*.log
================================================================================REQUIRED process [oak-4] has died!
process has died [pid 7209, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet load depthai_ros_driver/Camera oak_nodelet_manager __name:=oak __log:=/home/souz/.ros/log/9c5c8d34-09d8-11ee-a7cf-3b6c5c5cf392/oak-4.log].
log file: /home/souz/.ros/log/9c5c8d34-09d8-11ee-a7cf-3b6c5c5cf392/oak-4*.log
Initiating shutdown!
================================================================================
[oak-4] killing on exit
[oak_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
```
I have also tried to change the config file with width and height  640×480 (480P) but no success.

Serafadam, 2023-06-14:
@youngjae41 @HomeworldL those errors are related to IMU configuration as those cameras might not have it installed, I'll add a fix for that this week, for now you can mitigate that by setting `camera_i_enable_imu: false`

youngjae41, 2023-06-15:
@Serafadam Do you add it as a param to launch file?

closed 2023-09-11 by Serafadam