luxonis/depthai-ros #328: [custom] Provide a general summary of the issue

osama-z-salah, 2023-06-06:
my platform is: Linux Ubuntu 22 (x86_64) + ROS Humble+ OAK-D-Pro

When I launch these launch files:
ros2 launch depthai_ros_driver example_segmentation.launch.py 
ros2 launch depthai_ros_driver stereo_from_rosbag.launch.py 

I have get the following error:

Serafadam, 2023-06-06:
Hi, thanks for the information, could you post more information according to [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)?

osama-z-salah, 2023-06-07:
**Describe the bug:** 
[ERROR] [component_container-2]: process has died [pid 58940, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level info --ros-args -r __node:=oak_container -r __ns:=/'].

**Minimal Reproducible Example:**
1. When i run the stereo_inertial_node in depthai_examples, it able to run succesfully without any error.
2. When i run camera in depthai_ros_driver, at 1st it able to run and in middle way, i got error where the process is died.

**Is depthai-ros built from source or installed from apt?** 
from source 

**Is depthai/depthai-core library installed from rosdep or manually? For rosdep install, check if ros-<rosdistro>-depthai package is installed, manual install can be checked with ldconfig -p | grep depthai**

I install the depthai package, using sudo apt-get install ros-foxy-depthai

**Please include versions of following packages - apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver**

Package: ros-humble-depthai
Version: 2.21.2-1jammy.20230425.234727

Package: ros-humble-depthai-ros
Version: 2.7.2-1jammy.20230513.013547

Package: ros-humble-depthai-bridge
Version: 2.7.2-1jammy.20230513.010952

Package: ros-humble-depthai-ros-msgs
Version: 2.7.2-1jammy.20230513.003520

Package: ros-humble-depthai-ros-driver
Version: 2.7.2-1jammy.20230513.012742

jmrp98, 2023-06-07:
I also had the process dying when using the 2.7.2 Humble  built from source in Ubuntu 20.04 Focal in an Nvidia Jetson AGX Orin. I tried using the provided Docker version and the same happens. I rolled back to 2.7.1 and I don't have problem there , I tested 2.7.1 built from source and provided  Docker container as well. I don't remember if the error message was the same as the person who opened this issue , but I can run it again tomorrow and share here if that helps.

Serafadam, 2023-06-07:
Could you run `export DEPTHAI_DEBUG=1` and post logs that you get after running launch files?

osama-z-salah, 2023-06-14:
nanoamr@nanoamr:~/ros2_ws$ ros2 launch depthai_ros_driver camera.launch.py 
[INFO] [launch]: All log files can be found below /home/nanoamr/.ros/log/2023-06-14-18-57-40-522575-nanoamr-79330
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [79344]
[INFO] [component_container-2]: process started with pid [79346]
[component_container-2] [DEBUG] [1686740260.684359845] [rclcpp]: signal handler installed
[component_container-2] [DEBUG] [1686740260.684390375] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[component_container-2] [DEBUG] [1686740260.684402947] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[component_container-2] [DEBUG] [1686740260.684430444] [rcl]: Initializing node 'ComponentManager' in namespace ''
[component_container-2] [DEBUG] [1686740260.684438028] [rcl]: Using domain ID of '0'
[component_container-2] [DEBUG] [1686740260.690127020] [rcl]: Initializing publisher for topic name '/rosout'
[component_container-2] [DEBUG] [1686740260.690141205] [rcl]: Expanded and remapped topic name '/rosout'
[component_container-2] [DEBUG] [1686740260.690814858] [rcl]: Publisher initialized
[component_container-2] [DEBUG] [1686740260.690824797] [rcl]: Node initialized
[component_container-2] [DEBUG] [1686740260.690884019] [rcl]: Initializing subscription for topic name '/parameter_events'
[component_container-2] [DEBUG] [1686740260.690887178] [rcl]: Expanded and remapped topic name '/parameter_events'
[component_container-2] [DEBUG] [1686740260.691513083] [rcl]: Subscription initialized
[component_container-2] [DEBUG] [1686740260.691594846] [rcl]: Initializing service for service name '~/_container/load_node'
[component_container-2] [DEBUG] [1686740260.691600232] [rcl]: Expanded and remapped service name '/oak_container/_container/load_node'
[component_container-2] [DEBUG] [1686740260.691855190] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1686740260.691862105] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/load_nodeRequest
[component_container-2] [DEBUG] [1686740260.691863950] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/load_nodeReply
[component_container-2] [DEBUG] [1686740260.691865161] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1686740260.691892820] [rcl]: Service initialized
[component_container-2] [DEBUG] [1686740260.691902537] [rcl]: Initializing service for service name '~/_container/unload_node'
[component_container-2] [DEBUG] [1686740260.691906810] [rcl]: Expanded and remapped service name '/oak_container/_container/unload_node'
[component_container-2] [DEBUG] [1686740260.692021189] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1686740260.692024088] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/unload_nodeRequest
[component_container-2] [DEBUG] [1686740260.692025461] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/unload_nodeReply
[component_container-2] [DEBUG] [1686740260.692026607] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1686740260.692053658] [rcl]: Service initialized
[component_container-2] [DEBUG] [1686740260.692060410] [rcl]: Initializing service for service name '~/_container/list_nodes'
[component_container-2] [DEBUG] [1686740260.692063730] [rcl]: Expanded and remapped service name '/oak_container/_container/list_nodes'
[component_container-2] [DEBUG] [1686740260.692485525] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1686740260.692496768] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/list_nodesRequest
[component_container-2] [DEBUG] [1686740260.692500364] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/list_nodesReply
[component_container-2] [DEBUG] [1686740260.692502940] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1686740260.692544316] [rcl]: Service initialized
[component_container-2] [DEBUG] [1686740260.694447634] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1686740260.694472844] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1686740260.694616943] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1686740260.694638960] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1686740260.694649044] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1686740260.694656222] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1686740260.694659917] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1686740260.694662031] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1686740260.694665204] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1686740260.694666961] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1686740260.694669960] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1686740260.694672191] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1686740260.694685037] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1686740260.694688361] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1686740260.694693381] [rcl]: Subscription taking message
[... cut ...]
[component_container-2] [1844301071C7AB0F00] [1.13] [21.923] [system] [info] Cpu Usage - LeonOS 86.95%, LeonRT: 49.77%
[ERROR] [component_container-2]: process has died [pid 79346, exit code -11, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level debug --ros-args -r __node:=oak_container -r __ns:=/'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1686740343.198757355] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 79344]

Shivam7Sharma, 2024-07-01:
Hi,

I am getting the same error when I launch the camera.launch.py in any mode.

How do I solve this?

 The following is the debug output link:
 [Link](https://drive.google.com/file/d/1kHsRQblpu2uSca5E4dy3BbVnuztjfXCo/view?usp=drive_link)

Serafadam, 2024-07-02:
Hi @Shivam7Sharma , could you run the camera node with gdb to find the exact point of error? You can try using `use_gdb:=true` launch argument which should make it easier

Shivam7Sharma, 2024-07-02:
Thank you for the quick reply. The following is the output with the gdb:

```
shivam157@ubuntu:~/dai_ws_non_prod$ ros2 launch depthai_ros_driver camera.launch.py use_gdb:=true
[INFO] [launch]: All log files can be found below /home/shivam157/.ros/log/2024-07-02-11-21-14-677889-ubuntu-9096
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gdb-1]: process started with pid [9113]
[gdb-1] GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
[gdb-1] Copyright (C) 2022 Free Software Foundation, Inc.
[gdb-1] License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
[gdb-1] This is free software: you are free to change and redistribute it.
[gdb-1] There is NO WARRANTY, to the extent permitted by law.
[gdb-1] Type "show copying" and "show warranty" for details.
[gdb-1] This GDB was configured as "aarch64-linux-gnu".
[gdb-1] Type "show configuration" for configuration details.
[gdb-1] For bug reporting instructions, please see:
[gdb-1] <https://www.gnu.org/software/gdb/bugs/>.
[gdb-1] Find the GDB manual and other documentation resources online at:
[gdb-1]     <http://www.gnu.org/software/gdb/documentation/>.
[gdb-1] 
[gdb-1] For help, type "help".
[gdb-1] Type "apropos word" to search for commands related to "word"...
[gdb-1] Reading symbols from /opt/ros/humble/lib/rclcpp_components/component_container...
[gdb-1] (No debugging symbols found in /opt/ros/humble/lib/rclcpp_components/component_container)
[gdb-1] Starting program: /opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level info --ros-args -r __node:=oak_container -r __ns:=/
[gdb-1] [Thread debugging using libthread_db enabled]
[gdb-1] Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[gdb-1] [New Thread 0xfffff667e8e0 (LWP 9134)]
[gdb-1] [New Thread 0xfffff5e6e8e0 (LWP 9135)]
[gdb-1] [New Thread 0xfffff55d78e0 (LWP 9136)]
[gdb-1] [New Thread 0xfffff4dc78e0 (LWP 9137)]
[gdb-1] [New Thread 0xfffff45b78e0 (LWP 9138)]
[gdb-1] [New Thread 0xfffff3da78e0 (LWP 9139)]
[gdb-1] [New Thread 0xfffff35978e0 (LWP 9140)]
[gdb-1] [New Thread 0xfffff2cc78e0 (LWP 9141)]
[gdb-1] [New Thread 0xfffff249e8e0 (LWP 9154)]
[gdb-1] [INFO] [1719933676.666819772] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[gdb-1] [INFO] [1719933676.828926148] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[gdb-1] [INFO] [1719933676.829058808] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[gdb-1] [INFO] [1719933676.889019062] [oak_state_publisher]: got segment oak
[gdb-1] [INFO] [1719933676.889169257] [oak_state_publisher]: got segment oak-d-base-frame
[gdb-1] [INFO] [1719933676.889189095] [oak_state_publisher]: got segment oak_imu_frame
[gdb-1] [INFO] [1719933676.889198855] [oak_state_publisher]: got segment oak_left_camera_frame
[gdb-1] [INFO] [1719933676.889207590] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[gdb-1] [INFO] [1719933676.889216229] [oak_state_publisher]: got segment oak_model_origin
[gdb-1] [INFO] [1719933676.889224580] [oak_state_publisher]: got segment oak_rgb_camera_frame
[gdb-1] [INFO] [1719933676.889234851] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[gdb-1] [INFO] [1719933676.889243555] [oak_state_publisher]: got segment oak_right_camera_frame
[gdb-1] [INFO] [1719933676.889251362] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[gdb-1] [INFO] [1719933676.912779848] [oak_container]: Load Library: /home/shivam157/dai_ws_non_prod/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container 'oak_container'
[gdb-1] [New Thread 0xffffceffe8e0 (LWP 9173)]
[gdb-1] [New Thread 0xffffce7ee8e0 (LWP 9174)]
[gdb-1] [New Thread 0xffffc9fde8e0 (LWP 9175)]
[gdb-1] [New Thread 0xffffc97ce8e0 (LWP 9176)]
[gdb-1] [New Thread 0xffffc6fbe8e0 (LWP 9177)]
[gdb-1] [INFO] [1719933679.188285504] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[gdb-1] [INFO] [1719933679.188433331] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[gdb-1] [New Thread 0xffffc274e8e0 (LWP 9178)]
[gdb-1] [New Thread 0xffffc1f3e8e0 (LWP 9179)]
[gdb-1] [New Thread 0xffffbbffe8e0 (LWP 9180)]
[gdb-1] [INFO] [1719933679.224965184] [oak]: No ip/mxid specified, connecting to the next available device.
[gdb-1] [Thread 0xffffc1f3e8e0 (LWP 9179) exited]
[gdb-1] [Thread 0xffffc274e8e0 (LWP 9178) exited]
[gdb-1] [New Thread 0xffffc172e8e0 (LWP 9243)]
[gdb-1] [New Thread 0xffffc0f1e8e0 (LWP 9244)]
[gdb-1] [New Thread 0xffffbb7ee8e0 (LWP 9248)]
[gdb-1] [New Thread 0xffffbafde8e0 (LWP 9249)]
[gdb-1] [New Thread 0xffffba7ce8e0 (LWP 9251)]
[gdb-1] [New Thread 0xffffb976b8e0 (LWP 9253)]
[gdb-1] [INFO] [1719933683.890287715] [oak]: Camera with MXID: 14442C10E14FCBD600 and Name: 1.2.1 connected!
[gdb-1] [INFO] [1719933683.891209364] [oak]: USB SPEED: SUPER
[gdb-1] [INFO] [1719933683.925123486] [oak]: Device type: OAK-FFC-3P
[gdb-1] [INFO] [1719933683.928247922] [oak]: Pipeline type: Depth
[gdb-1] 
[gdb-1] Thread 1 "component_conta" received signal SIGSEGV, Segmentation fault.
[gdb-1] 0x0000fffff79adbf4 in __GI___libc_free (mem=0x7468676972) at ./malloc/malloc.c:3368
[gdb-1] 3368    ./malloc/malloc.c: No such file or directory.
```

Serafadam, 2024-07-04:
Could you install additional debug symbols?
```
sudo apt install -y ros-humble-rclcpp-dbgsym ros-humble-rclcpp-components-dbgsym ros-humble-depthai-ros-driver-dbgsym
```

Shivam7Sharma, 2024-07-04:
@Serafadam  I installed the things you said. What should I do now?

Serafadam, 2024-07-10:
Hi, sorry for the delay, if you installed them you should be able to get more debug output from running those commands

Shivam7Sharma, 2024-07-19:
@Serafadam Thank for your reply. The following is the output which I think is the same as before:

```
shivam157@ubuntu:~/dai_ws_non_prod/src$ sudo apt install -y ros-humble-rclcpp-dbgsym ros-humble-rclcpp-components-dbgsym ros-humble-depthai-ros-driver-dbgsym
[sudo] password for shivam157: 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-humble-rclcpp-components-dbgsym is already the newest version (16.0.9-1jammy.20240524.033335).
ros-humble-rclcpp-dbgsym is already the newest version (16.0.9-1jammy.20240524.023929).
The following packages were automatically installed and are no longer required:
  cppcheck libbullet-dev libbullet3.06 libdecor-0-dev libibus-1.0-dev libpulse-dev libsdl2-dev libsndio-dev libxv-dev
  libxxf86vm-dev libz3-4 pybind11-dev pydocstyle pyflakes3 python3-flake8 python3-mccabe python3-netifaces
  python3-pycodestyle python3-pydocstyle python3-pyflakes python3-snowballstemmer ros-humble-action-tutorials-cpp
  ros-humble-action-tutorials-interfaces ros-humble-action-tutorials-py ros-humble-actionlib-msgs
  ros-humble-ament-cmake-copyright ros-humble-ament-cmake-cppcheck ros-humble-ament-cmake-cpplint
  ros-humble-ament-cmake-flake8 ros-humble-ament-cmake-lint-cmake ros-humble-ament-cmake-pep257
  ros-humble-ament-cmake-uncrustify ros-humble-ament-cmake-xmllint ros-humble-ament-copyright ros-humble-ament-cppcheck
  ros-humble-ament-cpplint ros-humble-ament-flake8 ros-humble-ament-lint ros-humble-ament-lint-cmake
  ros-humble-ament-lint-common ros-humble-ament-pep257 ros-humble-ament-uncrustify ros-humble-ament-xmllint
  ros-humble-common-interfaces ros-humble-composition ros-humble-demo-nodes-cpp ros-humble-demo-nodes-cpp-native
  ros-humble-demo-nodes-py ros-humble-depthimage-to-laserscan ros-humble-dummy-map-server
  ros-humble-dummy-robot-bringup ros-humble-dummy-sensors ros-humble-example-interfaces
  ros-humble-examples-rclcpp-minimal-action-client ros-humble-examples-rclcpp-minimal-action-server
  ros-humble-examples-rclcpp-minimal-client ros-humble-examples-rclcpp-minimal-composition
  ros-humble-examples-rclcpp-minimal-publisher ros-humble-examples-rclcpp-minimal-service
  ros-humble-examples-rclcpp-minimal-subscriber ros-humble-examples-rclcpp-minimal-timer
  ros-humble-examples-rclcpp-multithreaded-executor ros-humble-examples-rclpy-executors
  ros-humble-examples-rclpy-minimal-action-client ros-humble-examples-rclpy-minimal-action-server
  ros-humble-examples-rclpy-minimal-client ros-humble-examples-rclpy-minimal-publisher
  ros-humble-examples-rclpy-minimal-service ros-humble-examples-rclpy-minimal-subscriber ros-humble-geometry2
  ros-humble-image-tools ros-humble-intra-process-demo ros-humble-joy ros-humble-keyboard-handler
  ros-humble-launch-testing-ros ros-humble-lifecycle ros-humble-logging-demo ros-humble-pendulum-control
  ros-humble-pendulum-msgs ros-humble-pybind11-vendor ros-humble-qt-dotgraph ros-humble-qt-gui-py-common
  ros-humble-quality-of-service-demo-cpp ros-humble-quality-of-service-demo-py ros-humble-ros-base ros-humble-ros-core
  ros-humble-ros2action ros-humble-ros2bag ros-humble-ros2cli ros-humble-ros2cli-common-extensions
  ros-humble-ros2component ros-humble-ros2doctor ros-humble-ros2interface ros-humble-ros2launch
  ros-humble-ros2lifecycle ros-humble-ros2multicast ros-humble-ros2node ros-humble-ros2param ros-humble-ros2pkg
  ros-humble-ros2run ros-humble-ros2service ros-humble-ros2topic ros-humble-rosbag2 ros-humble-rosbag2-compression
  ros-humble-rosbag2-compression-zstd ros-humble-rosbag2-interfaces ros-humble-rosbag2-py ros-humble-rosbag2-transport
  ros-humble-rosidl-runtime-py ros-humble-rqt-action ros-humble-rqt-bag ros-humble-rqt-bag-plugins
  ros-humble-rqt-console ros-humble-rqt-graph ros-humble-rqt-gui ros-humble-rqt-gui-cpp ros-humble-rqt-gui-py
  ros-humble-rqt-msg ros-humble-rqt-plot ros-humble-rqt-publisher ros-humble-rqt-py-common ros-humble-rqt-py-console
  ros-humble-rqt-reconfigure ros-humble-rqt-service-caller ros-humble-rqt-shell ros-humble-rqt-srv ros-humble-rqt-topic
  ros-humble-rttest ros-humble-rviz2 ros-humble-sdl2-vendor ros-humble-shape-msgs ros-humble-sros2
  ros-humble-sros2-cmake ros-humble-teleop-twist-joy ros-humble-teleop-twist-keyboard ros-humble-tf2-bullet
  ros-humble-tf2-eigen-kdl ros-humble-tf2-kdl ros-humble-tf2-tools ros-humble-tlsf ros-humble-tlsf-cpp
  ros-humble-topic-monitor ros-humble-trajectory-msgs ros-humble-turtlesim ros-humble-uncrustify-vendor
  ros-humble-zstd-vendor uncrustify
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  ros-humble-compressed-depth-image-transport ros-humble-compressed-image-transport ros-humble-depth-image-proc
  ros-humble-depthai ros-humble-depthai-bridge ros-humble-depthai-descriptions ros-humble-depthai-examples
  ros-humble-depthai-ros-driver ros-humble-depthai-ros-msgs ros-humble-image-pipeline ros-humble-image-proc
  ros-humble-image-publisher ros-humble-image-rotate ros-humble-image-transport-plugins ros-humble-image-view
  ros-humble-stereo-image-proc ros-humble-theora-image-transport ros-humble-tracetools-image-pipeline
The following NEW packages will be installed:
  ros-humble-compressed-depth-image-transport ros-humble-compressed-image-transport ros-humble-depth-image-proc
  ros-humble-depthai ros-humble-depthai-bridge ros-humble-depthai-descriptions ros-humble-depthai-examples
[... cut ...]
[gdb-1] [INFO] [1721412260.835496335] [oak]: Camera with MXID: 14442C10E14FCBD600 and Name: 1.2.1 connected!
[gdb-1] [INFO] [1721412260.839200025] [oak]: USB SPEED: SUPER
[gdb-1] [INFO] [1721412260.875123464] [oak]: Device type: OAK-FFC-3P
[gdb-1] [INFO] [1721412260.879013526] [oak]: Pipeline type: Depth
[gdb-1] 
[gdb-1] Thread 1 "component_conta" received signal SIGSEGV, Segmentation fault.
[gdb-1] 0x0000fffff79adbf4 in __GI___libc_free (mem=0x6570007468676972) at ./malloc/malloc.c:3368
[gdb-1] 3368    ./malloc/malloc.c: No such file or directory.

```

Serafadam, 2024-08-05:
Hi, sorry for the late reply, could you output the logs after running `export DEPTHAI_DEBUG=1`? Additionally, does it work if you change pipeline type to `stereo`?

Shivam7Sharma, 2024-08-05:
> Hi, sorry for the late reply, could you output the logs after running `export DEPTHAI_DEBUG=1`? Additionally, does it work if you change pipeline type to `stereo`?

After I install the ros-humble-rclcpp-dbgsym ros-humble-rclcpp-components-dbgsym ros-humble-depthai-ros-driver-dbgsym packages: the depthai library is linked to a different OpenCV (installed from source Opencv 4.10) than the Cvbridge OpenCV (ROS 2 Opencv4.5). So it gives an error when I run the pipeline in my own ROS 2 driver. I don't want this to happen. This only happens when I install the packages you mentioned. Is there a way to avoid this so that it doesn't mess up with my code?

Serafadam, 2024-08-05:
Hi, these packages ` ros-humble-rclcpp-dbgsym ros-humble-rclcpp-components-dbgsym ros-humble-depthai-ros-driver-dbgsym` are debug packages only and are not required for the driver to work so you can uninstall them. Additionally, `depthai-core` library ideally should be linked to the same OpenCV library that ROS libraries use otherwise there might be some undefined behavior

Shivam7Sharma, 2024-08-05:
I have uninstalled those packages. I gave the depthai debug output in my previous reply:
https://github.com/luxonis/depthai-ros/issues/328#issuecomment-2201073920

Does the following mean depthai core is linked to Opencv 4.5?:
shivam157@ubuntu:~$ cd ros2_ws/
shivam157@ubuntu:~/ros2_ws$ source install/setup.bash
shivam157@ubuntu:~/ros2_ws$ python3
Python 3.10.12 (main, Jul 29 2024, 16:56:48) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import depthai
>>> import cv2
>>> depthai_version = depthai.__version__
>>> opencv_version = cv2.__version__
>>> (depthai_version, opencv_version)
('2.26.0.0', '4.5.5')
>>> 

4.5 is the Opencv version used by ROS 2.  If this is not the correct way to check, then let me know how to find the version linked to depthai Core.

Serafadam, 2024-08-06:
These are Python libraries which can differ from ROS libraries. Since it seems gdb needs to be run separately, here are steps that you can do to get better information:
```
sudo apt install xterm
```
Edit [this line](https://github.com/luxonis/depthai-ros/blob/82ce67cf91182b11b26504af3973dc312824970e/depthai_ros_driver/launch/camera.launch.py#L67)
to following `xterm -e gdb -ex run --args`
Run the driver with `use_gdb:=true`. A new console window should pop out with debug information coming from the driver. After the crash, you can type in `bt` to get more information on where exactly it crashed in the code, it would be great if you could share that information.

Shivam7Sharma, 2024-08-06:
> These are Python libraries which can differ from ROS libraries. Since it seems gdb needs to be run separately, here are steps that you can do to get better information:
>
> ```
> sudo apt install xterm [...]

I made the changes but no new window opened up. I saw following output in the terminals:

1.```
 shivam157@ubuntu:~/dai_ws_non_prod$ ros2 launch depthai_ros_driver camera.launch.py use_gdb:=true
[INFO] [launch]: All log files can be found below /home/shivam157/.ros/log/2024-08-06-11-51-32-809983-ubuntu-11805
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gdb-1]: process started with pid [11826]
[gdb-1] GNU gdb (Ubuntu 12.1-0ubuntu1~22.04.2) 12.1
[gdb-1] Copyright (C) 2022 Free Software Foundation, Inc.
[gdb-1] License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
[gdb-1] This is free software: you are free to change and redistribute it.
[gdb-1] There is NO WARRANTY, to the extent permitted by law.
[gdb-1] Type "show copying" and "show warranty" for details.
[gdb-1] This GDB was configured as "aarch64-linux-gnu".
[gdb-1] Type "show configuration" for configuration details.
[gdb-1] For bug reporting instructions, please see:
[gdb-1] <https://www.gnu.org/software/gdb/bugs/>.
[gdb-1] Find the GDB manual and other documentation resources online at:
[gdb-1]     <http://www.gnu.org/software/gdb/documentation/>.
[gdb-1] 
[gdb-1] For help, type "help".
[gdb-1] Type "apropos word" to search for commands related to "word"...
[gdb-1] Reading symbols from /opt/ros/humble/lib/rclcpp_components/component_container...
[gdb-1] (No debugging symbols found in /opt/ros/humble/lib/rclcpp_components/component_container)
[gdb-1] Starting program: /opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level info --ros-args -r __node:=oak_container -r __ns:=/
[gdb-1] [Thread debugging using libthread_db enabled]
[gdb-1] Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
[gdb-1] [New Thread 0xfffff667e8e0 (LWP 11863)]
[gdb-1] [New Thread 0xfffff5e6e8e0 (LWP 11864)]
[gdb-1] [New Thread 0xfffff55d78e0 (LWP 11865)]
[gdb-1] [New Thread 0xfffff4dc78e0 (LWP 11866)]
[gdb-1] [New Thread 0xfffff45b78e0 (LWP 11867)]
[gdb-1] [New Thread 0xfffff3da78e0 (LWP 11868)]
[gdb-1] [New Thread 0xfffff35978e0 (LWP 11869)]
[gdb-1] [New Thread 0xfffff2cc78e0 (LWP 11870)]
[gdb-1] [New Thread 0xfffff249e8e0 (LWP 11871)]
[gdb-1] [INFO] [1722959494.227844711] [oak_container]: Load Library: /opt/ros/humble/lib/librobot_state_publisher_node.so
[gdb-1] [INFO] [1722959494.349106073] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[gdb-1] [INFO] [1722959494.349302717] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<robot_state_publisher::RobotStatePublisher>
[gdb-1] [INFO] [1722959494.419471109] [oak_state_publisher]: got segment oak
[gdb-1] [INFO] [1722959494.419600616] [oak_state_publisher]: got segment oak-d-base-frame
[gdb-1] [INFO] [1722959494.419621352] [oak_state_publisher]: got segment oak_imu_frame
[gdb-1] [INFO] [1722959494.419633512] [oak_state_publisher]: got segment oak_left_camera_frame
[gdb-1] [INFO] [1722959494.419643401] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[gdb-1] [INFO] [1722959494.419652233] [oak_state_publisher]: got segment oak_model_origin
[gdb-1] [INFO] [1722959494.419660841] [oak_state_publisher]: got segment oak_rgb_camera_frame
[gdb-1] [INFO] [1722959494.419669065] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[gdb-1] [INFO] [1722959494.419677673] [oak_state_publisher]: got segment oak_right_camera_frame
[gdb-1] [INFO] [1722959494.419685322] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[gdb-1] [INFO] [1722959494.450183378] [oak_container]: Load Library: /home/shivam157/dai_ws_non_prod/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak_state_publisher' in container 'oak_container'
[gdb-1] [New Thread 0xffffc9e6e8e0 (LWP 11913)]
[gdb-1] [New Thread 0xffffc965e8e0 (LWP 11914)]
[gdb-1] [New Thread 0xffffc8e4e8e0 (LWP 11915)]
[gdb-1] [New Thread 0xffffbfffe8e0 (LWP 11916)]
[gdb-1] [New Thread 0xffffbb7ee8e0 (LWP 11917)]
[gdb-1] [INFO] [1722959499.855914297] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[gdb-1] [INFO] [1722959499.856067517] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[gdb-1] [New Thread 0xffffb8fde8e0 (LWP 11918)]
[gdb-1] [New Thread 0xffffb87ce8e0 (LWP 11919)]
[gdb-1] [New Thread 0xffffb6fbc8e0 (LWP 11920)]
[gdb-1] [INFO] [1722959499.879174652] [oak]: No ip/mxid specified, connecting to the next available device.
[gdb-1] [Thread 0xffffb87ce8e0 (LWP 11919) exited]
[gdb-1] [Thread 0xffffb8fde8e0 (LWP 11918) exited]
[gdb-1] [New Thread 0xffffb7fbe8e0 (LWP 11984)]
[... 6 lines cut ...]
[gdb-1] [INFO] [1722959504.510668847] [oak]: USB SPEED: SUPER
[gdb-1] [INFO] [1722959504.552905237] [oak]: Device type: OAK-FFC-3P
[gdb-1] [INFO] [1722959504.557032214] [oak]: Pipeline type: Depth
[gdb-1] 
[gdb-1] Thread 1 "component_conta" received signal SIGSEGV, Segmentation fault.
[gdb-1] __memcpy_generic () at ../sysdeps/aarch64/multiarch/../memcpy.S:111
[gdb-1] 111     ../sysdeps/aarch64/multiarch/../memcpy.S: No such file or directory.
```

2. Looks like related to VNC :

```
[... cut ...]
thunar-volman: Unsupported USB device type "usb".
thunar-volman: Unsupported USB device type "(null)".
thunar-volman: Unsupported USB device type "usb".
thunar-volman: Unsupported input device type "/dev/input/event10".
thunar-volman: Unsupported input device type "/dev/input/event11".
thunar-volman: Unsupported input device type "(null)".
thunar-volman: Unsupported input device type "(null)".
thunar-volman: Unsupported USB device type "usbhid".
thunar-volman: Unsupported USB device type "usbhid".

```

Serafadam, 2024-08-07:
Hi @Shivam7Sharma, did you rebuild the workspace after editing the launch file? I've found that sometimes that is needed even with `symlink-install`. You could also try adding some print statement to verify that you are launching the modified version of launch file.

Shivam7Sharma, 2024-08-07:
I didn't get this error after building and sourcing the workspace (I have tried this before, but it didn't work before). Can we please solve other issues, like:

https://discuss.luxonis.com/d/4963-low-and-drop-in-fps-and-unsynchronized/4
And 
[Poor depth on the depthAI ROS 2 driver](https://discuss.luxonis.com/d/4903-no-distortion-coefficient-and-rectification-matrix-generated-by-calibrationpy/44)