luxonis/depthai-ros #330: [BUG] depthai_ros_driver camera.launch.py crashes "double free detected in tcache 2"

FabienDanieau, 2023-06-15:
**Describe the bug**
When I ran `ros2 launch depthai_ros_driver camera.launch.py`, it crashes at a random time with the error
```
[component_container-2] free(): double free detected in tcache 2
[ERROR] [component_container-2]: process has died [pid 55444, exit code -6, cmd '/opt/ros/humble/install/lib/rclcpp_components/component_container --ros-args --log-level debug --ros-args -r __node:=oak_container -r __ns:=/'].
```

**Minimal Reproducible Example**

Run `ros2 launch depthai_ros_driver camera.launch.py` with default parameters.

(`stereo_inertial_node` in `depthai_examples` works well.)

**Attach system log**
Logs are here:
[log.txt](https://github.com/luxonis/depthai-ros/files/11759861/log.txt)
[log_system_information.json.txt](https://github.com/luxonis/depthai-ros/files/11759863/log_system_information.json.txt)

I am using a Jetson Orin Nano.

FabienDanieau, 2023-06-19:
It is built from source. I've just updated depthai-core to 2.22.0, and depthai-ros to 2.7.3
It also happens with `ros2 launch depthai_ros_driver example_marker_publish.launch.py`, maybe more quickly

Hitomamacs, 2023-06-21:
Same problem here

Serafadam, 2023-06-22:
Does it also happen with `rgbd_pcl.launch.py`?

FabienDanieau, 2023-06-22:
Looks good on this side

FabienDanieau, 2023-07-10:
Any news on this issue? I'd like to use the ROS driver with 2d object detection but it crashes too often

tkelestemur-bdai, 2023-07-10:
Same problem here -- I've also installed from source from the latest humble branch.

tkelestemur-bdai, 2023-07-11:
@Serafadam I've encountered this problem on Ubuntu 22.04 with ROS2 Humble. I'm able to run the driver on v2.7.1 but then the point cloud rates is extremely slow. It gives a cloud at every minute or so. I'll give it a try with Docker. I can send the full log if it helps.

Serafadam, 2023-07-11:
Sure, it would be nice if you provide information as in [bug template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md). When it comes to the pointcloud rate, that could depend on the network/DDS setup

BOOTCFG, 2023-07-13:
Hi, I was experiencing the same issue, both on a Ubuntu 22 laptop and a PI4. After attaching myself with a debugger, I narrowed the error to the IMU interpolation function (to be exact [line 182 of ImuConverter.hpp](https://github.com/luxonis/depthai-ros/blob/6693cfec253d629960aae1940de0361e352327b1/depthai_bridge/include/depthai_bridge/ImuConverter.hpp#L182C21-L182C49)).

The workaround I found is to set the sync method to `COPY`. Here is my pcl.yaml:
```
/oak:
  ros__parameters:
    camera:
      i_nn_type: none
    right:
      i_publish_topic: true
    stereo:
      i_align_depth: false
      i_subpixel: true
    imu:
      i_sync_method: COPY
```

Serafadam, 2023-07-13:
Although I'm not sure if this is the cause as stereo_inertial_method uses same sync method, could you try it out @FabienDanieau  @tkelestemur-bdai ?

FabienDanieau, 2023-07-14:
Indeed I do not have the error with `i_sync_method: COPY`

ksharma-bdai, 2023-07-26:
@Serafadam +1, this workaround resolves the issue for me as well (same lab as @tkelestemur-bdai)

@BOOTCFG nice detective work, thanks!

Serafadam, 2023-07-28:
Hi @FabienDanieau, @ksharma-bdai could you check if this branch works for you?

ksharma-bdai, 2023-07-31:
> Hi @FabienDanieau, @ksharma-bdai could you check if this branch works for you?

I'll try to give it a test today or tomorrow (otherwise I'll be away until next week). Assuming the branch is: https://github.com/luxonis/depthai-ros/tree/imu_sync_fix_humble ?

Serafadam, 2023-07-31:
Yes, sorry, for some reason the link didn't display properly

ksharma-bdai, 2023-07-31:
> Yes, sorry, for some reason the link didn't display properly

Unfortunately it's still crashing when built from source (verified it's running the branch with a few printouts). Happy to run with any debug that could be useful. This was tested with the OAK-D Pro Wide USB.

Our devices will be under test and I'll be away starting Wednesday, but can resume testing next week.

Serafadam, 2023-08-01:
I've pushed some additional changes

ksharma-bdai, 2023-08-01:
> I've pushed some additional changes

No crashes after ~10 minutes on the updated branch with the same OAK-D Pro Wide USB camera (crashes within ~1 minute on apt version). I'm currently only able to verify it does not crash and rgb/depth streams to rviz, I don't have a known good regression test suite - so it would be nice if anyone else can confirm,

Thanks for the updates and let me know if there's anything else I can help with (will be out of contact the rest of this week after today).

asimonov, 2023-08-01:
I am also seeing the issue. Just to add - it is on Ethernet version of OAK-D Pro W.

asimonov, 2023-08-02:
> @asimonov you can try the fix for noetic [here](https://github.com/luxonis/depthai-ros/tree/imu_sync_fix_noetic)

yes, this works without crashes for me. 
ubuntu 20 inside docker on jetson orin NX with jetpack 5.1.1.
Oak-D Pro W Ethernet version.

FabienDanieau, 2023-08-14:
> Hi @FabienDanieau, @ksharma-bdai could you check if this branch works for you?
Sorry I was away. I'll test it asap.

Serafadam, 2023-09-11:
Closing due to inactivity, please reopen if the issues still appear.

closed 2023-09-11 by Serafadam