luxonis/depthai-ros #331: [QQ] Oak D Lite IMU to base camera frame transform ?

hidmic, 2023-06-15:
I have an Oak D Lite in my hands (awesome device BTW!), and this one has an IMU I'm trying to use. However, I have the impression its orientation does not match [that of the regular Oak D](https://github.com/luxonis/depthai-ros/blob/c77c4e9f9d13ce80f0826103cbd792ff5003f57d/depthai_descriptions/urdf/include/depthai_macro.urdf.xacro#L17-L25). Instead of a +90 degree pitch, I see -90 degree yaw followed by -90 degree roll (intrinsic rotations). Almost as if it was mounted on the other side of the board and rotated. 

Am I missing something here?

asimonov, 2023-09-19:
@saching13 

I am using urdf for Luxonis OAK-D Pro with my Luxonis Oak-D pro W device. 

I am also running OpenVINS with it. OpenVINS reports position/rotation of the IMU in the world frame.

It looks like TF coming from URDF is not matching what OpenVINS reports. Is this correct? Can you tell how I can know orientation of IMU inside my Luxonis Oak-D Pro W without opening it? Is it in documentation somewhere?

saching13, 2023-09-19:
We will have an update on this within next 2 weeks.
For now to get the correct orientation. when the node is publishing the IMU values. Plot them in plotjuggler or foxglove to see which axis gets value when placed on one of the sides of the device. do this for 3 sides to know which one represents which axises

josephduchesne, 2024-11-19:
Any update on this? The xacro file with OAK-D-LITE selected still has no IMU transform.

Serafadam, 2024-11-25:
Hi, currently we are working on next version of depthai-core (V3), which will allow publishing of all IMU transforms straight from the camera's firmware, although for now it's hard to give an estimate on when it will be released for ROS

Interpause, 2025-02-05:
seems to be root cause for https://github.com/luxonis/depthai-ros/issues/128#issuecomment-1270523739 as well

Jakubach, 2025-03-01:
@Serafadam Is there any solution for that? I would like to run Oak-D-Lite camera with ROS2. It's publishing imu data wrt. "oak_imu_frame" which does not exist in tf tree and in the camera's URDF.