luxonis/depthai-ros #333: [BUG] SpatialYolo output seems to have the Y coordinate inverted

RemiFabre, 2023-06-18:
Hi,
Could you please confirm what is the frame in wich the output of SpatialYolo is supposed to be?
In my understanding, the Y coordinate is inverted. This is how we fixed it for our pipeline, but it's likely that this is not the correct place to fix this (it probably comes from before that point):
https://github.com/pollen-robotics/depthai-ros/commit/f4ff935f8c3a7f925a02e389aa44c833e7e370c0

Apologies if this is the intended behaviour.
Best,

saching13, 2023-06-19:
Yes.  It was a mistake in the fw that went unnoticed. We are adding a fix in the fw on the next release. 
Apologize for the confusion.

saching13, 2023-06-19:
@RemiFabre  I would also be interested in knowing your use-case.