luxonis/depthai-ros #334: Launch File for OAK-1 Lite W - ROS Noetic

David9696, 2023-06-20:
Hi,

I just received my OAK-1 Lite W camera.

May I know which launch file should I use to have the yolo detection result being published in ROS noetic?

Thanks.

Regards,

David

sagarvl96, 2023-08-10:
@Serafadam what about yolov4_spatial_node? Does it have a similar implementation as stereo_inertial_node?

Serafadam, 2023-08-11:
Hi, `yolov4_publisher` is pretty similar to `stereo_inertial_node`, although you should be able to run custom Yolo models on OAK-1 with `depthai_ros_driver`, if not, please let us know.

sagarvl96, 2023-08-11:
> Hi, `yolov4_publisher` is pretty similar to `stereo_inertial_node`, although you should be able to run custom Yolo models on OAK-1 with `depthai_ros_driver`, if not, please let us know.

Thank you for the quick response. Should I just change the path to blob file in the `i_nn_config_path:` and launch the camera driver to run the custom yolo model? Or is there some other way?

Serafadam, 2023-08-16:
Hi, you should edit or create new NN config file ([here is the example](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/nn/yolo.json) one that can be used), `i_nn_config_path` is the path to that json (currently only absolute paths are supported).

closed 2023-09-11 by Serafadam