luxonis/depthai-ros #348: [custom] precision depth

FabienDanieau, 2023-07-10:
Hi,

I am using a charuco board to double check the 3D position of the point cloud. I notice an offset in the z direction, using `ros2 launch depthai_examples stereo_inertial_node.launch.py enableDotProjector:=true dotProjectormA:=800.0` or `ros2 launch depthai_ros_driver rgbd_pcl.launch.py`. x and y looks well aligned.
Is there a parameter to tune to adjust the precision of the depth map? Thanks

[image]
[image]

Serafadam, 2023-07-13:
Hi, could you record a rosbag that we could investigate?

FabienDanieau, 2023-07-14:
Sure sent by email

FabienDanieau, 2023-07-14:
Note that my board has a [legacy charuco pattern ](https://github.com/opencv/opencv/pull/23153/files) so you need Opencv 4.8 with setlegacypattern(true) or Opencv <=4.5.

saching13, 2023-08-03:
@FabienDanieau  the file has stereo/points topic which is showing up empty on my side. can you send more data?

And also how was the charuco grid added and how did you try to align so that I can try the same experiment on my side. 
Do you have some [RME](https://docs.luxonis.com/en/latest/pages/support/#creating-minimal-reproducible-example) ?

[image]

saching13, 2023-10-05:
Closing due to inactivity. Feel free to open it again if there is more data available.

closed 2023-10-05 by saching13