luxonis/depthai-ros #349: [custom] Ability to publish RGB from multiple OAK-D PoE cameras at the same time

ahmadchalhoub, 2023-07-11:
Hello,
I am currently using `camera.launch.py` from the _depthai_ros_driver_ package to publish RGB frames from an OAK-D PoE camera as a ROS2 topic. From my understanding, this currently only supports publishing data from a single camera. Is there any configuration that I can easily do to allow publishing data from multiple OAK-D PoE cameras at the same time, or would I need to implement it myself?

Would really appreciate the help with this!

Thanks,
Ahmad

Serafadam, 2023-07-12:
Hi, to launch multiple cameras you can refer to [multicam launch](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/launch/example_multicam.launch.py), you'll need to provide a config file with specific camera IPs.

ymei, 2023-08-01:
I have the same question.  I have two PoE cams with IP `192.168.2.1` and `192.168.2.2`.  Where should these IPs go?  In the code the comment says mx_ids
```
# put mx_ids here
    cams = ["oak_d_w", "oak_d_lite"]
```
but obviously mx_ids cannot be put there.

I just want to have raw left&right b/w images and imu values from both cameras.

Thanks.

ymei, 2023-08-01:
I see in `multicam_example.yaml` there's the `i_mx_id` field but not the `ip` (or camera name?) field.  Where should the IP go exactly?

Serafadam, 2023-08-01:
You can set `i_ip` parameter to the desired IP

closed 2023-09-11 by Serafadam