luxonis/depthai-ros #350: Pointcloud artifacts

martinlucan, 2023-07-13:
Hi,
I try my best to use OAK-D Pro in conjunction with RTABmap to create a map of interior space for further robot localization. I have spent times tuning camera parameters, although I still cannot get rid of random point cloud artifacts (as seen in the picture close to the camera), that spoil the cloud map. I have tried various filters combination, as well as extended disparity and subpixel mode. Is there any other advice on this issue?

[image]

Serafadam, 2023-07-13:
hi, for limiting the depth range you can use depth threshold filter. Could you also share a rosbag with your image data?

martinlucan, 2024-06-26:
Hi @Serafadam , how I am supposed to use the depth threshold filter in Noetic driver? I checked a complete [list](https://docs.luxonis.com/software/ros/depthai-ros/driver/#List%20of%20parameters) parameters, there is `i_enable_threshold_filter `param, but I don't see any param that actually sets the threshold value. Besides that, is there any documentation that describes all the parameters available?

Serafadam, 2024-07-02:
Hi, sorry, those are missing indeed and will be fixed in docs, you can check the parameter names and values [here](https://github.com/luxonis/depthai-ros/blob/a15039cf468818a7451ce7e9f921c73037be8c3e/depthai_ros_driver/src/param_handlers/stereo_param_handler.cpp#L152)