luxonis/depthai-ros #353: [BUG] {Compilation of package with ROS Noetic on RPI4 make it crashed  }

ChengTszYin, 2023-07-22:
I tried to install the package from source on my RPI 4B which installed with ROS Noetic. I compile the package with `catkin_make` but it crashed everytimes I tried. Are there any solution available on my case? Thank you
[image]

Serafadam, 2023-07-24:
Hi, to limit resource usage during compilation you can execute `MAKEFLAGS="-j1 -l1" catkin_make_isolated`, you can also temporarily increase [RAM with swap](https://pimylifeup.com/raspberry-pi-swap-file/)

ChengTszYin, 2023-07-25:
Thank, I successfully built it. It is my fault.

closed 2023-07-25 by ChengTszYin