luxonis/depthai-ros #354: [custom] Provide a general summary of the issue

osama-z-salah, 2023-07-24:
I am using the OAK-D Pro for my AMR (Autonomous Mobile Robot). However, I have two state publishers, namely /oak_state_publisher and /robot_state_publisher. I have made their parent frames the same to connect both trees. However, I am facing an issue with time stamp synchronization, as the tf (transform) of each one is running under different time stamps.

Serafadam, 2023-07-25:
Hi, attaching camera to the robot should be working without problems, could you share more information? 
- What's the time difference?
- Is camera driver running on the same host?
- How did you "attach" those together?

osama-z-salah, 2023-07-25:
[image]
yes the camera driver is running on the same host
and I attach them by using the same parent_frame for both "base_link"
parent_frame = LaunchConfiguration('parent_frame',  default = 'base_link')
    cam_pos_x    = LaunchConfiguration('cam_pos_x',     default = '-0.4815')
    cam_pos_y    = LaunchConfiguration('cam_pos_y',     default = '0.0')
    cam_pos_z    = LaunchConfiguration('cam_pos_z',     default = '0.26')
    cam_roll     = LaunchConfiguration('cam_roll',      default = '0.o')
    cam_pitch    = LaunchConfiguration('cam_pitch',     default = '0.0')
    cam_yaw      = LaunchConfiguration('cam_yaw',       default = '3.14159')

Serafadam, 2023-08-02:
@osama-z-salah This second transform is from detected marker, not the camera, could you show the timestamp between robot base frame and oak base frame?

closed 2023-09-11 by Serafadam