luxonis/depthai-ros #355: ERROR: the following packages/stacks could not hace their rosdep keys resolved to system dependencies

bb6698, 2023-07-27:
Hi! I have installed ros2 distribution humble since installation source in the Nvidia Jetson AGX Orin. I want install depthai-ros. 

I try rosdep install --from-paths src --ignore-src -r -y following instructions install from source package:depthai-ros. 

The problem is: 
[image]

I try also rosdep install --from-paths src --ignore-src -r --rosdristro humble but I have same error. 

Please, Can anybody help me? 

Cheers.

Serafadam, 2023-07-27:
Hi, default ROS distro for focal is Foxy (which is end-of-life), so even if you install Humble from source you won't be able to access humble packages via rosdep. Easiest way to get that system is to use Docker images https://github.com/dusty-nv/jetson-containers

bb6698, 2023-07-28:
Thanks for your answer!! I'll follow your advice :)

Serafadam, 2023-08-16:
Closing, please reopen if needed.