luxonis/depthai-ros #356: How to use depthai API in ROS for spatial (distance) detection

hangcui1201, 2023-08-01:
Hi,

A quick question that how can I use the depthai API in ROS for spatial detection? 

I lunched "roslaunch depthai_examples stereo_node.launch". I have left, right and depth images. I also got the python demo code such as spatial_location_calculator.py work. In spatial_location_calculator.py, it uses the depthai to setup the device and do the spatial detection. 

In my case, when I launch "roslaunch depthai_examples stereo_node.launch", the parameters of the camera has been set up. How can I use the depthai API to do spatial detection from the published images, such as /stereo_publisher/left/image, /stereo_publisher/right/image and /stereo_publisher/stereo/depth?

I am using oak-d-lite.

Thank you very much for the help!

Best,
Hang

Serafadam, 2023-08-02:
Hi, you can refer either to [stereo_inertial_node](https://github.com/luxonis/depthai-ros/blob/humble/depthai_examples/src/stereo_inertial_publisher.cpp) to see how to set up custom pipeline, or you can use depthai_ros_driver node which also has detections enabled by default (though those two nodes publish detections in different formats).

saching13, 2023-08-03:
depthai-ros uses depthai api inside it. If you have a python code you want to convert to ros. Just add ros publishers to it. Or use of our nodes in ros driver which does the same. They can be configured with yaml here.

closed 2023-09-11 by Serafadam