luxonis/depthai-ros #361: Error: Depth resolution (1280x720) does not match RGB resolution (1280x720)

julianhund, 2023-08-04:
When I run "ros2 launch depthai_ros_driver rgbd_pcl.launch.py" I got the following Error:
[component_container-2] [ERROR] [1691152826.505330441] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)

Log-File:
1691152806.7447321 [component_container-2] [INFO] [1691152806.744490674] [oak]: Device type: OAK-D-PRO-W-97
1691152806.7471197 [component_container-2] [INFO] [1691152806.746911994] [oak]: Pipeline type: RGBD
1691152806.9809377 [component_container-2] [INFO] [1691152806.980735986] [oak]: Finished setting up pipeline.
1691152807.0401533 [component_container-2] [18443010E1EE870E00] [1.4.2] [0.927] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
1691152807.2883236 [component_container-2] [INFO] [1691152807.287414685] [oak]: Camera ready!
1691152807.2921410 [component_container-2] [INFO] [1691152807.289150476] [oak_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
1691152807.2943358 [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
1691152807.3126044 [component_container-2] [INFO] [1691152807.311861453] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
1691152807.3132336 [component_container-2] [INFO] [1691152807.312008315] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
1691152807.3135865 [component_container-2] [INFO] [1691152807.312036711] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
1691152807.3138990 [component_container-2] [INFO] [1691152807.312058297] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
1691152807.3144405 [component_container-2] [INFO] [1691152807.312077833] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
1691152807.3153996 [component_container-2] [INFO] [1691152807.312096953] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
1691152807.3159177 [component_container-2] [INFO] [1691152807.312116834] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziRadialNode>
1691152807.3161850 [component_container-2] [INFO] [1691152807.312136389] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
1691152807.3164363 [component_container-2] [INFO] [1691152807.312155862] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
1691152807.3265450 [INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzrgb_node' in container 'oak_container'
1691152807.4546072 [component_container-2] [ERROR] [1691152807.454301139] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
1691152807.4709277 [component_container-2] [ERROR] [1691152807.470650526] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
1691152807.5033290 [component_container-2] [ERROR] [1691152807.503056358] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
1691152807.5377488 [component_container-2] [ERROR] [1691152807.537376189] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
1691152807.5870776 [component_container-2] [ERROR] [1691152807.586230310] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
1691152807.6155798 [component_container-2] [ERROR] [1691152807.615002322] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
1691152807.6493702 [component_container-2] [ERROR] [1691152807.648775254] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
1691152807.6808860 [component_container-2] [ERROR] [1691152807.680299402] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)

Serafadam, 2023-08-04:
Hi, well this is a weird one :sweat_smile:  could you show some more logs? For example topic data ( from rostopic echo /rqt, but rosbag would be preferable)

julianhund, 2023-08-07:
Hello, 

I cant upload the bag it is to big with a very small recording. 
With this link you can download it:

https://seafile.rlp.net/d/a662c2e48be0474ba397/

Thank you!

This time I get first the Error:
[ERROR] [1691388172.414031802] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).

Log-Output:
[INFO] [robot_state_publisher-1]: process started with pid [11813]
[INFO] [component_container-2]: process started with pid [11815]
[robot_state_publisher-1] [INFO] [1691388166.255118115] [oak_state_publisher]: got segment oak
[robot_state_publisher-1] [INFO] [1691388166.257084027] [oak_state_publisher]: got segment oak-d-base-frame
[robot_state_publisher-1] [INFO] [1691388166.257133540] [oak_state_publisher]: got segment oak_imu_frame
[robot_state_publisher-1] [INFO] [1691388166.257170280] [oak_state_publisher]: got segment oak_left_camera_frame
[robot_state_publisher-1] [INFO] [1691388166.257206315] [oak_state_publisher]: got segment oak_left_camera_optical_frame
[robot_state_publisher-1] [INFO] [1691388166.257245993] [oak_state_publisher]: got segment oak_model_origin
[robot_state_publisher-1] [INFO] [1691388166.257282240] [oak_state_publisher]: got segment oak_rgb_camera_frame
[robot_state_publisher-1] [INFO] [1691388166.257317294] [oak_state_publisher]: got segment oak_rgb_camera_optical_frame
[robot_state_publisher-1] [INFO] [1691388166.257352264] [oak_state_publisher]: got segment oak_right_camera_frame
[robot_state_publisher-1] [INFO] [1691388166.257390961] [oak_state_publisher]: got segment oak_right_camera_optical_frame
[component_container-2] [INFO] [1691388166.474098748] [oak_container]: Load Library: /home/julian/Software/Assemblyassist/dai_ws/install/depthai_ros_driver/lib/libdepthai_ros_driver.so
[component_container-2] [INFO] [1691388166.610822918] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1691388166.610914335] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depthai_ros_driver::Camera>
[component_container-2] [INFO] [1691388166.616937910] [oak]: No ip/mxid specified, connecting to the next available device.
[component_container-2] [INFO] [1691388169.017902708] [oak]: Camera with MXID: 18443010E1EE870E00 and Name: 1.2 connected!
[component_container-2] [INFO] [1691388169.019401129] [oak]: USB SPEED: SUPER
[component_container-2] [INFO] [1691388169.041526583] [oak]: Device type: OAK-D-PRO-W-97
[component_container-2] [INFO] [1691388169.052734379] [oak]: Pipeline type: RGBD
[component_container-2] [INFO] [1691388169.293687617] [oak]: Finished setting up pipeline.
[component_container-2] [18443010E1EE870E00] [1.2] [0.995] [ColorCamera(0)] [warning] Unsupported resolution set for detected camera OV9782, needs 800_P or 720_P. Defaulting to 800_P
[component_container-2] [ERROR] [1691388169.510878102] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[component_container-2] [ERROR] [1691388169.559720050] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images (input format is: bgr8).
[component_container-2] [INFO] [1691388169.594677120] [oak]: Camera ready!
[component_container-2] [INFO] [1691388169.595373416] [oak_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-2] [INFO] [1691388169.603562116] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1691388169.604339431] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-2] [INFO] [1691388169.604803598] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-2] [INFO] [1691388169.605277601] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-2] [INFO] [1691388169.605720912] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-2] [INFO] [1691388169.606156381] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-2] [INFO] [1691388169.606577382] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziRadialNode>
[component_container-2] [INFO] [1691388169.607014840] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[component_container-2] [INFO] [1691388169.607452192] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzrgb_node' in container 'oak_container'
[component_container-2] [ERROR] [1691388169.643453055] [compressed_depth_image_transport]: Compressed Depth Image Transport - Compression requires single-channel 32bit-floating point or 16bit raw depth images  [... cut ...]
[component_container-2] [ERROR] [1691388175.872586782] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
[component_container-2] [ERROR] [1691388175.894877910] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
[component_container-2] [ERROR] [1691388175.926729590] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)
[component_container-2] [ERROR] [1691388175.960706664] [point_cloud_xyzrgb_node]: Depth resolution (1280x720) does not match RGB resolution (1280x720)

julianhund, 2023-08-07:
[image]

This is the rqt-Graph

Serafadam, 2023-08-10:
Could you try setting `i_resolution: '720'` parameter in the config?

julianhund, 2023-08-14:
Yes I setted the parameter to 720 as well as i_nn_type: none.

Now I have this error when I run rgbd_pcl.launch.py:

omponent_container-2] [INFO] [1692004697.519664985] [oak_container]: Load Library: /opt/ros/humble/lib/libdepth_image_proc.so
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/oak' in container '/oak_container'
[component_container-2] [INFO] [1692004697.530541296] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1692004697.530585684] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-2] [INFO] [1692004697.530592092] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-2] [INFO] [1692004697.530596911] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-2] [INFO] [1692004697.530601596] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-2] [INFO] [1692004697.530606267] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-2] [INFO] [1692004697.530615180] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziRadialNode>
[component_container-2] [INFO] [1692004697.530620459] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[component_container-2] [INFO] [1692004697.530625337] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzrgb_node' in container 'oak_container'
[component_container-2] terminate called after throwing an instance of 'cv::Exception'
[component_container-2]   what():  OpenCV(4.8.0-dev) ./modules/imgproc/src/resize.cpp:4051: error: (-215:Assertion failed) !ssize.empty() in function 'resize'
[component_container-2]

julianhund, 2023-08-14:
[rosbag2_2023_08_14-11_31_25.zip](https://github.com/luxonis/depthai-ros/files/12333431/rosbag2_2023_08_14-11_31_25.zip)

This is a very small recording

Serafadam, 2023-08-16:
Hi @julianhund, could you try setting `rgb: i_set_isp_scale: false` and see if it works?

julianhund, 2023-08-17:
Seems like this problem is solved, thanks.

Now I have another Error:

[component_container-2] terminate called after throwing an instance of 'cv::Exception'
[component_container-2]   what():  OpenCV(4.8.0-dev) ./modules/imgproc/src/resize.cpp:4051: error: (-215:Assertion failed) !ssize.empty() in function 'resize'
[component_container-2] 
[ERROR] [component_container-2]: process has died [pid 12617, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level info --ros-args -r __node:=oak_container -r __ns:=/'].
[rosbag2_2023_08_17-13_16_11.zip](https://github.com/luxonis/depthai-ros/files/12368650/rosbag2_2023_08_17-13_16_11.zip)

Serafadam, 2023-08-17:
Could you also share logs? To enable debug logs you can run `export DEPTHAI_DEBUG=1`

julianhund, 2023-08-18:
[launch.log](https://github.com/luxonis/depthai-ros/files/12380364/launch.log)
[robot_state_publisher_26400_1692368927058.log](https://github.com/luxonis/depthai-ros/files/12380365/robot_state_publisher_26400_1692368927058.log)
[component_container_26402_1692368927058.log](https://github.com/luxonis/depthai-ros/files/12380366/component_container_26402_1692368927058.log)
[python3_26385_1692368927008.log](https://github.com/luxonis/depthai-ros/files/12380368/python3_26385_1692368927008.log)

Serafadam, 2023-08-21:
Hi @julianhund, camera driver appears to be running normally, although there is an error related to opencv. Did you build it from source? If yes, what version?

julianhund, 2023-08-21:
Hello, 
yes i builded it from source: 
I have version 4.8.0 in C++ and 4.5.5 in Python

Serafadam, 2023-08-24:
Could you share your parameter file?

julianhund, 2023-08-25:
file:///home/julian/camera.yaml

julianhund, 2023-08-25:
.yaml Files are not supported to upload, I changed it to .txt
[camera.txt](https://github.com/luxonis/depthai-ros/files/12436976/camera.txt)

Serafadam, 2023-08-25:
Hi, just to be sure:
You are launching `rgbd_pcl.launch.py`, right? Are you passing a new parameters file, or using the default one?
This launch file has a [separate parameter file](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/rgbd.yaml), whereas the one that you uploaded looks like a [parameter file](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/camera.yaml) from [camera.launch.py](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/launch/camera.launch.py).

closed 2023-09-29 by julianhund