luxonis/depthai-ros #362: Frame rate around 20fps

Hero9555, 2023-08-04:
I used the OAK-D-W with the driver with default config parameters. Buth the frame rate of all three cameras are around 20fps. I tried to changed the Parameters in camera.yaml than launch the camera.launch but the frame rate remain the same. Any suggestions?

Serafadam, 2023-08-07:
Hi, framerate can be influenced by number of things such as resolution and bandwidth used. You can try using different USB cable (and different port) to see if things improve. It would be also nice to have some information of the system as described in [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)

Serafadam, 2023-10-16:
Closing due to inactivity, please reopen if you have further questions/remarks.

closed 2023-10-16 by Serafadam