luxonis/depthai-ros #363: [custom] Provide a general summary of the issue[depthai] [warning] Skipping X_LINK_UNBOOTED device with name "1.1" ()

ujasindro, 2023-08-04:
I am trying to Run the provided ROS drivers for OAK-D-PRO camera inside the docker container in Humble. However, when I run the launch command it throws and error as shown in the picture below.

Could you please help me to resolve this issue?

[image]

Serafadam, 2023-08-07:
Hi, you probably don't have udev rules set up on host, you can do that with 
```
echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
```
Let me know if that helps.

ujasindro, 2023-08-08:
Unfortunately No luck. Its still throwing me the same error

Serafadam, 2023-08-10:
Could you try running other examples and see if they are working for you? Alternatively, are you able to run [the demo script?](https://docs.luxonis.com/en/latest/#demo-script).
Here's some more information that could be helpful when troubleshooting:
https://docs.luxonis.com/en/latest/pages/troubleshooting/#depthai-can-t-connect-to-an-oak-camera

Bug report template to provide more information is [here](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)

ujasindro, 2023-08-10:
Yes. I am able to run the demo script. But none of the examples are working for me. I can run the example in ROS 1 but in ROS 2 it gives me the ' runtime error;

ujasindro, 2023-08-10:
I have got it working. I just forgot to add the usb rules to my docker file as well. Thank you for all the help :)

Serafadam, 2023-08-11:
Happy to help :slightly_smiling_face:

closed 2023-08-11 by Serafadam