luxonis/depthai-ros #369: Projection matrix in camera_info is not aligned with the definition in OpenCV

yanmingzou, 2023-08-14:
I launched the camera with the command ros2 launch depthai_exmaples stereo_inertial_node rectify:=false depth_aligned:=false
When I echo the /left/camera_info ROS topic, I find the projection matrix is: 
804.0103
0.0
643.925
_**60.203**_
0.0
804.
...
But according to the definition of projection matrix in OpenCV, the fourth element should be 0. (which is 60.2 currently)
And the fourth element of the right camera's projection matrix should be a negative value, but it is set as 0 now.

So, what is the exact definition of your camera frames, projection matrixes, and rotation matrixes? Are there any reasons to use a different definition from OpenCV? How can I use those parameters when my program is based on OpenCV? Do you have examples in C/C++? If I use my own calibration tools and get results based on the OpenCV definition, how can I convert the results before writing to the camera's flash memory? Thanks a lot.

Yanming

saching13, 2023-08-22:
>But according to the definition of projection matrix in OpenCV, the fourth element should be 0. (which is 60.2 currently)

Can you link the references please ?

Projection Matrixes are defined as the comments in [here](http://docs.ros.org/en/melodic/api/sensor_msgs/html/msg/CameraInfo.html)

yanmingzou, 2023-08-22:
https://docs.opencv.org/4.2.0/d9/d0c/group__calib3d.html#ga617b1685d4059c6040827800e72ad2b6

saching13, 2023-09-06:
Since our default depth output is usually rectified right frame. it was inverted to match it. But since if users want to use it externally in the default way I think we can modify it to reflect the left being thew origin. 

Will change it soon. If you need it soon feel free to change it [here](https://github.com/luxonis/depthai-ros/blob/7233130f6f3dbfee35a81f006ea5d0a902ab97fc/depthai_bridge/src/ImageConverter.cpp#L379-L388) and send a PR.

Serafadam, 2023-09-12:
Closing this, since the option is available in newest version of `depthai_ros_driver`. Please reopen if you have further comments/questions.

closed 2023-09-12 by Serafadam