luxonis/depthai-ros #370: [question] hange of distortion_model param in depthai_examples

martinlucan, 2023-08-15:
Hi there,
please, how can I change the distortion_model in `/camera_info` message? My Oak-D Pro uses `rational_polynomial` model, while in `depthai_examples/params/camera `configs, the `plumb_bob` model is used (which is desired). Even if the `/stereo_publisher/camera_param_uri `path is correctly set, the values from config files are ignored. Thank you.

Serafadam, 2023-08-16:
Hi, `stereo_publisher` example code at this moment doesn't support loading external calibration file (you would have to edit the example code to add that, or write a new node using that setup). Loading ROS calibrations from parameters is supported in `depthai_ros_driver`.

martinlucan, 2023-08-17:
Hi, @Serafadam , sure, I tried to set up a path to the calibration file. Unfortunately, it evokes the` [ERROR] [1692269031.217107656]: Invalid camera calibration URL: depthai_ros_driver/config/calibration/rgb.yaml`. I kept the original path, but it does not accept local or global paths. 
Moreover, could you please give me a hint on how to set the camera to really output the non-rectified image_raw in separate topics for left and right cameras?

Serafadam, 2023-08-17:
Hi, `CameraInfoManager` has specific path convention mentioned [here](https://github.com/luxonis/depthai-ros#recalibration), to use global path you should prepend it with `file:///`

martinlucan, 2023-08-17:
@Serafadam Thank you. Regarding non-rectified image `i_publish_topic` works reliably in ROS1 driver, although does not work in ROS2 driver. The configuration used is completely the same.

Serafadam, 2023-08-17:
Could you share your setup? I've just [edited this line](https://github.com/luxonis/depthai-ros/blob/f1c0656b67c3e6cf1476ed7245c603655a3547cf/depthai_ros_driver/config/camera.yaml#L41) and it is working in my case

martinlucan, 2023-08-17:
@Serafadam I use Foxy branch version. I rebuild the packages and changed only `i_publish_topic`, but the camera streams /stereo topics anyway.

Serafadam, 2023-08-18:
@martinlucan I would recommend switching to Humble version if possible (you can use Docker container for that) as Foxy distribution reached end of life this year and thus is not officially supported. Nevertheless, setting `publish_topic` should work just fine (tested it right now).

One caveat with Foxy is that parameters for composable nodes have a bit different format (node name doesn't have `/` sign and `ros__parameters` field is missing, for comparison [Humble parameters file](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/camera.yaml) and [Foxy parameters file](https://github.com/luxonis/depthai-ros/blob/foxy/depthai_ros_driver/config/camera.yaml).

Serafadam, 2023-08-22:
Hi, could you show the result of `ros2 topic list`?

martinlucan, 2023-08-28:
@Serafadam see the `ros2 topic list ` output, while `/camera_info` being empty. /camera_raw for left and right cameras are publishing.

```
/joint_states
/oak/imu/data
/oak/left/camera_info
/oak/left/image_raw
/oak/left/image_raw/compressed
/oak/left/image_raw/compressedDepth
/oak/left/image_raw/theora
/oak/right/camera_info
/oak/right/image_raw
/oak/right/image_raw/compressed
/oak/right/image_raw/compressedDepth
/oak/right/image_raw/theora
/parameter_events
/robot_description
/rosout
/tf
/tf_static
```

Serafadam, 2023-08-28:
Hi @martinlucan, which sensor information are you expecting on `/camera_info` topic? Generally, info topic for each sensor is published on `/camera_name/sensor_name/camera_info`, so for example `/oak/left/camera_info`

martinlucan, 2023-08-28:
@Serafadam sorry, I shorted the topic name, I definitely meant `/oak/left/camera_info`, according to the topic list.

Serafadam, 2023-08-29:
Could you share logs that you get when running the driver? Additionally, you can run `export DEPTHAI_DEBUG=1` before launching to enable more log information.

martinlucan, 2023-08-30:
@Serafadam please see the attached DEGUB output as well as config file 
[debug_log.txt](https://github.com/luxonis/depthai-ros/files/12473103/debug_log.txt)
[config.txt](https://github.com/luxonis/depthai-ros/files/12473110/config.txt)

Serafadam, 2023-08-30:
Thanks @martinlucan, I think I found a bug on latest development branch, fix will be merged today

martinlucan, 2023-08-30:
@Serafadam Thank you, looking forward to the merge. Could you please elaborate on what is the bug related to? I am just being curious.

Serafadam, 2023-08-30:
One of the newest updates introduced [lazy publishing](https://github.com/luxonis/depthai-ros/blob/4b3fe01e90dcf5af04e1539c480dac65c7513e98/depthai_ros_driver/src/dai_nodes/sensors/sensor_helpers.cpp#L119) mechanism which checks whether there are subscribers to a topic. What needs to be updated is to use image_transport::CameraPublisher `getNumSubscribers()` method to check for all types of subscribers as for now the check was only against `image` topic. In other words, until this is merged you should be able to get `camera_info` messages if you also subscribe to the image topic in some other way (such as displaying it in rviz/rqt).

martinlucan, 2023-08-30:
@Serafadam Thank you, I solved it temporarily by creating a "dummy" subscriber, to force the /camera_info topic to publish.

Serafadam, 2023-09-11:
Changes are merged, closing the issue.

closed 2023-09-11 by Serafadam