luxonis/depthai-ros #371: custom YOLO model with ROS and ROS2

saivojjala, 2023-08-15:
Which yolo versions can I use with ROS 1 and 2. What parameters would need to be altered in the yolov4_publisher.launch.py for launching a custom yolov7-tiny. Is it possible to set the arguments from a json file obtained from the blob converter like in[ this script ](https://github.com/luxonis/depthai-experiments/blob/master/gen2-yolo/device-decoding/main_api.py)

Serafadam, 2023-08-16:
Hi, could you share your edited code? From the error message it seems that there could be multiple Xout streams set up with the same name. As for loading configs, this is supported in `depthai_ros_driver`, example nodes in `depthai_examples` are just a showcase of different setups and aren't developed actively (although we accept PR's to them).

saivojjala, 2023-08-16:
Hi @Serafadam, here is the modified code. I believe I manaed to solve one error, [yolov4_spatial_node-2] what(): Streams have duplicate name 'detections' . However in doing so I created a new error: yolov4_spatial_node-2]   what():  Cannot create XLinkStream using empty stream name. 
The other error, [yolov4_spatial_node-2] terminate called after throwing an instance of 'std::invalid_argument', still persists and the process still dies. 

#include <cstdio>
#include <iostream>

#include "camera_info_manager/camera_info_manager.hpp"
#include "depthai_bridge/BridgePublisher.hpp"
#include "depthai_bridge/ImageConverter.hpp"
#include "depthai_bridge/ImgDetectionConverter.hpp"
#include "depthai_bridge/SpatialDetectionConverter.hpp"
#include "depthai_ros_msgs/msg/spatial_detection_array.hpp"
#include "rclcpp/executors.hpp"
#include "rclcpp/node.hpp"
#include "sensor_msgs/msg/image.hpp"

// Inludes common necessary includes for development using depthai library
#include "depthai/device/DataQueue.hpp"
#include "depthai/device/Device.hpp"
#include "depthai/pipeline/Pipeline.hpp"
#include "depthai/pipeline/node/ColorCamera.hpp"
#include "depthai/pipeline/node/DetectionNetwork.hpp"
#include "depthai/pipeline/node/MonoCamera.hpp"
#include "depthai/pipeline/node/SpatialDetectionNetwork.hpp"
#include "depthai/pipeline/node/StereoDepth.hpp"
#include "depthai/pipeline/node/XLinkOut.hpp"

const std::vector<std::string> label_map = {
        "abdominal_spatula", "arterial_clamp", "bipolar_cable", "bipolar_forcep", "de_bakey_pean",
    "durogrip_needle_holder", "durotip_prep_scissor", "durotip_surgery_scissor", "grain_tong",
    "grasping_forcep", "hook", "hook_cushing", "hook_stove", "intestinal_clamp", "intestinal_forcep",
    "kidney_dish", "lab_dish", "metz_scissor", "mosquito_clamp", "narrow_belly_hook", "prep_clamp",
    "probe", "scalpel_handle", "sharp_hook", "slime_line_electrode", "spike", "sponge_plier",
    "suction_needle", "surge_retract", "surgical_tweezer", "towel_clamp", "washing_plier", "wide_belly_hook"};

dai::Pipeline createPipeline(bool spatial_camera, bool syncNN, bool subpixel, std::string nnPath, int confidence, int LRchecktresh, std::string resolution) {
    dai::Pipeline pipeline;
    dai::node::MonoCamera::Properties::SensorResolution monoResolution;
    auto colorCam = pipeline.create<dai::node::ColorCamera>();
    auto camRgb = pipeline.create<dai::node::ColorCamera>();  // non spatial add in
    auto spatialDetectionNetwork = pipeline.create<dai::node::YoloSpatialDetectionNetwork>();
    auto detectionNetwork = pipeline.create<dai::node::YoloDetectionNetwork>();
    auto monoLeft = pipeline.create<dai::node::MonoCamera>();
    auto monoRight = pipeline.create<dai::node::MonoCamera>();
    auto stereo = pipeline.create<dai::node::StereoDepth>();

    // create xlink connections
    auto xoutRgb = pipeline.create<dai::node::XLinkOut>();
    auto xoutDepth = pipeline.create<dai::node::XLinkOut>();
    auto xoutNN = pipeline.create<dai::node::XLinkOut>();
    auto nnOut = pipeline.create<dai::node::XLinkOut>();  // non spatial add in

    if(spatial_camera == true) {
        xoutDepth->setStreamName("depth");
        xoutRgb->setStreamName("preview");
        xoutNN->setStreamName("detections");
        camRgb->setPreviewSize(512, 288);
        colorCam->setResolution(dai::ColorCameraProperties::SensorResolution::THE_1080_P);
        colorCam->setInterleaved(false);
        colorCam->setColorOrder(dai::ColorCameraProperties::ColorOrder::BGR);

        if(resolution == "720p") {
            monoResolution = dai::node::MonoCamera::Properties::SensorResolution::THE_720_P;
        } else if(resolution == "400p") {
            monoResolution = dai::node::MonoCamera::Properties::SensorResolution::THE_400_P;
        } else if(resolution == "800p") {
            monoResolution = dai::node::MonoCamera::Properties::SensorResolution::THE_800_P;
        } else if(resolution == "480p") {
            monoResolution = dai::node::MonoCamera::Properties::SensorResolution::THE_480_P;
        } else {
[... 209 lines cut ...]
        node,
        std::string("color/yolov4_Spatial_detections"),
        std::bind(&dai::rosBridge::SpatialDetectionConverter::toRosMsg, &detConverter, std::placeholders::_1, std::placeholders::_2),
        30);

    dai::rosBridge::ImageConverter depthConverter(tfPrefix + "_right_camera_optical_frame", true);
    auto rightCameraInfo = depthConverter.calibrationToCameraInfo(calibrationHandler, dai::CameraBoardSocket::CAM_C, width, height);
    dai::rosBridge::BridgePublisher<sensor_msgs::msg::Image, dai::ImgFrame> depthPublish(
        depthQueue,
        node,
        std::string("stereo/depth"),
        std::bind(&dai::rosBridge::ImageConverter::toRosMsg, &depthConverter, std::placeholders::_1, std::placeholders::_2),
        30,
        rightCameraInfo,
        "stereo");

    depthPublish.addPublisherCallback();

    detectionPublish.addPublisherCallback();
    rgbPublish.addPublisherCallback();  // addPublisherCallback works only when the dataqueue is non blocking.

    rclcpp::spin(node);

    return 0;
}

Serafadam, 2023-08-16:
I don't have the blob so I can't verify all of it, but I suspect that the culprit is (line 49)
`    auto nnOut = pipeline.create<dai::node::XLinkOut>();  // non spatial add in
`
you should remove it if you don't use that stream.

saivojjala, 2023-08-16:
Unfortunately no, I still get the same error after commenting the line 

[yolov4_spatial_node-2] terminate called after throwing an instance of 'std::invalid_argument'
[yolov4_spatial_node-2]   what():  Cannot create XLinkStream using empty stream name
[ERROR] [yolov4_spatial_node-2]: process has died [pid 1218757, exit code -6, cmd '/home/darvis/dai_ws/install/depthai_examples/lib/depthai_examples/yolov4_spatial_node --ros-args --params-file /tmp/launch_params_0_h4o6lu --params-file /tmp/launch_params_suj2pw41 --params-file /tmp/launch_params_vv19o0zs --params-file /tmp/launch_params_s_xszq4o --params-file /tmp/launch_params_dirfob0i --params-file /tmp/launch_params_gentv7rr --params-file /tmp/launch_params_k0g3kcly'].
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[robot_state_publisher-1] [INFO] [1692200080.425033858] [rclcpp]: signal_handler(signum=2)
[INFO] [robot_state_publisher-1]: process has finished cleanly [pid 1218755]

Serafadam, 2023-08-16:
Hi, you can try running following code, unfortunately can't test it properly (I would still recommend using depthai_ros_driver instead):

```
#include <cstdio>
#include <iostream>

#include "camera_info_manager/camera_info_manager.hpp"
#include "depthai_bridge/BridgePublisher.hpp"
#include "depthai_bridge/ImageConverter.hpp"
#include "depthai_bridge/ImgDetectionConverter.hpp"
#include "depthai_bridge/SpatialDetectionConverter.hpp"
#include "depthai_ros_msgs/msg/spatial_detection_array.hpp"
#include "rclcpp/executors.hpp"
#include "rclcpp/node.hpp"
#include "sensor_msgs/msg/image.hpp"

// Inludes common necessary includes for development using depthai library
#include "depthai/device/DataQueue.hpp"
#include "depthai/device/Device.hpp"
#include "depthai/pipeline/Pipeline.hpp"
#include "depthai/pipeline/node/ColorCamera.hpp"
#include "depthai/pipeline/node/DetectionNetwork.hpp"
#include "depthai/pipeline/node/MonoCamera.hpp"
#include "depthai/pipeline/node/SpatialDetectionNetwork.hpp"
#include "depthai/pipeline/node/StereoDepth.hpp"
#include "depthai/pipeline/node/XLinkOut.hpp"

const std::vector<std::string> label_map = {"abdominal_spatula",
                                            "arterial_clamp",
                                            "bipolar_cable",
                                            "bipolar_forcep",
                                            "de_bakey_pean",
                                            "durogrip_needle_holder",
                                            "durotip_prep_scissor",
                                            "durotip_surgery_scissor",
                                            "grain_tong",
                                            "grasping_forcep",
                                            "hook",
                                            "hook_cushing",
                                            "hook_stove",
                                            "intestinal_clamp",
                                            "intestinal_forcep",
                                            "kidney_dish",
                                            "lab_dish",
                                            "metz_scissor",
                                            "mosquito_clamp",
                                            "narrow_belly_hook",
                                            "prep_clamp",
                                            "probe",
                                            "scalpel_handle",
                                            "sharp_hook",
                                            "slime_line_electrode",
                                            "spike",
                                            "sponge_plier",
                                            "suction_needle",
                                            "surge_retract",
                                            "surgical_tweezer",
                                            "towel_clamp",
                                            "washing_plier",
                                            "wide_belly_hook"};

dai::Pipeline createPipeline(bool spatial_camera, bool syncNN, bool subpixel, std::string nnPath, int confidence, int LRchecktresh, std::string resolution) {
    dai::Pipeline pipeline;
    dai::node::MonoCamera::Properties::SensorResolution monoResolution;
    auto camRgb = pipeline.create<dai::node::ColorCamera>();

    auto monoLeft = pipeline.create<dai::node::MonoCamera>();
    auto monoRight = pipeline.create<dai::node::MonoCamera>();

    // create xlink connections
[... 209 lines cut ...]
            30,
            rightCameraInfo,
            "stereo");

        depthPublish.addPublisherCallback();

        detectionPublish.addPublisherCallback();
    } else {
        dai::rosBridge::ImgDetectionConverter detConverter(tfPrefix + "_rgb_camera_optical_frame", 512, 288, false, false);
        dai::rosBridge::BridgePublisher<vision_msgs::msg::Detection2DArray, dai::ImgDetections> detectionPublish(
            detectionQueue,
            node,
            std::string("color/yolov4_detections"),
            std::bind(&dai::rosBridge::ImgDetectionConverter::toRosMsg, &detConverter, std::placeholders::_1, std::placeholders::_2),
            30);
        detectionPublish.addPublisherCallback();
    }
    rgbPublish.addPublisherCallback();  // addPublisherCallback works only when the dataqueue is non blocking.

    rclcpp::spin(node);

    return 0;
}

```

saivojjala, 2023-08-16:
@Serafadam This code runs, however not perfectly. There is still an error 
[yolov4_spatial_node-2] [2023-08-16 20:20:04.448] [depthai] [error] Callback with id: 1 throwed an exception: Invalid topic name: topic name must not contain characters other than alphanumerics, '_', '~', '{', or '}':
[yolov4_spatial_node-2]   '�w���U'

How can I use the depthai_ros_driver with YOLO?

Serafadam, 2023-08-17:
This seems like a ROS-specific warning, did you change topic names in BridgePublishers?

Regarding `depthai_ros_driver`:
- Network configuration is based on a json file ([here is the default yolov4](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/nn/yolo.json) one).
- If you want to use a custom model, set "zoo": "path" and "model_name": "PATH_TO_MODEL"
- Edit the rest of the parameters accordingly
- In the driver parameters you pass the path to the JSON config in `nn: i_nn_config_path`
- By [default](https://github.com/luxonis/depthai-ros/blob/f1c0656b67c3e6cf1476ed7245c603655a3547cf/depthai_ros_driver/config/camera.yaml#L50) it's set to use mobilenet [config file](https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/nn/mobilenet.json)
- Set `camera: i_nn_type:` to either `spatial` or `rgb`
- If you want to see bounding boxes, you can also use one of the filters in `depthai_filters`. [Here](https://github.com/luxonis/depthai-ros/blob/humble/depthai_filters/launch/example_det2d_overlay.launch.py) is an example for 2D overlays, next to it you can find an example for publishing bounding boxes in 3D. Please note that By default these are set for Mobilenet, to use Yolo, you need to edit [their config files](https://github.com/luxonis/depthai-ros/blob/humble/depthai_filters/config/detection.yaml). Since detection results are published without id labels, you also need to specify label map as a parameter to the overlay node.

Serafadam, 2023-08-18:
Hi @sagarvl96, what changes did you implement in the `stereo_inertial_publisher` node?

saivojjala, 2023-08-18:
Hi @Serafadam I didn't change the names of the topics.

sagarvl96, 2023-08-18:
> Hi @sagarvl96, what changes did you implement in the `stereo_inertial_publisher` node?

        ```
   // yolo specific parameters
            spatialDetectionNetwork->setNumClasses(1);
            spatialDetectionNetwork->setCoordinateSize(4);
            spatialDetectionNetwork->setAnchors({10, 13, 16, 30, 33, 23, 30, 61, 62, 45, 59, 119, 116, 90, 156, 198, 373, 326});
            spatialDetectionNetwork->setAnchorMasks({{"side80", {0, 1, 2}}, {"side40", {3, 4, 5}}, {"side20", {6, 7, 8}}});
            spatialDetectionNetwork->setIouThreshold(0.5f);
```
Changed only these @Serafadam

sagarvl96, 2023-08-18:
> Hi @Serafadam I didn't change the names of the topics. Also, I made all the changes to the config as you mentioned. But when I launch camera.launch.py it shows NN Family as mobilenet.

I think you haven't given the path to the json file properly. That's why it still takes the default value,i.e., mobilenet. Maybe you can change log_level to debug and see if `nn: i_nn_config_path` is set properly or not!

saivojjala, 2023-08-18:
@sagarvl96 thanks, but it was actually something to do with the way i named the file. In the camera.yaml file, where can I change the input image size. Terminal shows the input is 512x512, but this value is nowhere in the config. My model requires 512x288

Serafadam, 2023-08-21:
@sagarvl96 

> Changed only these @Serafadam

Could you try running the same model with bare C++/Python DepthAI libraries? [Here is the example code for Python](https://github.com/luxonis/depthai-python/blob/main/examples/SpatialDetection/spatial_tiny_yolo.py) and here [is the one for C++](https://github.com/luxonis/depthai-core/blob/main/examples/SpatialDetection/spatial_tiny_yolo.cpp).

@saivojjala 
Until now, same size has been used for preview width/height (I will add setting those separately in next update), although NN node should resize input image according to model inputs, is there an error in your case? Or are the results not correct? Could you share some logs?

saivojjala, 2023-08-21:
@Serafadam the NN node does rezise input, however width and height are both set to the same. For example, my model takes 512x288 but the resize is done to 512x512. I retrained my model to work with 512x512 now, however the driver doesn't seem to be working.  I get the following error: 

[component_container-2] [18443010A1AB1BF500] [3.5] [1692606601.797] [host] [warning] Monitor thread (device: 18443010A1AB1BF500 [3.5]) - ping was missed, closing the device connection
F: [global] [    603797] [EventRead00Thr] usbPlatformRead:999	Cannot find file descriptor by key: 55

I checked for the topics, and they are showing, but nothing is being published to them. When I try visualizing the camera feed on rviz I get the error:

[INFO] [1692606512.607990427] [rviz2]: Stereo is NOT SUPPORTED

I actually havent till now been able to visualize the camera feed at all while using the driver. For example, even though in the camera config I have set i_output_isp as false, I cant see the video output. 

Also, although  I am currently working with ROS2, I intend to switch back to ROS1, as not all the packages I require are on ROS1. Does the depthai_ros_driver work the same in both distros are would it be completely different.

Serafadam, 2023-08-22:
@saivojjala are you also unable to run the "default" launch files (such as camera.launch.py or rgbd_pcl.launch.py)?

> the NN node does resize input, however width and height are both set to the same. For example, my model takes 512x288 but the resize is done to 512x512.

This is updated on the newest version of Humble, and will be available in the next release

> [INFO] [1692606512.607990427] [rviz2]: Stereo is NOT SUPPORTED

This is just a minor info message from rviz.

> I actually havent till now been able to visualize the camera feed at all while using the driver. For example, even though in the camera config I have set i_output_isp as false, I cant see the video output.

Could you share your parameters?

> Also, although I am currently working with ROS2, I intend to switch back to ROS1, as not all the packages I require are on ROS1. Does the depthai_ros_driver work the same in both distros or would it be completely different.

Noetic distribution uses a bit different convention for ROS parameters, and is updated regularly, although new features are added to Humble first.

saivojjala, 2023-08-22:
@Serafadam No I am not able to run the default files. The launch file shows no errors on terminal, but I dont see any output video GUI as given in the readme. When I try to view the camera feed on rviz also I encounter issues (attached in the below image)

[image]

The parameters in camera.yaml are set aaas follows: 

/oak:
  ros__parameters:
    camera:
      i_enable_imu: true
      i_enable_ir: true
      i_floodlight_brightness: 0
      i_ip: ''
      i_laser_dot_brightness: 800
      i_mx_id: ''
      i_nn_type: spatial
      i_pipeline_type: RGBD
      i_usb_port_id: ''
      i_usb_speed: SUPER_PLUS
      i_pipeline_dump: false
      i_calibration_dump: false
      i_external_calibration_path: ''
    imu:
      i_acc_freq: 400
      i_acc_cov: 0.0
      i_batch_report_threshold: 1
      i_get_base_device_timestamp: false
      i_enable_rotation: false
      i_gyro_cov: 0.0
      i_gyro_freq: 400
      i_mag_cov: 0.0
      i_mag_freq: 100
      i_max_batch_reports: 5
      i_message_type: IMU
      i_rot_cov: -1.0
      i_rot_freq: 400
      i_sync_method: LINEAR_INTERPOLATE_ACCEL
    left:
      i_board_socket_id: 1
      i_calibration_file: ''
      i_get_base_device_timestamp: false
      i_fps: 30.0
      i_height: 720
      i_low_bandwidth: false
      i_low_bandwidth_quality: 50
      i_max_q_size: 30
      i_publish_topic: false
      i_simulate_from_topic: false
      i_disable_node: false
      i_resolution: '720'
      i_width: 1280
      r_exposure: 1000
      r_iso: 800
      r_set_man_exposure: false
    nn:
      i_nn_config_path: depthai_ros_driver/yolo
      i_disable_resize: false
      i_enable_passthrough: false
      i_enable_passthrough_depth: false
    rgb:
      i_board_socket_id: 0
      i_simulate_from_topic: false
      i_get_base_device_timestamp: false
      i_disable_node: false
      i_calibration_file: ''
      i_enable_preview: false
      i_fps: 30.0
      i_height: 720
      i_interleaved: false
      i_keep_preview_aspect_ratio: true
      i_low_bandwidth: false
      i_low_bandwidth_quality: 50
      i_max_q_size: 30
      i_preview_size: 300
      i_publish_topic: true
      i_resolution: '1080'
      i_set_isp_scale: true
      i_isp_num: 2
      i_isp_den: 3
      i_output_isp: false
      i_width: 1280
      r_exposure: 20000
      r_focus: 1
      r_iso: 800
      r_set_man_exposure: false
      r_set_man_focus: false
      r_set_man_whitebalance: false
      r_whitebalance: 3300
    right:
      i_board_socket_id: 2
      i_calibration_file: ''
      i_get_base_device_timestamp: false
      i_fps: 30.0
      i_height: 720
      i_low_bandwidth: false
      i_low_bandwidth_quality: 50
      i_max_q_size: 30
      i_publish_topic: false
      i_simulate_from_topic: false
      i_disable_node: false
      i_resolution: '720'
      i_width: 1280
      r_exposure: 1000
      r_iso: 800
      r_set_man_exposure: false
    stereo:
      i_align_depth: true
      i_get_base_device_timestamp: false
      i_output_disparity: false
      i_bilateral_sigma: 0
      i_board_socket_id: 0
      i_depth_filter_size: 5
      i_depth_preset: HIGH_ACCURACY
      i_disparity_width: DISPARITY_96
      i_enable_companding: false
      i_enable_decimation_filter: false
      i_enable_distortion_correction: false
      i_enable_spatial_filter: false
      i_enable_speckle_filter: false
      i_enable_temporal_filter: false
      i_enable_threshold_filter: false
      i_enable_brightness_filter: false
      i_brightness_filter_min_brightness: 1
      i_brightness_filter_max_brightness: 256
      i_extended_disp: false
      i_height: 720
      i_low_bandwidth: false
      i_low_bandwidth_quality: 50
      i_lr_check: true
      i_lrc_threshold: 10
      i_max_q_size: 30
      i_rectify_edge_fill_color: 0
      i_stereo_conf_threshold: 240
      i_set_input_size: false
      i_input_width: 1280
      i_input_height: 720
      i_width: 1280
    use_sim_time: false
    
    
  Does YOLO detection + spatial work on noetic?

saivojjala, 2023-08-23:
Hey @Serafadam, this is an extension to yesterday's message. I can see the output video from the oak/rgb/image_raw topics on the rqt image viewer, and I can see some detection on the oak/nn/spatial_detections topic as well. Is there a way I can visualize the bounding boxes?

Serafadam, 2023-08-24:
Hi, for visualization you can use SpatialBB filter from `depthai_filters`, you can see the launch file [here](https://github.com/luxonis/depthai-ros/blob/humble/depthai_filters/launch/spatial_bb.launch.py), one change that you would need to make is to set correct `label_map` parameter for the overlay node.

saivojjala, 2023-08-25:
Okay, I believe I made all the required changes: changed the default model from mobilenet to yolo, added the label map to the spatial_bb.cpp node, and a few minor parameters like size and cam input. Although, according to the spatial_bb.cpp node the final image is published to the /overlay topic, but I see nothing on that topic, not even the camera feed. Which topic is the final video output visible on. I can see some detection on the  oak/nn/spatial_detection topic

Serafadam, 2023-08-25:
Hi, if nothing is being output on the `/overlay` topic, then it's probably due to the node not receiving messages on subscribed topics - they either have different names (most probable) or different timestamps. You can check the topic graph using `rqt_graph` tool or by using `ros2 node info spatial_bb_node` or `ros2 topic info -v TOPIC_NAME` on selected topic

saivojjala, 2023-08-25:
That seems about right, ros2 node info spatial_bb_node shows no node found, which is weird because on launching the spatial_bb.launch.py I get the message "Loaded node '/spatial_bb_node' in container 'oak_container'". Also when I run the model using the API it works. 

There is an error regarding calibration, could that be why: 
No calibration! Publishing empty camera_info.

or maybe its the way I added the labels, I added it as an array directly to the cpp code, as  I noticed it was later referenced by index. Should I have used a map?

Serafadam, 2023-08-28:
@saivojjala Could you paste your logs or record a rosbag and post it here?

saivojjala, 2023-08-29:
Hi @Serafadam , the below drive link contains the bag file. 
https://drive.google.com/drive/folders/1E4QjsKY0VNsL5YWNrEVWn3NEKiXXESpL?usp=sharing

Serafadam, 2023-08-29:
Thanks, looking at topics things seem fine. Could you post log information? It would be also best if you run `export DEPTHAI_DEBUG=1` before launching the camera. Additionally, output from `ros2 node list` and `ros2 doctor` would be useful.

saivojjala, 2023-08-30:
@Serafadam here is the log file for when camera.launch.py is launched from depthai_ros_driver package.
[launch.log](https://github.com/luxonis/depthai-ros/files/12471030/launch.log)

and the log file from depthai_filters spatial.launch.py is here.
[launch.log](https://github.com/luxonis/depthai-ros/files/12471049/launch.log)

the output of ros2 node list is as follows: 
1. for camera.launch.py
/launch_ros_3063673
/oak
/oak_container
/oak_state_publisher

2. for spatial_bb.launch.py
/launch_ros_3065671
/oak
/oak_container
/oak_state_publisher
/point_cloud_xyzrgb_node
/rectify_color_node
/spatial_bb_node

the output of ros2 doctor is:
1. for camera.launch.py
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz_ogre_vendor has been updated to a new version. local: 11.2.6 < latest: 11.2.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: depthimage_to_laserscan has been updated to a new version. local: 2.5.0 < latest: 2.5.1
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz2 has been updated to a new version. local: 11.2.6 < latest: 11.2.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz_rendering has been updated to a new version. local: 11.2.6 < latest: 11.2.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: image_transport has been updated to a new version. local: 3.1.5 < latest: 3.1.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: camera_calibration_parsers has been updated to a new version. local: 3.1.5 < latest: 3.1.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz_default_plugins has been updated to a new version. local: 11.2.6 < latest: 11.2.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz_common has been updated to a new version. local: 11.2.6 < latest: 11.2.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz_assimp_vendor has been updated to a new version. local: 11.2.6 < latest: 11.2.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: camera_info_manager has been updated to a new version. local: 3.1.5 < latest: 3.1.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 45: UserWarning: Subscriber without publisher detected on /joint_states.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/imu/data.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/nn/spatial_detections.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/rgb/camera_info.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/rgb/image_raw.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/rgb/image_raw/compressed.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/rgb/image_raw/compressedDepth.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/rgb/image_raw/theora.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/stereo/camera_info.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/stereo/image_raw.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/stereo/image_raw/compressed.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/stereo/image_raw/compressedDepth.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /oak/stereo/image_raw/theora.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /robot_description.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /tf.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /tf_static.

All 5 checks passed

2. for spatial_bb.launch.py

/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz_ogre_vendor has been updated to a new version. local: 11.2.6 < latest: 11.2.7
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: depthimage_to_laserscan has been updated to a new version. local: 2.5.0 < latest: 2.5.1
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/package.py: 112: UserWarning: rviz2 has been updated to a new version. local: 11.2.6 < latest: 11.2.7
[... cut ...]
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /robot_description.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /spatial_bb.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /tf.
/opt/ros/humble/lib/python3.10/site-packages/ros2doctor/api/topic.py: 42: UserWarning: Publisher without subscriber detected on /tf_static.

All 4 checks passed

Serafadam, 2023-08-31:
Thanks for the logs, could you additionally post a screenshot from rqt_graph tool?

saivojjala, 2023-08-31:
RQT Graph for camera.launch from depthai_ros_driver package: 
[image]

RQT Graph for spatial.launch from depthai_filters package: 
[image]

Serafadam, 2023-09-01:
Thanks, topic names seem okay, same with configuration. If you try to visualize pointcloud or marker message in rviz, nothing appears still? If not, for further debugging I would need a rosbag (legnth around 5s) that contains following topics:
- `/tf`
- `/tf_static`
- `/oak/nn/spatial_detections`
- `/oak/stereo/camera_info`
- `/oak/rgb/preview/image_raw`

Additionally, if it's possible, could you share your model and configuration json?

saivojjala, 2023-09-02:
Hi @Serafadam everything you requested is updated in the drive link
https://drive.google.com/drive/folders/1E4QjsKY0VNsL5YWNrEVWn3NEKiXXESpL?usp=sharing

Serafadam, 2023-09-04:
Hi, I was able to get images both on `/overlay` and `/spatial_bb`, one modification I had to implement was to edit launch file to add node name and parameters to the spatial_bb node
```
        LoadComposableNodes(
            target_container=name+"_container",
            composable_node_descriptions=[
                    ComposableNode(
                        package="depthai_filters",
                        plugin="depthai_filters::SpatialBB",
                        name="spatial_bb_node",
                        remappings=[
                                    ('stereo/camera_info', name+'/stereo/camera_info'),
                                    ('nn/spatial_detections', name+'/nn/spatial_detections'),
                                    ('rgb/preview/image_raw', name+'/rgb/preview/image_raw'),
                                    ],
                        parameters=[params_file],
                    ),
            ],
        ),
```

config file modification:
```
/oak:
  ros__parameters:
    camera:
      i_nn_type: spatial
    rgb:
      i_enable_preview: true
      i_keep_preview_aspect_ratio: false
      i_preview_size: 512
    nn:
      i_enable_passthrough: true
      i_disable_resize: true
      i_nn_config_path: path_to_config
    stereo:
      i_subpixel: true
/spatial_bb_node:
  ros__parameters:
    desqueeze: true
    label_map: 
            - "abdominal_spatula"
            - "arterial_clamp"
            - "bipolar_cable"
            - "bipolar_forcep"
            - "de_bakey_pean"
            - "durogrip_needle_holder"
            - "durotip_prep_scissor"
            - "durotip_surgery_scissor"
            - "grain_tong"
            - "grasping_forcep"
            - "hook"
            - "hook_cushing"
            - "hook_stove"
            - "intestinal_clamp"
            - "intestinal_forcep"
            - "kidney_dish"
            - "lab_dish"
            - "metz_scissor"
            - "mosquito_clamp"
            - "narrow_belly_hook"
            - "prep_clamp"
            - "probe"
            - "scalpel_handle"
            - "sharp_hook"
            - "slime_line_electrode"
            - "spike"
            - "sponge_plier"
            - "suction_needle"
            - "surge_retract"
            - "surgical_tweezer"
            - "towel_clamp"
            - "washing_plier"
            - "wide_belly_hook"
```

That launch file fix will be merged later today to Humble distribution. If you have any further questions please let me know.

saivojjala, 2023-09-04:
Hi @Serafadam, Thank you so much for all your help. It works for me as well now. One last question I have is regarding the update which would allow us to use models of non-square size. By when would it be integrated with the ros driver ?

Serafadam, 2023-09-05:
Glad I could help, the preview size options are available in [the newest Humble release](https://github.com/luxonis/depthai-ros/releases/tag/v2.8.0-humble) (will be available on apt in next package sync). Noetic & Iron versions will also be updated soon.

closed 2023-09-05 by saivojjala