luxonis/depthai-ros #374: depthai filter package colcon build fail in humble

tolas92, 2023-08-21:
i installed depthai-ros in humble running on ubuntu 20.04, but since the binary build(sudo apt install ros-humble-depthai-ros) did not have the depthai filter package. i tried to build  the depthai filter package from the source but ended up getting this below error : 
~/dai_ws$ colcon build 
Starting >>> depthai_filters
--- stderr: depthai_filters                             
In file included from /home/tolasing/dai_ws/src/depthai_filters/src/feature_tracker_overlay.cpp:1:
/home/tolasing/dai_ws/src/depthai_filters/include/depthai_filters/feature_tracker_overlay.hpp:4:10: fatal error: depthai_ros_msgs/msg/tracked_features.hpp: No such file or directory
    4 | #include "depthai_ros_msgs/msg/tracked_features.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/depthai_filters.dir/build.make:132: CMakeFiles/depthai_filters.dir/src/feature_tracker_overlay.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:137: CMakeFiles/depthai_filters.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< depthai_filters [0.47s, exited with code 2]

Summary: 0 packages finished [0.69s]
  1 package failed: depthai_filters
  1 package had stderr output: depthai_filters

how can i install the depthai filter package?

Serafadam, 2023-08-21:
Hi, I will add depthai_filters in the next update, to build it from source you'll need to additionally build whole repository together, as newest version of `humble` branch hasn't been officially released yet (it will be probably this week)

tolas92, 2023-08-21:
yes , its installed , thank you.

closed 2023-08-21 by tolas92