luxonis/depthai-ros #376: [BUG] {Got Error when launching the camera.launch.py}

Jenanaputra, 2023-08-22:
Hi @Serafadam , Iam using ubuntu 22.04 and working with ros2 humble. I installed the depthai-ros from source. I got the error when trying to launch `ros2 launch depthai_ros_driver camera.launch.py`.
`[INFO] [launch]: All log files can be found below /home/jenanaputra/.ros/log/2023-08-21-16-44-45-285141-manuel-ThinkPad-P50-10726
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [10740]
[INFO] [component_container-2]: process started with pid [10742]
[component_container-2] [DEBUG] [1692607485.596530462] [rclcpp]: signal handler installed
[component_container-2] [DEBUG] [1692607485.596681835] [rcl]: Initializing wait set with '0' subscriptions, '2' guard conditions, '0' timers, '0' clients, '0' services
[component_container-2] [DEBUG] [1692607485.596595934] [rclcpp]: deferred_signal_handler(): waiting for SIGINT/SIGTERM or uninstall
[component_container-2] [DEBUG] [1692607485.596744854] [rcl]: Initializing node 'ComponentManager' in namespace ''
[component_container-2] [DEBUG] [1692607485.596758340] [rcl]: Using domain ID of '1'
[component_container-2] [DEBUG] [1692607485.606209171] [rcl]: Initializing publisher for topic name '/rosout'
[component_container-2] [DEBUG] [1692607485.606244805] [rcl]: Expanded and remapped topic name '/rosout'
[component_container-2] [DEBUG] [1692607485.607690994] [rcl]: Publisher initialized
[component_container-2] [DEBUG] [1692607485.607714900] [rcl]: Node initialized
[component_container-2] [DEBUG] [1692607485.607880597] [rcl]: Initializing subscription for topic name '/parameter_events'
[component_container-2] [DEBUG] [1692607485.607897139] [rcl]: Expanded and remapped topic name '/parameter_events'
[component_container-2] [DEBUG] [1692607485.609038206] [rcl]: Subscription initialized
[component_container-2] [DEBUG] [1692607485.609226689] [rcl]: Initializing service for service name '~/_container/load_node'
[component_container-2] [DEBUG] [1692607485.609243146] [rcl]: Expanded and remapped service name '/oak_container/_container/load_node'
[component_container-2] [DEBUG] [1692607485.609728444] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1692607485.609741180] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/load_nodeRequest
[component_container-2] [DEBUG] [1692607485.609749371] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/load_nodeReply
[component_container-2] [DEBUG] [1692607485.609757294] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1692607485.609825783] [rcl]: Service initialized
[component_container-2] [DEBUG] [1692607485.609850374] [rcl]: Initializing service for service name '~/_container/unload_node'
[component_container-2] [DEBUG] [1692607485.609872488] [rcl]: Expanded and remapped service name '/oak_container/_container/unload_node'
[component_container-2] [DEBUG] [1692607485.610206501] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1692607485.610218098] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/unload_nodeRequest
[component_container-2] [DEBUG] [1692607485.610226047] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/unload_nodeReply
[component_container-2] [DEBUG] [1692607485.610233625] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1692607485.610290192] [rcl]: Service initialized
[component_container-2] [DEBUG] [1692607485.610314530] [rcl]: Initializing service for service name '~/_container/list_nodes'
[component_container-2] [DEBUG] [1692607485.610329225] [rcl]: Expanded and remapped service name '/oak_container/_container/list_nodes'
[component_container-2] [DEBUG] [1692607485.610648741] [rmw_fastrtps_cpp]: ************ Service Details *********
[component_container-2] [DEBUG] [1692607485.610660409] [rmw_fastrtps_cpp]: Sub Topic rq/oak_container/_container/list_nodesRequest
[component_container-2] [DEBUG] [1692607485.610668288] [rmw_fastrtps_cpp]: Pub Topic rr/oak_container/_container/list_nodesReply
[component_container-2] [DEBUG] [1692607485.610675851] [rmw_fastrtps_cpp]: ***********
[component_container-2] [DEBUG] [1692607485.610722043] [rcl]: Service initialized
[component_container-2] [DEBUG] [1692607485.613249653] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613299981] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613696153] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613710558] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613724228] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613731252] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613770015] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613778050] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613797895] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613812755] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613865672] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613886678] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613903881] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613914316] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613929130] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613938646] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.613952228] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.613962741] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.614021392] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.614033648] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.614057176] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.614067277] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.614096534] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.614106870] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.614130404] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.614140500] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.616046255] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.616085488] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.616104554] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.616117876] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607485.616132112] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607485.616154916] [rcl]: Subscription take succeeded: true
[... 115 lines cut ...]
[component_container-2] [DEBUG] [1692607485.910491758] [rcl]: Publisher initialized
[component_container-2] [DEBUG] [1692607485.910733121] [rcl]: Initializing subscription for topic name '/parameter_events'
[component_container-2] [DEBUG] [1692607485.910749727] [rcl]: Expanded and remapped topic name '/parameter_events'
[component_container-2] [DEBUG] [1692607485.911126435] [rcl]: Subscription initialized
[component_container-2] [DEBUG] [1692607485.911705730] [oak]: Setting param camera.i_pipeline_type with value RGBD
[component_container-2] [DEBUG] [1692607485.911779011] [oak]: Setting param camera.i_nn_type with value spatial
[component_container-2] [DEBUG] [1692607485.911850231] [oak]: Setting param camera.i_enable_imu with value 1
[component_container-2] [DEBUG] [1692607485.911907738] [oak]: Setting param camera.i_enable_ir with value 1
[component_container-2] [DEBUG] [1692607485.911964373] [oak]: Setting param camera.i_usb_speed with value SUPER_PLUS
[component_container-2] [DEBUG] [1692607485.912035331] [oak]: Setting param camera.i_mx_id with value 
[component_container-2] [DEBUG] [1692607485.912103157] [oak]: Setting param camera.i_ip with value 
[... cut ...]
[component_container-2] [2023-08-21 16:44:46.138] [debug] Device - BoardConfig: {"camera":[],"emmc":null,"gpio":[],"imu":null,"logDevicePrints":true,"logPath":null,"logSizeMax":null,"logVerbosity":null,"network":{"mtu":0,"xlinkTcpNoDelay":true},"nonExclusiveMode":false,"pcieInternalClock":null,"sysctl":[],"uart":[],"usb":{"flashBootedPid":63037,"flashBootedVid":999,"maxSpeed":4,"pid":63035,"vid":999},"usb3PhyInternalClock":null,"watchdogInitialDelayMs":null,"watchdogTimeoutMs":null} 
[component_container-2] libnop:`

Jenanaputra, 2023-08-22:
`[component_container-2] 0000: b9 11 b9 05 81 e7 03 81 3b f6 81 e7 03 81 3d f6 04 b9 02 00 01 ba 00 be be bb 00 bb 00 be be be
[component_container-2] 0020: be be be 01 00 bb 00 be
[component_container-2] [2023-08-21 16:44:46.408] [debug] Resources - Archive 'depthai-device-fwp-3575b77f20e796b4e79953bf3d2ba22f0416ee8b.tar.xz' open: 2ms, archive read: 492ms
[component_container-2] [2023-08-21 16:44:46.439] [debug] Searching for booted device: DeviceInfo(name=192.168.8.224, mxid=19443010A188211300, X_LINK_BOOTLOADER, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[component_container-2] [2023-08-21 16:44:46.717] [debug] Connected bootloader version 0.0.19
[component_container-2] [2023-08-21 16:44:46.717] [info] New bootloader version available. Device has: 0.0.19, available: 0.0.24
[component_container-2] [2023-08-21 16:44:47.620] [debug] Booting FW with Bootloader. Version 0.0.19, Time taken: 902ms
[component_container-2] [2023-08-21 16:44:47.620] [debug] DeviceBootloader about to be closed...
[component_container-2] [2023-08-21 16:44:47.620] [debug] XLinkResetRemote of linkId: (0)
[component_container-2] [2023-08-21 16:44:48.869] [debug] DeviceBootloader closed, 1249
[component_container-2] [2023-08-21 16:44:48.871] [debug] Searching for booted device: DeviceInfo(name=192.168.8.224, mxid=19443010A188211300, X_LINK_BOOTED, X_LINK_TCP_IP, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[component_container-2] [19443010A188211300] [192.168.8.224] [5.284] [system] [warning] PRINT:LeonCss: GPIO boot mode 0x3, interface SPI_MASTER_EFF
[component_container-2] Setting aons(0..4) back to boot from flash (offset = 0)
[component_container-2] ====ENABLE WATCHDOG====1
[component_container-2] initial keepalive, countdown: 7
[component_container-2] PLL0: 700000 AUX_IO0: 24000 AUX_IO1: 24000 MCFG: 24000 MECFG: 24000
[component_container-2] Board init ret 3
[component_container-2] DepthAI Firmware - version: 3575b77f20e796b4e79953bf3d2ba22f0416ee8b
[component_container-2] eeprom configuration version: 55AA0007
[component_container-2] Reading VERSION 7 --- ->
[component_container-2] eeprom configuration load from user area, status: 0
[component_container-2] Reading VERSION 7 --- ->
[component_container-2] Didn't find any boards matching the eeprom data
[component_container-2] Board options: 00000001
[component_container-2] --> brdInit ...
[component_container-2] brdInitAuxDevices: Error: SC = 27: io_initialize expander_cam_gpios_1 [OK]
[component_container-2] 
[component_container-2] spi_N25Q_init: Flash JEDEC ID: 01 20 18
[component_container-2] CR1NV: 0x02
[component_container-2] CR2NV: 0x08
[component_container-2] QUAD mode already enabled
[component_container-2] inited hal ---
[component_container-2] def: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
[component_container-2] val: 53 46 45 34 10 06 08 34 10 53 14 45 46 08 06
[component_container-2] Opening bus for IR driver: 1
[component_container-2] Failed to probe IR driver LM3644
[component_container-2] Opening bus for IR driver: 2
[component_container-2] Failed to probe IR driver LM3644
[component_container-2] Opening bus for IR driver: 3
[component_container-2] Failed to probe IR driver LM3644
[component_container-2] Closing EEPROm!
[component_container-2] MyriaX board configuration
[component_container-2] pll0 frequency: 700000, ref0 frequency: 24000
[component_container-2] Setup network with bootloader config
[component_container-2] OsDrvCprSysDeviceAction() for PCIe failed: 13
[component_container-2] nexus0: <RTEMS Nexus device>
[component_container-2] pcib0: <MyriadX PCIe RC> on nexus0
[component_container-2] PCIE_REGS_PCIE_CFG_ADR = 0x60100
[component_container-2] PCIE_REGS_PCIE_PLL_CNTRL_ADR = 0x1
[component_container-2] pcib0: Link training complete, PCIe downstream link is UP
[component_container-2] pcib0: myriad_pcie_err_interrupt: SMLH LINK UP
[component_container-2] pcib0: myriad_pcie_err_interrupt: SMLH LINK DOWN
[component_container-2] pcib0: myriad_pcie_err_interrupt: RDLH LINK UP
[component_container-2] pcib0: myriad_pcie_err_interrupt: RDLH LINK DOWN
[component_container-2] pcib0: myriad_pcie_err_interrupt: CORE RST
[component_container-2] pcib0: myriad_pcie_err_interrupt: CORE RSTN
[component_container-2] pcib0: myriad_pcie_err_interrupt: LINK REQ RST
[component_container-2] pci0: <PCI bus> on pcib0
[component_container-2] memcpy_dma_init: got DMA id 62
[component_container-2] re0: <Realtek PCIe GBE Family Controller> at device 0.0 on pci0
[component_container-2] re0: Using Memory Mapping!
[component_container-2] re0: Using 1 MSI message
[component_container-2] Overriden invalid mac address, mac: 44:A9:2C:31:FE:C4
[component_container-2] re0: version:1.95.00
[component_container-2] re0: Ethernet address: 0x83d45178
[component_container-2] 
[component_container-2] This product is covered by one or more of the following patents:            
[component_container-2] US6,570,884, US6,115,776, and US6,327,625.
[component_container-2] info: re0: Ethernet address: 44:a9:2c:31:fe:c4
[component_container-2] myriad_mmc2: <MyriadX MMC> on nexus0
[... 220 lines cut ...]
[component_container-2] [DEBUG] [1692607493.290258226] [oak]: Setting param stereo.i_subpixel with value 0
[component_container-2] [DEBUG] [1692607493.290331476] [oak]: Setting param stereo.i_extended_disp with value 0
[component_container-2] [DEBUG] [1692607493.290396112] [oak]: Setting param stereo.i_rectify_edge_fill_color with value 0
[... cut ...]
[component_container-2] [DEBUG] [1692607493.436135368] [oak]: Setting param nn.i_num_pool_frames with value 4
[component_container-2] [DEBUG] [1692607493.436303058] [oak]: Setting param nn.i_num_inference_threads with value 2
[component_container-2] [DEBUG] [1692607493.436350947] [oak]: Setting param nn.i_max_q_size with value 30
[component_container-2] [DEBUG] [1692607493.436963212] [oak]: Setting param nn.i_label_map with value background aeroplane bicycle bird boat bottle bus car cat chair cow diningtable dog horse motorbike person pottedplant sheep sofa train tvmonitor`

Jenanaputra, 2023-08-22:
`[component_container-2] [DEBUG] [1692607493.437150461] [oak]: Node nn created
[component_container-2] [DEBUG] [1692607493.437307002] [oak]: Base node nn created
[component_container-2] [DEBUG] [1692607493.437591193] [oak]: Creating node imu
[component_container-2] [19443010A188211300] [192.168.8.224] [5.591] [system] [warning] PRINT:LeonCss: Error [-3] : Device not found error. It occurs when the device chip id is incorrectly read
[component_container-2] [19443010A188211300] [192.168.8.224] [5.745] [system] [warning] PRINT:LeonCss: SH2 Reset.
[component_container-2] [DEBUG] [1692607493.677834805] [oak]: Setting param imu.i_message_type with value IMU
[component_container-2] [DEBUG] [1692607493.678093547] [oak]: Setting param imu.i_sync_method with value LINEAR_INTERPOLATE_ACCEL
[component_container-2] [DEBUG] [1692607493.678252477] [oak]: Setting param imu.i_acc_cov with value 0
[component_container-2] [DEBUG] [1692607493.678304924] [oak]: Setting param imu.i_gyro_cov with value 0
[component_container-2] [DEBUG] [1692607493.678405651] [oak]: Setting param imu.i_rot_cov with value -1
[component_container-2] [DEBUG] [1692607493.678448747] [oak]: Setting param imu.i_mag_cov with value 0
[component_container-2] [DEBUG] [1692607493.678510830] [oak]: Setting param imu.i_enable_rotation with value 0
[component_container-2] [DEBUG] [1692607493.678572400] [oak]: Setting param imu.i_acc_freq with value 400
[component_container-2] [DEBUG] [1692607493.678655689] [oak]: Setting param imu.i_gyro_freq with value 400
[component_container-2] [DEBUG] [1692607493.678751720] [oak]: Setting param imu.i_batch_report_threshold with value 1
[component_container-2] [DEBUG] [1692607493.678794665] [oak]: Setting param imu.i_max_batch_reports with value 10
[component_container-2] [DEBUG] [1692607493.678890968] [oak]: Node imu created
[component_container-2] [INFO] [1692607493.678968308] [oak]: Finished setting up pipeline.
[component_container-2] [2023-08-21 16:44:53.704] [debug] Schema dump: {"connections":[{"node1Id":0,"node1Output":"preview","node1OutputGroup":"","node2Id":10,"node2Input":"inputImage","node2InputGroup":""},{"node1Id":9,"node1Output":"out","node1OutputGroup":"","node2Id":11,"node2Input":"in","node2InputGroup":""},{"node1Id":3,"node1Output":"depth","node1OutputGroup":"","node2Id":9,"node2Input":"inputDepth","node2InputGroup":""},{"node1Id":10,"node1Output":"out","node1OutputGroup":"","node2Id":9,"node2Input":"in","node2InputGroup":""},{"node1Id":6,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"right","node2InputGroup":""},{"node1Id":4,"node1Output":"out","node1OutputGroup":"","node2Id":3,"node2Input":"left","node2InputGroup":""},{"node1Id":3,"node1Output":"depth","node1OutputGroup":"","node2Id":8,"node2Input":"in","node2InputGroup":""},{"node1Id":7,"node1Output":"out","node1OutputGroup":"","node2Id":6,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":5,"node1Output":"out","node1OutputGroup":"","node2Id":4,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":12,"node1Output":"out","node1OutputGroup":"","node2Id":13,"node2Input":"in","node2InputGroup":""},{"node1Id":2,"node1Output":"out","node1OutputGroup":"","node2Id":0,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"isp","node1OutputGroup":"","node2Id":1,"node2Input":"in","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"sippBufferSize":18432,"sippDmaBufferSize":16384,"xlinkChunkSize":-1},"nodes":[[12,{"id":12,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":49,"name":"out","queueSize":8,"type":0,"waitForMessage":false}]],"name":"IMU","properties":[185,4,186,2,185,5,0,0,0,129,144,1,20,185,5,0,0,0,129,144,1,21,1,10,0]}],[13,{"id":13,"ioInfo":[[["","in"],{"blocking":true,"group":"","id":1,"name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":[185,3,136,0,0,128,191,189,7,105,109,117,95,105,109,117,0]}],[0,{"id":0,"ioInfo":[[["","video"],{"blocking":false,"group":"","id":9,"name":"video","queueSize":8,"type":0,"waitForMessage":false}],[["","still"],{"blocking":false,"group":"","id":7,"name":"still","queueSize":8,"type":0,"waitForMessage":false}],[["","isp"],{"blocking":false,"group":"","id":6,"name":"isp","queueSize":8,"type":0,"waitForMessage":false}],[["","preview"],{"blocking":false,"group":"","id":8,"name":"preview","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","id":5,"name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","id":4,"name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputConfig"],{"blocking":false,"group":"","id":3,"name":"inputConfig","queueSize":8,"type":3,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","id":2,"name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"ColorCamera","properties":[185,25,185,27,0,3,0,0,0,185,3,0,0,0,185,5,0,0,0,0,0,185,5,0,0,0,0,0,0,0,0,0,0,0,0,185,3,0,0,0,185,3,0,0,0,0,0,0,0,0,0,0,0,0,0,0,189,0,255,0,0,0,129,44,1,129,44,1,133,0,5,133,208,2,255,255,0,136,0,0,112,66,0,136,0,0,128,191,136,0,0,128,191,1,185,4,2,3,2,3,3,3,4,4,4]}],[1,{"id":1,"ioInfo":[[["","in"],{"blocking":true,"group":"","id":10,"name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkO [... cut ...]
[component_container-2] [DEBUG] [1692607494.692620728] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/libtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692627793] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/liblibtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692634590] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/libtheora_image_transport_component.so'`

Jenanaputra, 2023-08-22:
`[component_container-2] [DEBUG] [1692607494.692641291] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/liblibtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692649628] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/libtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692656948] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/liblibtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692664297] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/libtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692679948] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/liblibtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692687829] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/libtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692694567] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib/theora_image_transport/liblibtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692701202] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/libtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692707826] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/liblibtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692715016] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/libtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692722908] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/liblibtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692731150] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/libtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692738591] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/liblibtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692747191] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/libtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692754064] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/lib64/theora_image_transport/liblibtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692760895] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/libtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692769339] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/liblibtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692775998] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/libtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692782748] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/liblibtheora_image_transport_component.so'
[component_container-2] [DEBUG] [1692607494.692789611] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/libtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692796751] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/liblibtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692803493] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/libtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692810221] [pluginlib.ClassLoader]: [search path for 'theora_image_transport_component']: '/opt/ros/humble/bin/theora_image_transport/liblibtheora_image_transport_componentd.so'
[component_container-2] [DEBUG] [1692607494.692819619] [pluginlib.ClassLoader]: Iterating through all possible paths where theora_image_transport_component could be located...
[component_container-2] [DEBUG] [1692607494.692826754] [pluginlib.ClassLoader]: Checking path /opt/ros/humble/lib/libtheora_image_transport_component.so 
[... cut ...]
[component_container-2] [DEBUG] [1692607495.353732081] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353738800] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353744991] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353751239] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353778785] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353787264] [rcl]: Subscription take succeeded: true`

Jenanaputra, 2023-08-22:
`[component_container-2] [DEBUG] [1692607495.353793463] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353799350] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353817291] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353824212] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353830481] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353836351] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353854041] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353884838] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353899280] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353906499] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353941620] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353948572] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353954844] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353960748] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353978529] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353985374] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.353991539] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.353997544] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354016980] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354023880] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354030046] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354035933] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354053646] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354060478] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354066608] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354072522] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354090192] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354097104] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354103248] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354109201] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354126966] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354133776] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354139834] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354145898] [rcl]: Subscription take succeeded: true
[component_container-2] [DEBUG] [1692607495.354163664] [rcl]: Subscription taking message
[component_container-2] [DEBUG] [1692607495.354170634] [rcl]: Subscription take succeeded: true
[component_container-2] [19443010A188211300] [192.168.8.224] [7.406] [system] [warning] PRINT:LeonCss: W: 4 - 0 - 0x86cccfc0
[component_container-2] W: Isp cfg mising: 0 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.494] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 0 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.505] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 1 
[component_container-2] W: Isp cfg mising: 2 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.560] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 0 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.592] [system] [warning] PRINT:LeonCss: W: 4 - 0 - 0x8707c4c0
[component_container-2] W: Isp cfg mising: 0 
[component_container-2] W: Isp cfg mising: 1 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.615] [system] [warning] PRINT:LeonMss: MIPI HS error on ctrl 4, err 0x8000
[component_container-2] [19443010A188211300] [192.168.8.224] [7.637] [system] [warning] PRINT:LeonCss: W: Mipi internal error, frame skipped: 0 
[component_container-2] W: 4 - 0 - 0x86d9e9c0
[component_container-2] [19443010A188211300] [192.168.8.224] [7.659] [system] [warning] PRINT:LeonCss: W: 4 - 0 - 0x86ed90c0
[component_container-2] [19443010A188211300] [192.168.8.224] [7.747] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 2 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.758] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 1 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.769] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 0 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.780] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 2 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.890] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 0 
[component_container-2] W: Isp cfg mising: 1 
[component_container-2] [19443010A188211300] [192.168.8.224] [7.901] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 2 
[... cut ...]
[component_container-2] [19443010A188211300] [192.168.8.224] [28.611] [system] [warning] PRINT:LeonCss: memcpy_dma: needed 2 dst reads
[component_container-2] [19443010A188211300] [192.168.8.224] [28.655] [system] [warning] PRINT:LeonCss: W: Isp cfg mising: 1 
[component_container-2] W: Isp cfg mising: 2 
[component_container-2] free(): double free detected in tcache 2
[ERROR] [component_container-2]: process has died [pid 10742, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container --ros-args --log-level debug --ros-args -r __node:=oak_container -r __ns:=/'].`

Jenanaputra, 2023-08-22:
I also tried to print the pipeline graph when launch the camera.launch.py. The output seems like this :
[image]

Jenanaputra, 2023-08-22:
The output for the 'log_system_information.py' is like this :
`{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.19.0-50-generic-x86_64-with-glibc2.35",
    "processor": "x86_64",
    "python_build": "main Jun 11 2023 05:26:28",
    "python_compiler": "GCC 11.4.0",
    "python_implementation": "CPython",
    "python_version": "3.10.12",
    "release": "5.19.0-50-generic",
    "system": "Linux",
    "version": "#50-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul 10 18:24:29 UTC 2023",
    "win32_ver": "",
    "packages": [
        "action-msgs==1.2.1",
        "action-tutorials-interfaces==0.20.2",
        "action-tutorials-py==0.20.2",
        "actionlib-msgs==4.2.2",
        "ament-cmake-test==1.3.3",
        "ament-copyright==0.12.4",
        "ament-cppcheck==0.12.4",
        "ament-cpplint==0.12.4",
        "ament-flake8==0.12.4",
        "ament-index-python==1.4.0",
        "ament-lint==0.12.4",
        "ament-lint-cmake==0.12.4",
        "ament-package==0.14.0",
        "ament-pep257==0.12.4",
        "ament-uncrustify==0.12.4",
        "ament-xmllint==0.12.4",
        "angles==1.15.0",
        "appdirs==1.4.4",
        "apturl==0.5.2",
        "argcomplete==1.8.1",
        "asciinema==2.1.0",
        "asciitree==0.3.3",
        "attrs==21.2.0",
        "bcrypt==3.2.0",
        "beautifulsoup4==4.10.0",
        "beniget==0.4.1",
        "blinker==1.4",
        "bluerobotics-ping==0.1.4",
        "bond==3.0.2",
        "Brlapi==0.8.3",
        "Brotli==1.0.9",
        "builtin-interfaces==1.2.1",
        "cachetools==5.0.0",
        "camera-calibration==3.0.0",
        "cartographer-ros-msgs==2.0.9000",
        "catkin==0.8.10",
        "catkin-pkg-modules==0.5.2",
        "certifi==2020.6.20",
        "chardet==4.0.0",
        "click==8.0.3",
        "colcon-argcomplete==0.3.3",
        "colcon-bash==0.4.2",
        "colcon-cd==0.1.1",
        "colcon-cmake==0.2.27",
        "colcon-common-extensions==0.3.0",
        "colcon-core==0.11.0",
        "colcon-defaults==0.2.7",
        "colcon-devtools==0.2.3",
        "colcon-installed-package-information==0.1.0",
        "colcon-library-path==0.2.1",
        "colcon-metadata==0.2.5",
        "colcon-notification==0.2.15",
        "colcon-output==0.2.12",
        "colcon-override-check==0.0.1",
        "colcon-package-information==0.3.3",
[... 316 lines cut ...]
        "unattended-upgrades==0.1",
        "unicodedata2==14.0.0",
        "unique-identifier-msgs==2.2.1",
        "uritemplate==3.0.1",
        "urllib3==1.26.5",
        "usb-creator==0.3.7",
        "vcstool==0.3.0",
        "velodyne-msgs==2.3.0",
        "vision-msgs==4.1.0",
        "visualization-msgs==4.2.2",
        "wadllib==1.3.6",
        "webencodings==0.5.1",
        "wheel==0.37.1",
        "xacro==2.0.8",
        "xdg==5",
        "xkit==0.0.0",
        "zipp==1.0.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux manuel-ThinkPad-P50 5.19.0-50-generic #50-Ubuntu SMP PREEMPT_DYNAMIC Mon Jul 10 18:24:29 UTC 2023 x86_64 x86_64"
    ]
}`

Jenanaputra, 2023-08-22:
The output of `apt show ros-$ROS_DISTRO-depthai ros-$ROS_DISTRO-depthai-ros ros-$ROS_DISTRO-depthai-bridge ros-$ROS_DISTRO-depthai-ros-msgs ros-$ROS_DISTRO-depthai-ros-driver` is like this :

`Package: ros-humble-depthai
Version: 2.22.0-1jammy.20230623.022440
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 20.7 MB
Depends: libc6 (>= 2.34), libgcc-s1 (>= 4.2), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 12), libopencv-dev, libusb-1.0-0-dev, nlohmann-json3-dev, ros-humble-ros-workspace
Homepage: https://www.luxonis.com/
Download-Size: 10.5 MB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: DepthAI core is a C++ library which comes with firmware and an API to interact with OAK Platform

Package: ros-humble-depthai-ros
Version: 2.7.5-1jammy.20230814.161608
Priority: optional
Section: misc
Maintainer: sachin <sachin@luxonis.com>
Installed-Size: 42.0 kB
Depends: ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-descriptions, ros-humble-depthai-examples, ros-humble-depthai-ros-driver, ros-humble-depthai-ros-msgs, ros-humble-ros-workspace
Download-Size: 5,728 B
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: The depthai-ros package

Package: ros-humble-depthai-bridge
Version: 2.7.5-1jammy.20230814.154313
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 299 kB
Depends: libc6 (>= 2.32), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgcodecs4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), libboost-dev, libopencv-dev, ros-humble-camera-info-manager, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-ros-msgs, ros-humble-image-transport, ros-humble-rclcpp, ros-humble-robot-state-publisher, ros-humble-ros-environment, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-stereo-msgs, ros-humble-vision-msgs, ros-humble-xacro, ros-humble-ros-workspace
Download-Size: 76.8 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: The depthai_bridge package

Package: ros-humble-depthai-ros-msgs
Version: 2.7.5-1jammy.20230810.171220
Priority: optional
Section: misc
Maintainer: Sachin Guruswamy <sachin@luxonis.com>
Installed-Size: 1,500 kB
Depends: libc6 (>= 2.14), libgcc-s1 (>= 3.3.1), libpython3.10 (>= 3.10.0), libstdc++6 (>= 11), ros-humble-fastcdr, ros-humble-builtin-interfaces, ros-humble-geometry-msgs, ros-humble-rclcpp, ros-humble-rosidl-default-generators, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 114 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Package to keep interface independent of the driver

Package: ros-humble-depthai-ros-driver
Version: 2.7.5-1jammy.20230814.160555
Priority: optional
Section: misc
Maintainer: Adam Serafin <adam.serafin@luxonis.com>
Installed-Size: 2,047 kB
Depends: libc6 (>= 2.34), libconsole-bridge1.0 (>= 1.0.1+dfsg2), libgcc-s1 (>= 3.3.1), libopencv-core4.5d (>= 4.5.4+dfsg), libopencv-imgproc4.5d (>= 4.5.4+dfsg), libstdc++6 (>= 11), libtinyxml2-9 (>= 8.0.0), ros-humble-ament-cmake-auto, ros-humble-camera-calibration, ros-humble-cv-bridge, ros-humble-depthai, ros-humble-depthai-bridge, ros-humble-depthai-descriptions, ros-humble-depthai-examples, ros-humble-depthai-ros-msgs, ros-humble-diagnostic-msgs, ros-humble-image-pipeline, ros-humble-image-transport, ros-humble-image-transport-plugins, ros-humble-pluginlib, ros-humble-rclcpp, ros-humble-rclcpp-components, ros-humble-sensor-msgs, ros-humble-std-msgs, ros-humble-std-srvs, ros-humble-vision-msgs, ros-humble-ros-workspace
Download-Size: 482 kB
APT-Sources: http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
Description: Depthai ROS Monolithic node.

N: There is 1 additional record. Please use the '-a' switch to see it`

Those are the errors that i got, Do you have any idea about the issue?

Thanks

Serafadam, 2023-08-22:
Hi, thanks for the verbose output! I'll try to reproduce that setup, although looking by that error it might take some time. Now a few questions/remarks:
- [PoE cameras have higher CPU](https://docs.luxonis.com/projects/api/en/latest/tutorials/debugging/#cpu-usage) usage, as seeen in the logs `Cpu Usage - LeonOS 100.00%, LeonRT: 11.42%`, could you try lowering camera's FPS to 15 (and, for example enable low_bandwidth mode for rgb) and check if that helps? New Humble release will also include setISP3A parameter for each of the streams which should also lower the CPU usage.
- Additionally, you could try running camera driver with `gdb`, newest `humble` branch has an argument to enable that, or you could just create another launch file with those arguments.

closed 2023-09-11 by Serafadam