luxonis/depthai-ros #387: Oak D Lite is not doing any object detection

Femi1984, 2023-09-03:
Can  Oak D Lite camera do object and spatial detection? If so what are the specific params that need to be considered to make it possible. I'm having a hard time making the camera do object detection with this depth-ros package. The RGB and spatial shows in output image but no detection

Serafadam, 2023-09-04:
Hi, 
Yes, OAK D Lite can perform spatial detections. In this repository we have several code examples, could you add more information as mentioned in [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)?

Femi1984, 2023-09-04:
I'm not getting any errors. I'm launching the example_marker_publisher from the depthai_ros_driver repository. When launched,  the raw_image shows in rviz, but //oak/nn/ text_marker does not show. When I echo the /oak/nn/text_marker topic, no data is passing through it.I'm trying to accomplish spatial_mobilenet.py from the depthai_python example repository with the oak d lite in ros. I'm not sure how to go  about this

Femi1984, 2023-09-05:
I'm using foxy, 20.04

Femi1984, 2023-09-05:
This is error that I'm gettin

[component_container-2] [ERROR] [1693941467.630870360] [oak_container]: Component constructor threw an exception: IMU(12) - IMU invalid settings!: IMU not detected. Your board doesn't have IMU or requires firmware update (Set imu.enableFirmwareUpdate(True) explicitly.). 1, 321
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'oak' of type 'depthai_ros_driver::Camera' in container '/oak_container': Component constructor threw an exception: IMU(12) - IMU invalid settings!: IMU not detected. Your board doesn't have IMU or requires firmware update (Set imu.enableFirmwareUpdate(True) explicitly.). 1, 321
[component_container-2] [INFO] [1693941467.639676439] [oak_container]: Load Library: /opt/ros/foxy/lib/librectify.so
[component_container-2] [INFO] [1693941467.648952381] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[component_container-2] [INFO] [1693941467.649035037] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/rectify_color_node' in container 'oak_container'
[component_container-2] [INFO] [1693941467.683528661] [oak_container]: Load Library: /opt/ros/foxy/lib/libdepth_image_proc.so
[component_container-2] [INFO] [1693941467.694078856] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::ConvertMetricNode>
[component_container-2] [INFO] [1693941467.694201097] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::CropForemostNode>
[component_container-2] [INFO] [1693941467.694220842] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::DisparityNode>
[component_container-2] [INFO] [1693941467.694234730] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzNode>
[component_container-2] [INFO] [1693941467.694246730] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzRadialNode>
[component_container-2] [INFO] [1693941467.694257962] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziNode>
[component_container-2] [INFO] [1693941467.694268170] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyziRadialNode>
[component_container-2] [INFO] [1693941467.694278186] [oak_container]: Found class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[component_container-2] [INFO] [1693941467.694288010] [oak_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<depth_image_proc::PointCloudXyzrgbNode>
[component_container-2] [ERROR] [1693941467.703559439] [oak_container]: Failed to find class with the requested plugin name 'depthai_filters::SpatialBB' in the loaded library
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/point_cloud_xyzrgb_node' in container 'oak_container'
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'spatial_bb_node' of type 'depthai_filters::SpatialBB' in container 'oak_container': Failed to find class with the requested plugin

Serafadam, 2023-09-06:
Hi, Foxy distribution is not supported as it is EOL. To try out SpatialBB, you could try running it in docker as such:
`docker run -it -v /dev/:/dev/  --privileged -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix luxonis/depthai-ros:v2.8.0-humble ros2 launch depthai_filters spatial_bb.launch.py`

Serafadam, 2023-10-16:
Closing due to inactivity, please reopen if you have further questions/remarks.

closed 2023-10-16 by Serafadam