luxonis/depthai-ros #388: How can i get the left and right depth by launch examples?

jackslife2022, 2023-09-04:
I want to get the depth be published, which launch example can publish the right and left depth?
thanks

danilo-pejovic, 2023-09-04:
What do you mean by right and left depth? Depth image is calculated from 2 cameras (right and left one) - so we would only have 1 depth image. 

If you want to just publish streams from left and right camera you can do so by (for example) running camera.launch.py file from depthai_ros_driver package and changing parameters that you can see here [https://github.com/luxonis/depthai-ros/blob/humble/depthai_ros_driver/config/camera.yaml](url) to suit your needs. If you want to publish regular streams from left and right cameras you would need to change 

i_publish_topic: false to i_publish_topic: true under left and right camera field.

Serafadam, 2023-10-16:
Closing due to inactivity, please reopen if you have further questions/remarks.

closed 2023-10-16 by Serafadam