luxonis/depthai-ros #390: Cannot disable enableDotProjector and enableFloodLight

jeffrrrrr, 2023-09-05:
In v2.7.5-noetic branch
`roslaunch depthai_examples/launch/stereo_inertial_node.launch 
`
with this setting
```
    <arg name="enableDotProjector"    default="false"/>
    <arg name="enableFloodLight"      default="false"/>
    <arg name="dotProjectormA"        default="200.0"/>
    <arg name="floodLightmA"          default="200.0"/>
```
Device: OSK-D LR

But the output left and right images are still have dot projected as in the screen shot below.
[image]

Serafadam, 2023-09-05:
Hi, OAK-D-LR doesn't have IR filter, what's visible on the pictures is probably a filter since stereo_inertial_node is set up for Mono pair. Sensor type autodetection is done automatically in depthai_ros_driver.

jeffrrrrr, 2023-09-22:
hi, thanks for reply. is it a software level filter? can it be disabled?

Serafadam, 2023-09-28:
Hi, it should be solved automatically in depthai-ros-driver, if that doesn't happen please post additional debug information as in the [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md)