luxonis/depthai-ros #393: Unable to set required resolution on OAK-1 Lite camera as specified on the website.

makarandmandolkar, 2023-09-06:
**Describe the bug**
We are trying to use the OAK-1 Lite camera for our application. The website specifies 4K resolutions and 4208x3120 (13MP), but its been a little trouble achieving the same.
We are using depthai-ros noetic build from source, and launching the camera.launch from the depthai_ros_driver package.
It uses the camera.yaml file for the arguments.
On changing line 46 in the camera.yaml "rgb_i_resolution: '4K'", it was observed that it will still result in 2K output.
Moreover, any other value set in the line 46 will result in HD output or error.
Currently setting that value at 13MP results in a resolution of 2806X2080 and not  4208x3120.
Adjusting other values regarding the resolution in the config files didn't prove very helpful either.

It will be helpful to have a straightforward way/guide to achieve 4K, 13MP and HD resolution, just by changing the values in the config files.

**Screenshots**
[image]
[image]

**Pipeline Graph**

{"connections":[{"node1Id":2,"node1Output":"out","node1OutputGroup":"","node2Id":0,"node2Input":"inputControl","node2InputGroup":""},{"node1Id":0,"node1Output":"isp","node1OutputGroup":"","node2Id":1,"node2Input":"in","node2InputGroup":""}],"globalProperties":{"calibData":null,"cameraTuningBlobSize":null,"cameraTuningBlobUri":"","leonCssFrequencyHz":700000000.0,"leonMssFrequencyHz":700000000.0,"pipelineName":null,"pipelineVersion":null,"sippBufferSize":18432,"sippDmaBufferSize":16384,"xlinkChunkSize":-1},"nodes":[[0,{"id":0,"ioInfo":[[["","video"],{"blocking":false,"group":"","id":10,"name":"video","queueSize":8,"type":0,"waitForMessage":false}],[["","still"],{"blocking":false,"group":"","id":8,"name":"still","queueSize":8,"type":0,"waitForMessage":false}],[["","isp"],{"blocking":false,"group":"","id":7,"name":"isp","queueSize":8,"type":0,"waitForMessage":false}],[["","preview"],{"blocking":false,"group":"","id":9,"name":"preview","queueSize":8,"type":0,"waitForMessage":false}],[["","raw"],{"blocking":false,"group":"","id":6,"name":"raw","queueSize":8,"type":0,"waitForMessage":false}],[["","frameEvent"],{"blocking":false,"group":"","id":5,"name":"frameEvent","queueSize":8,"type":0,"waitForMessage":false}],[["","inputConfig"],{"blocking":false,"group":"","id":4,"name":"inputConfig","queueSize":8,"type":3,"waitForMessage":false}],[["","inputControl"],{"blocking":true,"group":"","id":3,"name":"inputControl","queueSize":8,"type":3,"waitForMessage":false}]],"name":"ColorCamera","properties":{"boardSocket":0,"cameraName":"","colorOrder":0,"fp16":false,"fps":30.0,"imageOrientation":-1,"initialControl":{"aeLockMode":false,"aeRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"afRegion":{"height":0,"priority":0,"width":0,"x":0,"y":0},"antiBandingMode":0,"autoFocusMode":3,"awbLockMode":false,"awbMode":0,"brightness":0,"chromaDenoise":0,"cmdMask":0,"contrast":0,"effectMode":0,"expCompensation":0,"expManual":{"exposureTimeUs":0,"frameDurationUs":0,"sensitivityIso":0},"frameSyncMode":0,"lensPosAutoInfinity":0,"lensPosAutoMacro":0,"lensPosition":0,"lowPowerNumFramesBurst":0,"lowPowerNumFramesDiscard":0,"lumaDenoise":0,"saturation":0,"sceneMode":0,"sharpness":0,"strobeConfig":{"activeLevel":0,"enable":0,"gpioNumber":0},"strobeTimings":{"durationUs":0,"exposureBeginOffsetUs":0,"exposureEndOffsetUs":0},"wbColorTemp":0},"interleaved":false,"isp3aFps":0,"ispScale":{"horizDenominator":3,"horizNumerator":2,"vertDenominator":3,"vertNumerator":2},"numFramesPoolIsp":3,"numFramesPoolPreview":4,"numFramesPoolRaw":3,"numFramesPoolStill":4,"numFramesPoolVideo":4,"previewHeight":300,"previewKeepAspectRatio":true,"previewWidth":300,"rawPacked":null,"resolution":3,"sensorCropX":-1.0,"sensorCropY":-1.0,"stillHeight":-1,"stillWidth":-1,"videoHeight":2806,"videoWidth":2080}}],[1,{"id":1,"ioInfo":[[["","in"],{"blocking":true,"group":"","id":2,"name":"in","queueSize":8,"type":3,"waitForMessage":true}]],"name":"XLinkOut","properties":{"maxFpsLimit":-1.0,"metadataOnly":false,"streamName":"rgb_isp"}}],[2,{"id":2,"ioInfo":[[["","out"],{"blocking":false,"group":"","id":1,"name":"out","queueSize":8,"type":0,"waitForMessage":false}]],"name":"XLinkIn","properties":{"maxDataSize":5242880,"numFrames":8,"streamName":"rgb_control"}}]]}

**Attach system log**
{
    "architecture": "64bit ELF",
    "machine": "x86_64",
    "platform": "Linux-5.15.0-78-generic-x86_64-with-glibc2.29",
    "processor": "x86_64",
    "python_build": "default May 26 2023 14:05:08",
    "python_compiler": "GCC 9.4.0",
    "python_implementation": "CPython",
    "python_version": "3.8.10",
    "release": "5.15.0-78-generic",
    "system": "Linux",
    "version": "#85~20.04.1-Ubuntu SMP Mon Jul 17 09:42:39 UTC 2023",
    "win32_ver": "",
    "packages": [
        "action-tutorials-py==0.14.4",
        "actionlib==1.14.0",
        "actionlib-tools==1.14.0",
        "aiohttp==3.8.4",
        "aiosignal==1.3.1",
        "ament-cmake-test==1.1.6",
        "ament-copyright==0.10.7",
        "ament-cppcheck==0.10.7",
        "ament-cpplint==0.10.7",
        "ament-flake8==0.10.7",
        "ament-index-python==1.2.0",
        "ament-lint==0.10.7",
        "ament-lint-cmake==0.10.7",
        "ament-package==0.13.1",
        "ament-pep257==0.10.7",
        "ament-uncrustify==0.10.7",
        "ament-xmllint==0.10.7",
        "angles==1.9.13",
        "aniso8601==7.0.0",
        "argcomplete==1.12.3",
        "async-timeout==4.0.2",
        "base-local-planner==1.17.3",
        "beeply==1.0.2",
        "beepy==1.0.7",
        "bondpy==1.8.6",
        "camera-calibration==1.17.0",
        "camera-calibration-parsers==1.12.0",
        "catkin==0.8.10",
        "cffi==1.15.1",
        "charset-normalizer==3.1.0",
        "click==8.1.3",
        "clickclick==20.10.2",
        "cmake==3.26.4",
        "connexion==2.6.0",
        "controller-manager==0.19.6",
        "controller-manager-msgs==0.19.6",
[... 249 lines cut ...]
        "toml==0.10.2",
        "tomli==2.0.1",
        "topic-monitor==0.14.4",
        "topic-tools==1.16.0",
        "torch==2.0.1",
        "tqdm==4.65.0",
        "triton==2.0.0",
        "turtlebot3-example==2.1.2",
        "turtlebot3-teleop==2.1.2",
        "typing-extensions==4.5.0",
        "urdfdom-py==1.1.0",
        "Werkzeug==2.2.3",
        "wifi==0.3.8",
        "xacro==1.14.16",
        "yarl==1.9.2",
        "zipp==3.15.0"
    ],
    "usb": [
        "NoLib"
    ],
    "uname": [
        "Linux makarand 5.15.0-78-generic #85~20.04.1-Ubuntu SMP Mon Jul 17 09:42:39 UTC 2023 x86_64 x86_64"
    ]
}
System info gathered successfully - saved as "log_system_information.json"

Serafadam, 2023-09-07:
Hi, thanks for the verbose information.
Could you check out if newest Noetic version still causes issues?

makarandmandolkar, 2023-09-07:
Thank you for your quick reply.
The launch file crashes, if I don't make the following change camera_i_pipeline_type: rgb.
Can you please guide how to set the require resolution as I see the camera.yaml file has been changed.

Serafadam, 2023-09-08:
Hi, would it be possible for you to test [this PR](https://github.com/luxonis/depthai-ros/pull/398) and see if things work better? Regarding parameter files, removing `rgb_i_width` and `rgb_i_height` parameters and setting following should be enough:
`rgb_i_resolution: 13MP`
`rgb_i_set_isp_scale: false`

makarandmandolkar, 2023-09-08:
Hi, thanks for the update.
I did try the PR, it looks great. There are a few observations though, like I was able to get the 13MP working with the said resolution. However, when I visualized the image topic in Rviz, it still appeared inverted.
Please let me know if you need more info.
Looking forward to hearing from your side.

[image]

Serafadam, 2023-09-11:
Hi, does the image also appear inverted when running the [demo script?](https://docs.luxonis.com/en/latest/#demo-script). In general, you can use `rgb_i_sensor_img_orientation: VERTICAL_FLIP`

rahulsharma11, 2023-10-04:
> Hi, would it be possible for you to test [this PR](https://github.com/luxonis/depthai-ros/pull/398) and see if things work better? Regarding parameter files, removing `rgb_i_width` and [...]

Hi.
As i can see your PR is merged with noetic branch. So i have freshly built the same and tried the same parameter but still not able to get the 13MP output. All time it gives 720p output.
Any suggestion ?
Thanks.

Serafadam, 2023-10-05:
Hi @rahulsharma11 , could you share your setup and resource usage? Are you subscribing to images over the network or on the device?

rahulsharma11, 2023-10-05:
Hi @Serafadam .

I am using Jetson Orin Nano platform. Over there i am able to get 
6fps -> 12MP
9fps -> 4k.

I also tried intel NUC  i7, over there i am getting
12fps -> 12MP
16fps -> 4k

I have directly connected the cameras over USB type C.
I am subscribing the topics directly on the board.

Serafadam, 2023-10-09:
Hi, I assume this is using ROS2 Humble or Noetic? There are several factors to consider here:
- In ROS2, `ros2 topic hz` won't provide accurate estimate as it also does deserialization of the message in Python (you can subscribe to camera info  or compressed topics to get more accurate estimates)
- In ROS2, [DDS tuning](https://docs.ros.org/en/foxy/How-To-Guides/DDS-tuning.html) might be required for larger messages depending on your setup
- You can test "real" camera FPS by running the ROS code with [depthai_pipeline_graph](https://github.com/luxonis/depthai_pipeline_graph)
- Setting FPS explicitly and/or enabling low bandwidth mode can also help with performance here

rahulsharma11, 2023-10-10:
Hi.
I am using ROS Noetic.

rahulsharma11, 2023-10-13:
> Hi, does the image also appear inverted when running the [demo script?](https://docs.luxonis.com/en/latest/#demo-script). In general, you can use `rgb_i_sensor_img_orientation: VERTICAL_FLIP`

I found setting the parameter as AUTO is making the feed correct. VERTICAL_FLIP is making the image correct but with horizontal mirror.

Serafadam, 2023-10-16:
Hi, newest version will have AUTO by default. Also, regarding Noetic frequency, I noticed high one core load increase when using `rostopic hz`. I suspect that the issue could be from the `rostopic` tool as it is not loaded as a nodelet so it cannot use shared memory transport (which could make a difference when subscribing to large messages).

rahulsharma11, 2023-10-16:
Thanks for rostopic node insight.
But i am dumping the data with rosbag record and i can find the topic count according to 6FPS(12MP) over there.
That means my USB only is limiting the data dump? i am using typec to typec in Thunderbolt port, so i am doubting if USB is limiting the data dump.

Serafadam, 2023-10-16:
According to [this issue](https://github.com/ros/ros_comm/issues/103), rosbag also doesn't support nodelet mechanism. USB speed should be printed out in logs after starting the camera. One additional tool to inspect the performance could be the [depthai pipeline graph](https://github.com/geaxgx/depthai_pipeline_graph). To use it,
``` bash
pip install git+https://github.com/luxonis/depthai_pipeline_graph.git
pip install depthai
pipeline_graph run "roslaunch depthai_ros_driver camera.launch"
```
This should output FPS of different nodes on a camera.

rahulsharma11, 2023-10-16:
ok.
Thanks for the suggestion. I will look into it.

Furthermore if dumping with rosbag record is also not good way to dump. If you can suggest few options to dump the data which is faster and good way, i can follow that too.

Thanks for your support.

Serafadam, 2023-10-16:
When it comes to other approaches, maybe using something like [Foxglove](https://foxglove.dev/) and foxglove_bridge could work better in this case

rahulsharma11, 2023-10-16:
I can see foxglove is very interactive tool for real time data visualisation, i think i can't use it to record the published topics.