luxonis/depthai-ros #396: OAK-FFC-4P:  Character [-] at element [4] is not valid in Graph Resource Name

martinlucan, 2023-09-07:
Hi @Serafadam ,
I try to lauch my OAK-FFC-4P, but following error pops out after commit `b8d5779c2642ac5066f03eb2651057850de99974`:

`Character [-] at element [4] is not valid in Graph Resource Name [CAMB-2L_i_max_q_size].  Valid characters are a-z, A-Z, 0-9, / and _.`

Could you please check it out? Thanks.

Serafadam, 2023-09-07:
Hi, could you run `export DEPTHAI_DEBUG=1` and post logs?

martinlucan, 2023-09-07:
Sure, here you go:
```

roslaunch depthai_ros_driver camera.launch 
... logging to /home/pr_admin/.ros/log/cae1d2ce-4d83-11ee-bc82-53e0d77ead1b/roslaunch-OC0001-27617.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://192.168.50.149:46847/

SUMMARY
========

PARAMETERS
 * /oak/camera_i_base_frame: oak-d_frame
 * /oak/camera_i_cam_pitch: 0.0
 * /oak/camera_i_cam_pos_x: 0.0
 * /oak/camera_i_cam_pos_y: 0.0
 * /oak/camera_i_cam_pos_z: 0.0
 * /oak/camera_i_cam_roll: 0.0
 * /oak/camera_i_cam_yaw: 0.0
 * /oak/camera_i_enable_imu: True
 * /oak/camera_i_enable_ir: True
 * /oak/camera_i_floodlight_brightness: 0.0
 * /oak/camera_i_imu_from_descr: False
 * /oak/camera_i_ip: 
 * /oak/camera_i_laser_dot_brightness: 800.0
 * /oak/camera_i_mx_id: 
 * /oak/camera_i_nn_type: spatial
 * /oak/camera_i_parent_frame: oak-d-base-frame
 * /oak/camera_i_pipeline_type: rgbd
 * /oak/camera_i_publish_tf_from_calibration: False
 * /oak/camera_i_tf_imu_from_descr: true
 * /oak/camera_i_usb_port_id: 
 * /oak/camera_i_usb_speed: SUPER_PLUS
 * /oak/left_i_board_socket_id: 1
 * /oak/nn_i_nn_config_path: depthai_ros_drive...
 * /oak/right_i_board_socket_id: 2
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    oak (nodelet/nodelet)
    oak_nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    set_log_level (rosservice/rosservice)

ROS_MASTER_URI=http://192.168.50.149:11311

process[set_log_level-1]: started with pid [27656]
process[oak_state_publisher-2]: started with pid [27657]
process[oak_nodelet_manager-3]: started with pid [27658]
process[oak-4]: started with pid [27659]
[ INFO] [1694096919.196779944]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1694096919.198573582]: waitForService: Service [/oak_nodelet_manager/load_nodelet] could not connect to host [192.168.50.149:45497], waiting...
[ INFO] [1694096919.209494276]: Initializing nodelet with 8 worker threads.
[ INFO] [1694096919.219261555]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[2023-09-07 16:28:39.330] [depthai] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-09-07 16:28:39.330] [depthai] [debug] Library information - version: 2.22.0, commit:  from , build: 2023-06-20 18:09:51 +0000
[2023-09-07 16:28:39.331] [depthai] [debug] Initialize - finished
[ INFO] [1694096919.332396276]: No ip/mxid/usb_id specified, connecting to the next available device.
[2023-09-07 16:28:39.394] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 2ms, archive read: 61ms
[2023-09-07 16:28:39.662] [depthai] [debug] Resources - Archive 'depthai-device-fwp-f033fd9c7eb0b3578d12f90302e87759c78cfb36.tar.xz' open: 1ms, archive read: 330ms
[2023-09-07 16:28:39.854] [depthai] [debug] Searching for booted device: DeviceInfo(name=3.1.4, mxid=14442C10013FE7D600, X_LINK_BOOTLOADER, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-09-07 16:28:39.859] [depthai] [debug] Connected bootloader version 0.0.22
[2023-09-07 16:28:39.859] [depthai] [info] New bootloader version available. Device has: 0.0.22, available: 0.0.26
[set_log_level-1] process has finished cleanly
[... 177 lines cut ...]
[DEBUG] [1694096923.143267020]: Param rgb_i_set_isp3a_fps with value 0
[DEBUG] [1694096923.143492001]: Param rgb_i_sensor_img_orientation with value NORMAL
[DEBUG] [1694096923.143617922]: Param rgb_i_publish_topic with value 1
[DEBUG] [1694096923.143798193]: Param rgb_i_low_bandwidth with value 0
[DEBUG] [1694096923.143914196]: Param rgb_i_output_isp with value 1
[DEBUG] [1694096923.144045892]: Param rgb_i_enable_preview with value 0
[DEBUG] [1694096923.144115477]: Node rgb created
[DEBUG] [1694096923.144272860]: Param rgb_i_enable_feature_tracker with value 0
[DEBUG] [1694096923.144280288]: Base node rgb created
[DEBUG] [1694096923.144756146]: Creating node stereo
[DEBUG] [1694096923.145377283]: Param stereo_i_left_socket_id with value 1
[DEBUG] [1694096923.145603447]: Param stereo_i_right_socket_id with value 2
[DEBUG] [1694096923.148057503]: Creating node CAMB-2L base
[DEBUG] [1694096923.148414194]: Destroying ClassLoader, base = depthai_ros_driver::pipeline_gen::BasePipeline, address = 0x7ffc31e7bf30
[ERROR] [1694096923.148509605]: Character [-] at element [4] is not valid in Graph Resource Name [CAMB-2L_i_max_q_size].  Valid characters are a-z, A-Z, 0-9, / and _.
[14442C10013FE7D600] [3.1.4] [1694096923.148] [host] [debug] Device about to be closed...
[2023-09-07 16:28:43.309] [depthai] [debug] XLinkResetRemote of linkId: (0)
[14442C10013FE7D600] [3.1.4] [1694096923.309] [host] [debug] Timesync thread exception caught: Couldn't read data from stream: '__timesync' (X_LINK_ERROR)
[14442C10013FE7D600] [3.1.4] [1694096923.309] [host] [debug] Log thread exception caught: Couldn't read data from stream: '__log' (X_LINK_ERROR)
[14442C10013FE7D600] [3.1.4] [1694096924.604] [host] [debug] Device closed, 1455
[FATAL] [1694096925.719805720]: Failed to load nodelet '/oak` of type `depthai_ros_driver/Camera` to manager `oak_nodelet_manager'
[oak_nodelet_manager-3] process has died [pid 27658, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager /nodelet_manager/load_nodelet:=oak/nodelet_manager/load_nodelet /nodelet_manager/unload_nodelet:=oak/nodelet_manager/unload_nodelet /nodelet_manager/list:=oak/nodelet_manager/list __name:=oak_nodelet_manager __log:=/home/pr_admin/.ros/log/cae1d2ce-4d83-11ee-bc82-53e0d77ead1b/oak_nodelet_manager-3.log].
log file: /home/pr_admin/.ros/log/cae1d2ce-4d83-11ee-bc82-53e0d77ead1b/oak_nodelet_manager-3*.log
================================================================================REQUIRED process [oak-4] has died!
```

Serafadam, 2023-09-08:
Thanks for the information, could you test the fix available [here](https://github.com/luxonis/depthai-ros/pull/398)?

martinlucan, 2023-09-08:
@Serafadam Unfortunately, the error still persists in `rgbd` mode.

Serafadam, 2023-09-08:
Ok, thanks for the info, could you test out this PR? https://github.com/luxonis/depthai-ros/pull/401

martinlucan, 2023-09-11:
@Serafadam unfortunately, I still get the same result.

Serafadam, 2023-09-11:
Hi @martinlucan, could you test that branch and output logs with the latest commits?

martinlucan, 2023-09-13:
@Serafadam the same behaviour after pulling the last commit. The error is present only while using OAK-FFC board, it works properly with OAK-Pro and OAK-Pro-W. Please see the log:

SUMMARY
========

PARAMETERS
 * /oak/camera_i_base_frame: oak-d_frame
 * /oak/camera_i_cam_pitch: 0.0
 * /oak/camera_i_cam_pos_x: 0.0
 * /oak/camera_i_cam_pos_y: 0.0
 * /oak/camera_i_cam_pos_z: 0.0
 * /oak/camera_i_cam_roll: 0.0
 * /oak/camera_i_cam_yaw: 0.0
 * /oak/camera_i_enable_imu: True
 * /oak/camera_i_enable_ir: True
 * /oak/camera_i_floodlight_brightness: 0.0
 * /oak/camera_i_imu_from_descr: False
 * /oak/camera_i_ip: 
 * /oak/camera_i_laser_dot_brightness: 800.0
 * /oak/camera_i_mx_id: 
 * /oak/camera_i_nn_type: spatial
 * /oak/camera_i_parent_frame: oak-d-base-frame
 * /oak/camera_i_pipeline_type: rgbd
 * /oak/camera_i_publish_tf_from_calibration: False
 * /oak/camera_i_usb_port_id: 
 * /oak/camera_i_usb_speed: SUPER_PLUS
 * /oak/nn_i_nn_config_path: depthai_ros_drive...
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    oak (nodelet/nodelet)
    oak_nodelet_manager (nodelet/nodelet)
    oak_state_publisher (robot_state_publisher/robot_state_publisher)
    set_log_level (rosservice/rosservice)

auto-starting new master
process[master]: started with pid [8473]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to e0850dcc-520a-11ee-8767-492a0077deed
process[rosout-1]: started with pid [8496]
started core service [/rosout]
process[set_log_level-2]: started with pid [8503]
process[oak_state_publisher-3]: started with pid [8505]
process[oak_nodelet_manager-4]: started with pid [8506]
process[oak-5]: started with pid [8507]
[ INFO] [1694591718.091179678]: Loading nodelet /oak of type depthai_ros_driver/Camera to manager oak_nodelet_manager with the following remappings:
[ INFO] [1694591718.094120674]: waitForService: Service [/oak_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1694591718.107210476]: Initializing nodelet with 12 worker threads.
[ INFO] [1694591718.115114128]: waitForService: Service [/oak_nodelet_manager/load_nodelet] is now available.
[2023-09-13 09:55:18.453] [depthai] [info] DEPTHAI_DEBUG enabled, lowered DEPTHAI_LEVEL to 'debug'
[2023-09-13 09:55:18.453] [depthai] [debug] Library information - version: 2.22.0, commit:  from , build: 2023-06-20 18:09:51 +0000
[2023-09-13 09:55:18.456] [depthai] [debug] Initialize - finished
[ INFO] [1694591718.459514362]: No ip/mxid/usb_id specified, connecting to the next available device.
[2023-09-13 09:55:18.612] [depthai] [debug] Resources - Archive 'depthai-bootloader-fwp-0.0.26.tar.xz' open: 5ms, archive read: 149ms
[2023-09-13 09:55:19.207] [depthai] [debug] Resources - Archive 'depthai-device-fwp-f033fd9c7eb0b3578d12f90302e87759c78cfb36.tar.xz' open: 7ms, archive read: 741ms
[2023-09-13 09:55:19.233] [depthai] [debug] Searching for booted device: DeviceInfo(name=1.3, mxid=14442C10013FE7D600, X_LINK_BOOTLOADER, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[2023-09-13 09:55:19.237] [depthai] [debug] Connected bootloader version 0.0.22
[2023-09-13 09:55:19.237] [depthai] [info] New bootloader version available. Device has: 0.0.22, available: 0.0.26
[set_log_level-2] process has finished cleanly
log file: /home/martinlucan/.ros/log/e0850dcc-520a-11ee-8767-492a0077deed/set_log_level-2*.log
[2023-09-13 09:55:19.989] [depthai] [debug] DeviceBootloader about to be closed...
[2023-09-13 09:55:19.990] [depthai] [debug] XLinkResetRemote of linkId: (0)
[2023-09-13 09:55:20.737] [depthai] [debug] DeviceBootloader closed, 748
[2023-09-13 09:55:20.748] [depthai] [debug] Searching for booted device: DeviceInfo(name=1.3, mxid=14442C10013FE7D600, X_LINK_BOOTED, X_LINK_USB_VSC, X_LINK_MYRIAD_X, X_LINK_SUCCESS), name used as hint only
[14442C10013FE7D600] [1.3] [2.020] [system] [warning] PRINT:LeonCss: GPIO boot mode 0x3, interface SPI_MASTER_EFF
Setting aons(0..4) back to boot from flash (offset = 0)
[... 165 lines cut ...]
[DEBUG] [1694591722.408977253]: Param rgb_i_fsync_continuous with value 0
[DEBUG] [1694591722.409833908]: Param rgb_i_fsync_trigger with value 0
[DEBUG] [1694591722.410686750]: Param rgb_i_set_isp3a_fps with value 0
[DEBUG] [1694591722.411534862]: Param rgb_i_sensor_img_orientation with value NORMAL
[DEBUG] [1694591722.412413358]: Param rgb_i_publish_topic with value 1
[DEBUG] [1694591722.412998356]: Param rgb_i_low_bandwidth with value 0
[DEBUG] [1694591722.413369275]: Param rgb_i_output_isp with value 1
[DEBUG] [1694591722.413878758]: Param rgb_i_enable_preview with value 0
[DEBUG] [1694591722.414133619]: Node rgb created
[DEBUG] [1694591722.414709363]: Param rgb_i_enable_feature_tracker with value 0
[DEBUG] [1694591722.414734396]: Base node rgb created
[DEBUG] [1694591722.416082612]: Creating node stereo
[DEBUG] [1694591722.417801064]: Param stereo_i_left_socket_id with value 1
[DEBUG] [1694591722.418686324]: Param stereo_i_right_socket_id with value 2
[DEBUG] [1694591722.422547804]: Creating node CAMB-2L base
[DEBUG] [1694591722.423293478]: Destroying ClassLoader, base = depthai_ros_driver::pipeline_gen::BasePipeline, address = 0x7ffeb7574790
[ERROR] [1694591722.423471654]: Character [-] at element [4] is not valid in Graph Resource Name [CAMB-2L_i_max_q_size].  Valid characters are a-z, A-Z, 0-9, / and _.
[14442C10013FE7D600] [1.3] [1694591722.423] [host] [debug] Device about to be closed...
[... cut ...]
[FATAL] [1694591725.320346554]: Failed to load nodelet '/oak` of type `depthai_ros_driver/Camera` to manager `oak_nodelet_manager'
[oak_nodelet_manager-4] process has died [pid 8506, exit code -11, cmd /opt/ros/noetic/lib/nodelet/nodelet manager /nodelet_manager/load_nodelet:=oak/nodelet_manager/load_nodelet /nodelet_manager/unload_nodelet:=oak/nodelet_manager/unload_nodelet /nodelet_manager/list:=oak/nodelet_manager/list __name:=oak_nodelet_manager __log:=/home/martinlucan/.ros/log/e0850dcc-520a-11ee-8767-492a0077deed/oak_nodelet_manager-4.log].

Serafadam, 2023-09-13:
Could you retry on the latest noetic branch? [This line](https://github.com/luxonis/depthai-ros/blob/1123a3bcb22ca51dc6c7823caf381d7f4e67011a/depthai_ros_driver/src/dai_nodes/stereo.cpp#L44) should also be printed before "Creating node base"

Serafadam, 2023-09-13:
@martinlucan, latest noetic version should also print out this line 
`"Creating stereo node with left sensor %s and right sensor %s",`
before 
`Creating node CAMB-2L base`, could you check if you correctly sourced the workspace after building?

martinlucan, 2023-09-14:
@Serafadam Thank you, it is working properly after the complete rebuild of the package

Serafadam, 2023-09-14:
Closing the issue since it's resolved.

closed 2023-09-14 by Serafadam