luxonis/depthai-ros #402: [BUG] [Humble] Missing dependency causes failure on the buildfarm

audrow, 2023-09-08:
The last release on Humble was missing a dependency for `tf2_geometry_msgs` in `geometry2`.
https://build.ros2.org/view/Hbin_uJ64/job/Hbin_uJ64__depthai_bridge__ubuntu_jammy_amd64__binary/62/

A colleague also pointed out it'd be good add a dependency on `tf2`, even though it will be implicitly included in `tf2_ros`.

Serafadam, 2023-09-11:
Hi, thanks for letting me know, it will be fixed soon, not sure why the[ CI showed green on this one ](https://build.ros2.org/job/Hpr__depthai-ros__ubuntu_jammy_amd64/202/)before merging.

sloretz, 2023-09-12:
Same issue on Noetic:

* https://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__depthai_bridge__ubuntu_focal_arm64__binary/51/console
* https://build.ros.org/job/Nbin_ufhf_uFhf__depthai_bridge__ubuntu_focal_armhf__binary/49/console
* https://build.ros.org/job/Nbin_uF64__depthai_bridge__ubuntu_focal_amd64__binary/

```
23:11:12 [ 75%] Building CXX object CMakeFiles/depthai_bridge.dir/src/TFPublisher.cpp.o
23:11:21 /usr/lib/ccache/c++  -DDEPTHAI_TARGET_CORE -DJSON_USE_IMPLICIT_CONVERSIONS=1 -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"depthai_bridge\" -DUSING_ROS -Ddepthai_bridge_EXPORTS -I/tmp/binarydeb/ros-noetic-depthai-bridge-2.8.0/include -I/opt/ros/noetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/usr/include/eigen3 -isystem /opt/ros/noetic/include -isystem /usr/include/opencv4 -isystem /opt/ros/noetic/include/depthai-shared/3rdparty -isystem /opt/ros/noetic/lib/arm-linux-gnueabihf/cmake/depthai/dependencies/include  -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-noetic-depthai-bridge-2.8.0=. -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC   -std=gnu++14 -o CMakeFiles/depthai_bridge.dir/src/TFPublisher.cpp.o -c /tmp/binarydeb/ros-noetic-depthai-bridge-2.8.0/src/TFPublisher.cpp
23:11:21 /tmp/binarydeb/ros-noetic-depthai-bridge-2.8.0/src/TFPublisher.cpp:21:10: fatal error: tf2_geometry_msgs/tf2_geometry_msgs.h: No such file or directory
23:11:24    21 | #include "tf2_geometry_msgs/tf2_geometry_msgs.h"
```

closed 2023-09-28 by Serafadam