luxonis/depthai-ros #408: [BUG] {Stereo baseline should be positive}

martinlucan, 2023-09-14:
Hi @Serafadam ,
I am using OAK-FFC-4 module for RTAB-Map Stereo odometry. I have been using this successfully until I built the last commit on `noetic` branch. currently, I see the following error. Has there been any changes in baseline interpretation? Do you advise to recalibrate the extrinstics?

`
[ERROR] [1694693705.371463281]: The stereo baseline (-0.041895) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.`

Serafadam, 2023-09-14:
Hi, could you share data that is being output on your camera_info topics?

martinlucan, 2023-09-18:
@Serafadam 

LEFT:
```
 header:
  seq: 172
  stamp:
    secs: 1694791756
    nsecs: 257870460
  frame_id: "oak_left_camera_optical_frame"
height: 800
width: 1280
distortion_model: "rational_polynomial"
D: [51.752418518066406, -370.51641845703125, -0.008457317017018795, 0.0016463500214740634, 788.4266967773438, 51.01250457763672, -366.0118713378906, 780.3660888671875]
K: [780.5100708007812, 0.0, 620.1886596679688, 0.0, 782.8507690429688, 367.64306640625, 0.0, 0.0, 1.0]
R: [0.9999654293060303, 0.0012884886236861348, 0.008211364038288593, -0.0012772120535373688, 0.9999982118606567, -0.0013783809263259172, -0.008213125169277191, 0.0013678455725312233, 0.9999653100967407]
P: [793.8073120117188, 0.0, 639.7841186523438, -33.25672651861009, 0.0, 795.9546508789062, 378.6341857910156, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

RIGHT:

```
header:
  seq: 8
  stamp:
    secs: 1694791693
    nsecs: 796834480
  frame_id: "oak_right_camera_optical_frame"
height: 800
width: 1280
distortion_model: "rational_polynomial"
D: [-18.02766227722168, 133.4366912841797, -0.008560389280319214, 0.0034420546144247055, -313.8127746582031, -18.0432071685791, 133.4146728515625, -313.45745849609375]
K: [793.8073120117188, 0.0, 639.7841186523438, 0.0, 795.9546508789062, 378.6341857910156, 0.0, 0.0, 1.0]
R: [0.9996832609176636, 0.0075179655104875565, 0.0240179393440485, -0.007550940383225679, 0.9999706745147705, 0.0012825404992327094, -0.024007592350244522, -0.0014634922845289111, 0.9997106790542603]
P: [793.8073120117188, 0.0, 639.7841186523438, 0.0, 0.0, 795.9546508789062, 378.6341857910156, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
---
```

Serafadam, 2023-09-28:
Hi @martinlucan, stereo socket logic has been updated a little bit to additionally use OpenCV format (changes [here](https://github.com/luxonis/depthai-ros/blob/6e88d5c84d470355bc37c9f08d17434e851c3c09/depthai_bridge/src/ImageConverter.cpp#L370)), by default the mechanism should be the same. Could you:
- Try changing `i_reverse_stereo_socket_order` for the sensors? (You can do that for individual sensors, and for rectified streams published by stereo node)
- Share camera_info topics from the older depthai-ros version for comparison?

mkarklins, 2023-10-05:
Hello!
Just had the same issue and reversing the camera order helped. 

I'm quite new to ROS2 in general, so I didn't yet understand how to change the socket order for the example node `stereo_inertial_node`, so I updated it in source code and recompiled, and it fixed the issue.

ciniminis1, 2024-08-23:
@Serafadam i temporarily fixed it in source code, although setting up the parameter does not work. `stereo_i_reverse_stereo_socket_order` seem to have no influence on the output, and `left_i_reverse_stereo_socket_order` or `right_i_reverse_stereo_socket_order` throw the error:

`/opt/ros/noetic/lib/nodelet/nodelet: symbol lookup error: /opt/ros/noetic/lib/libdepthai_ros_driver_sensor_nodes.so: undefined symbol: _ZN3dai3ros14ImageConverter24reverseStereoSocketOrderEv`

imranhussain14, 2025-05-19:
@ciniminis1 @mkarklins ,
i am also facing the same issue. can you please let me know how you update the source code. 
Last time i just swapped the camera_info topics in rtabmap launch file but i think its not a good solution because i see that some of my images are synchronized with wrong left/right images in slam.