luxonis/depthai-ros #409: [Help] rgbd_pcl low FPS

SpacePanda-42, 2023-09-14:
Hi, I have been trying to get depth and rgb images published simultaneously from a Turtlebot4 with an Oak-D-Lite. Currently I'm using rgbd_pcl.launch.py, but am having frame rate issues (~4-5 FPS). I have tried setting i_low_bandwidth to true for everything from camera.yaml in depthai_ros_driver/config/ as well as lowering the i_low_bandwidth_quality to 20. However, nothing seems to improve the performance. It looked like I should be editing camera.yaml, but it was mentioned in another issue that rgbd_pcl has its own .yaml file (which appears to now be gone) https://github.com/luxonis/depthai-ros/issues/329. 

Any insight would be appreciated. Not sure if I've modified the low_bandwidth setting incorrectly or if there's a better way to output lower resolution images.

Serafadam, 2023-09-15:
Hi, are you subscribing to the images from other PC? If yes, framerate can be influenced by network/DDS settings, [here you can find a guide on tuning DDS](https://docs.ros.org/en/rolling/How-To-Guides/DDS-tuning.html). If not, please share your config file and exact setup according to the [bug template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md).

alduxvm, 2023-09-15:
I had the same issue, until I realize it was the cable... I tested 3-4 cables until one gave me the proper FPS using the `rgb_stereo_node.launch` , I used to have 7-8 fps and now with a proper cable I have 30 on the color image