luxonis/depthai-ros #410: [BUG] OAK-D PRO W: No image and no camera_info received

YamadaNaoto712, 2023-09-17:
I use depthai-noetic with oak-d pro w on ubuntu20.04 of wsl2(windows11).
I run two commands below;
on 1st terminal: **roslaunch depthai_examples stereo_inertial_node.launch**
on 2nd terminal: **ROS_NAMESPACE=/stereo_inertial_publisher/ rosrun stereo_image_proc stereo_image_proc**

When Image Topic is /stereo_inertial_publisher/left/* or /stereo_inertial_publisher/right/*
 (example:/stereo_inertial_publisher/left/image_rect), no image and no camerainfo received.
[image]

When Image Topic is /stereo_inertial_publisher/stereo/* or /stereo_inertial_publisher/color/*, it works well.
[image]

I want data from /stereo_inertial_publisher/left/image_rect and /stereo_inertial_publisher/right/image_rect for SLAM.

Is there any solution to this?
Any advice would be very much appreciated!

Serafadam, 2023-09-28:
Hi, for publishing all the data I would recommend using `depthai_ros_driver` as you can enable rectified streams using parameters, as described in the [Readme](https://github.com/luxonis/depthai-ros#depth-alignment)

YamadaNaoto712, 2023-10-13:
Thank you.
 I modified **depthai_ros_driver/config/camera.yaml** and published left and right/image_raw, but I don’t know how to watch the streams from left and right/image_raw.
Could you tell me the way?
I suspect  **stereo_inertial_node.launch** need to be modified.

Serafadam, 2023-10-13:
Hi, if you enable the streams in the config file, you should be able to automatically view them in rviz after you launch the `camera` nodelet from `depthai_ros_driver`. `stereo_inertial_node` is a separate code example, so it doesn't need to be modified

YamadaNaoto712, 2023-10-15:
Thank you for your reply.

Although I launch   `camera.launch`,  rviz does not appear. 
After `rosrun rviz rviz`  rviz shows up but I don't catch any streams.

I will put my `camera.yaml`

/stereo_inertial_publisher:
  camera_i_enable_imu: true
  camera_i_enable_ir: true
  camera_i_floodlight_brightness: 0.0
  camera_i_ip: ''
  camera_i_laser_dot_brightness: 800.0
  camera_i_mx_id: ''
  camera_i_nn_type: spatial
  camera_i_pipeline_type: Stereo
  camera_i_usb_port_id: ''
  camera_i_usb_speed: SUPER_PLUS
  nn_i_nn_config_path: depthai_ros_driver/mobilenet
  right_i_publish_topic: true
  left_i_publish_topic: true

And after `roslaunch depthai_ros_driver camera.launch` below sentences appear

started roslaunch server http://Lenovo:43409/

SUMMARY
========

PARAMETERS
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.16.0
 * /stereo_inertial_publisher/camera_i_base_frame: oak-d_frame
 * /stereo_inertial_publisher/camera_i_cam_pitch: 0.0
 * /stereo_inertial_publisher/camera_i_cam_pos_x: 0.0
 * /stereo_inertial_publisher/camera_i_cam_pos_y: 0.0
 * /stereo_inertial_publisher/camera_i_cam_pos_z: 0.0
 * /stereo_inertial_publisher/camera_i_cam_roll: 0.0
 * /stereo_inertial_publisher/camera_i_cam_yaw: 0.0
 * /stereo_inertial_publisher/camera_i_enable_imu: True
 * /stereo_inertial_publisher/camera_i_enable_ir: True
 * /stereo_inertial_publisher/camera_i_floodlight_brightness: 0.0
 * /stereo_inertial_publisher/camera_i_imu_from_descr: False
 * /stereo_inertial_publisher/camera_i_ip:
 * /stereo_inertial_publisher/camera_i_laser_dot_brightness: 800.0
 * /stereo_inertial_publisher/camera_i_mx_id:
 * /stereo_inertial_publisher/camera_i_nn_type: spatial
 * /stereo_inertial_publisher/camera_i_parent_frame: oak-d-base-frame
 * /stereo_inertial_publisher/camera_i_pipeline_type: Stereo
 * /stereo_inertial_publisher/camera_i_publish_tf_from_calibration: False
 * /stereo_inertial_publisher/camera_i_usb_port_id:
 * /stereo_inertial_publisher/camera_i_usb_speed: SUPER_PLUS
 * /stereo_inertial_publisher/left_i_publish_topic: True
 * /stereo_inertial_publisher/nn_i_nn_config_path: depthai_ros_drive...
 * /stereo_inertial_publisher/right_i_publish_topic: True

NODES
  /
    stereo_inertial_publisher (nodelet/nodelet)
    stereo_inertial_publisher_nodelet_manager (nodelet/nodelet)
    stereo_inertial_publisher_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [7775]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to df06c76e-6b29-11ee-b16a-77bcc5a76c99
process[rosout-1]: started with pid [7785]
started core service [/rosout]
process[stereo_inertial_publisher_state_publisher-2]: started with pid [7792]
process[stereo_inertial_publisher_nodelet_manager-3]: started with pid [7793]
process[stereo_inertial_publisher-4]: started with pid [7794]
[ INFO] [1697353808.278760812]: Loading nodelet /stereo_inertial_publisher of type depthai_ros_driver/Camera to manager stereo_inertial_publisher_nodelet_manager with the following remappings:
[ INFO] [1697353808.281819392]: Initializing nodelet with 8 worker threads.
[ INFO] [1697353808.282346442]: waitForService: Service [/stereo_inertial_publisher_nodelet_manager/load_nodelet] has not been advertised, waiting...
[ INFO] [1697353808.304122134]: waitForService: Service [/stereo_inertial_publisher_nodelet_manager/load_nodelet] is now available.
[ INFO] [1697353808.406512029]: No ip/mxid/usb_id specified, connecting to the next available device.
[ INFO] [1697353814.615828274]: Camera with MXID: 1844301021BA341300 and Name: 1.1 connected!
[ INFO] [1697353814.621646709]: USB SPEED: HIGH
[ INFO] [1697353814.650150922]: Device type: OAK-D-PRO-W
[ INFO] [1697353814.664418886]: Pipeline type: Stereo
[ INFO] [1697353815.331447735]: Finished setting up pipeline.
[ INFO] [1697353817.292249160]: Camera ready!

Is there anything strange?

Serafadam, 2023-10-16:
Hi, Rviz is not enabled in camera.launch and needs to be run separately. Could you share the results of running `rostopic list` after running the camera node?

YamadaNaoto712, 2023-10-17:
ok this is the rostopic

 rostopic list
/diagnostics
/joint_states
/oak/imu/data
/oak/left/camera_info
/oak/left/image_raw
/oak/left/image_raw/compressed
/oak/left/image_raw/compressed/parameter_descriptions
/oak/left/image_raw/compressed/parameter_updates
/oak/left/image_raw/compressedDepth
/oak/left/image_raw/compressedDepth/parameter_descriptions
/oak/left/image_raw/compressedDepth/parameter_updates
/oak/left/image_raw/theora
/oak/left/image_raw/theora/parameter_descriptions
/oak/left/image_raw/theora/parameter_updates
/oak/nn/spatial_detections
/oak/parameter_descriptions
/oak/parameter_updates
/oak/rgb/camera_info
/oak/rgb/image_raw
/oak/rgb/image_raw/compressed
/oak/rgb/image_raw/compressed/parameter_descriptions
/oak/rgb/image_raw/compressed/parameter_updates
/oak/rgb/image_raw/compressedDepth
/oak/rgb/image_raw/compressedDepth/parameter_descriptions
/oak/rgb/image_raw/compressedDepth/parameter_updates
/oak/rgb/image_raw/theora
/oak/rgb/image_raw/theora/parameter_descriptions
/oak/rgb/image_raw/theora/parameter_updates
/oak/right/camera_info
/oak/right/image_raw
/oak/right/image_raw/compressed
/oak/right/image_raw/compressed/parameter_descriptions
/oak/right/image_raw/compressed/parameter_updates
/oak/right/image_raw/compressedDepth
/oak/right/image_raw/compressedDepth/parameter_descriptions
/oak/right/image_raw/compressedDepth/parameter_updates
/oak/right/image_raw/theora
/oak/right/image_raw/theora/parameter_descriptions
/oak/right/image_raw/theora/parameter_updates
/oak/stereo/camera_info
/oak/stereo/image_raw
/oak/stereo/image_raw/compressed
/oak/stereo/image_raw/compressed/parameter_descriptions
/oak/stereo/image_raw/compressed/parameter_updates
/oak/stereo/image_raw/compressedDepth
/oak/stereo/image_raw/compressedDepth/parameter_descriptions
/oak/stereo/image_raw/compressedDepth/parameter_updates
/oak/stereo/image_raw/theora
/oak/stereo/image_raw/theora/parameter_descriptions
/oak/stereo/image_raw/theora/parameter_updates
/oak_nodelet_manager/bond
/rosout
/rosout_agg
/tf
/tf_static

Serafadam, 2023-10-18:
Hi, according to the topci list, the left/right topics are published:
```
/oak/left/camera_info
/oak/left/image_raw
/oak/right/camera_info
/oak/right/image_raw
```

YamadaNaoto712, 2023-10-18:
Hi! my problem is solved just now !
This command made it possible!
`roslaunch depthai_examples stereo_inertial_node.launch enableRviz:=false depth_aligned:=false　stereo_fps:=4`

Thank you so much for all the help you've given me!

Serafadam, 2023-10-18:
Happy to help! Closing the issue, please reopen if you have further questions.