luxonis/depthai-ros #416: [Feature-Request] Get full stereo FOV by using alpha_scaling

BriceRenaudeau, 2023-09-25:
### Why:
We are using the OAK-D-PRO W. 

Using ROS driver, the field of view of the depth image is limited to the rectified image.

### What:
We want to get the full field of view for the depth image.

### How:
This limitation is due to 2 points:

1. The `alpha_scaling` parameter isn't exposed (I was ready to open a PR to expose it)
2. The `camera_info` topic is not modified to fit the field of view

In the following screenshot, you can see the stereo-depth image applying a 0.5 alpha_scaling, but the 3D points are not are the right place in the world (3D points must fit the laser scan). 
[image]

doisyg, 2023-09-25:
That would be incredibly useful to finally be able to use the full FoV in ROS ? @CenekAlbl

Serafadam, 2023-10-03:
Hi @BriceRenaudeau @doisyg , I've updated the codebase to modify the intrinsics based on alpha factor, you can test it out [on this branch](https://github.com/luxonis/depthai-ros/tree/alpha_scaling_info_update_humble). If the feature works as expected, then I will add it to the next release.

BriceRenaudeau, 2023-10-04:
Hi, bad news it seems it doesn't work.
Strangely the frequency has decreased, it cannot keep 15Hz and is down to ~13Hz.
The depth image applies the alpha_scaling but the camera info seems wrong.
Here is the camera_info I got (r is identity and d is zero):
```
---
header:
  stamp:
    sec: 1696413708
    nanosec: 857345536
  frame_id: oak_front_right_camera_optical_frame
height: 720
width: 1280
distortion_model: rational_polynomial
d:
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
- 0.0
k:
- 566.3484497070312
- 0.0
- 656.9843139648438
- 0.0
- 566.8724365234375
- 368.1535949707031
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 566.3484497070312
- 0.0
- 656.9843139648438
- 0.0
- 0.0
- 566.8724365234375
- 368.1535949707031
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
---
```

BriceRenaudeau, 2023-10-04:
The frequency issue doesn't seem to be related to this commit.

Serafadam, 2023-10-05:
Pushed out an update, `d` and `r` parameters should be updated

BriceRenaudeau, 2023-10-09:
Thanks, I tested it and it looks good to me. :+1: 

I still have frequency issues, I will investigate and create an issue about it.

I will be waiting for an Iron version.