luxonis/depthai-ros #420: [BUG] {How to publish rectified streams}

y2d2, 2023-09-29:
**Check if issue already exists**

- Per request made in https://github.com/luxonis/depthai-ros/issues/54 
- I also started a discussion on https://discuss.luxonis.com/d/2246-how-to-publish-the-rectified-streams-of-the-cameras-in-ros

**Describe the bug**
Not sure how the rectified streams are published in ROS2. But it seems not to publish
 
**Minimal Reproducible Example**
changed `/opt/ros/humble/share/depthai_ros_driver/config/camera.yaml` ( I had trouble making a separate yaml file to run so I adapted the base file directly.):
```
/oak:
  ros__parameters:
    camera:
      i_enable_imu: false
      i_enable_ir: true
      i_nn_type: none
      i_pipeline_type: Stereo
    stereo:
      i_publish_left_rect: false
      i_publish_right_rect: false
      i_publish_synced_rect_pair: false
```
run ` `
run `ros2 topic list` : 
```
ros2 topic list
/diagnostics
/joint_states
/oak/left/camera_info
/oak/left/image_raw
/oak/left/image_raw/compressed
/oak/left/image_raw/compressedDepth
/oak/left/image_raw/theora
/oak/right/camera_info
/oak/right/image_raw
/oak/right/image_raw/compressed
/oak/right/image_raw/compressedDepth
/oak/right/image_raw/theora
/parameter_events
/robot_description
/rosout
/tf
/tf_static
```
Changed  `/opt/ros/humble/share/depthai_ros_driver/config/camera.yaml`:
```
/oak:
  ros__parameters:
    camera:
      i_enable_imu: false
      i_enable_ir: true
      i_nn_type: none
      i_pipeline_type: Stereo
    stereo:
      i_publish_left_rect: true
      i_publish_right_rect: true
      i_publish_synced_rect_pair: true
```
run ` `
run `ros2 topic list` : 
```
/diagnostics
/joint_states
/oak/left/camera_info
/oak/left/image_raw
/oak/left/image_raw/compressed
/oak/left/image_raw/compressedDepth
/oak/left/image_raw/theora
/oak/right/camera_info
/oak/right/image_raw
/oak/right/image_raw/compressed
/oak/right/image_raw/compressedDepth
/oak/right/image_raw/theora
/parameter_events
/robot_description
/rosout
/tf
/tf_static
```
It seems nothing has changed in the published topics. 

**Expected behavior**
I was expecting to see topics that show the rectified stream. Unless the image_raw now contains the rectified stream

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Pipeline Graph**
I renamed the 12761857/1844301031DB391300_pipeline.json to .txt to be able to paste it here.
the pipeline has been taken with the i_publish_xx_rect set to true.
[1844301031DB391300_pipeline.txt](https://github.com/luxonis/depthai-ros/files/12761857/1844301031DB391300_pipeline.txt)

**Attach system log**
I renamed the log_system_information.json to .txt to be able to paste it here.
[log_system_information.txt](https://github.com/luxonis/depthai-ros/files/12761917/log_system_information.txt)

**Additional context**
I am not sure what other parameters have to be set to make the rectified streams work. 
I there documentation for all the parameters in this file? What their values can be and what their impact is?

Serafadam, 2023-09-29:
Hi, could you try setting pipeline type to depth and see if that works?

y2d2, 2023-09-29:
Hi @Serafadam, 
This worked, thanks! I appologise for using the wrong settings. Is there documentation on how the different parameters can be set and what the effects are
?

Serafadam, 2023-09-29:
Glad that it worked, it seems that the pipeline type section in the documentation needs some additional information. In short - `Stereo` is creating a pipeline for two separate sensors and `Depth` additionally creates Depth publishing node which can output rectified stereo streams.