luxonis/depthai-ros #422: OAK D PRO - MAXIMUN RESOLUTION IMAGES ARE NOT SENT

acordobes-airvant, 2023-10-03:
I have the OAK-D PRO camera model. In theory it is a 12MP (4056x3040) camera. On the one hand, I'm not able to set the camera to 4056x3040px. When I indicate the resolution of 4056x3040 I get the following error: 

`ColorCamera(0) - 'video' width or height (4056, 3040) bigger than maximum at current sensor resolution (3840, 2160)`

My configuration file is the following:

```
/oak:
  camera_i_enable_imu: true
  camera_i_enable_ir: true
  camera_i_floodlight_brightness: 0.0
  camera_i_ip: ''
  camera_i_laser_dot_brightness: 1000.0
  camera_i_mx_id: ''
  camera_i_nn_type: none
  camera_i_pipeline_type: RGBD
  camera_i_usb_port_id: ''
  camera_i_usb_speed: SUPER_PLUS
  imu_i_max_q_size: 30
  nn_i_nn_config_path: depthai_ros_driver/mobilenet
  rgb_i_board_socket_id: 0
  rgb_i_calibration_file: ''
  rgb_i_enable_preview: false
  rgb_i_fps: 1
  rgb_i_height: 3040
  rgb_i_interleaved: false
  rgb_i_keep_preview_aspect_ratio: true
  rgb_i_low_bandwidth: false
  rgb_i_low_bandwidth_quality: 50
  rgb_i_max_q_size: 30
  rgb_i_preview_size: 300
  rgb_i_publish_topic: true
  rgb_i_resolution: '12MP'
  rgb_i_set_isp_scale: false
  rgb_i_output_isp: false
  rgb_i_width: 4056
  rgb_r_exposure: 1000
  rgb_r_focus: 1
  rgb_r_iso: 800
  rgb_r_keep_preview_aspect_ratio: true
  rgb_r_set_man_exposure: false
  rgb_r_set_man_focus: false
  rgb_r_set_man_whitebalance: false
  rgb_r_whitebalance: 3300
  rgb_i_isp_num: 2
  rgb_i_isp_den: 3
```
For this reason I have to change the resolution to 3840x2160 px.

 On the other hand, at the resolution 3840x2160px, the maximum that tells me it is possible, it gives me the following error and only provides me with one image:
 
`[StereoDepth(3)] [error] Out of memory while creating pool for 'rectifiedLeft' frames. Number of frames: 3 each with size: 921600B  `

How can I receive images at the maximum resolution with a certain frequency?

Serafadam, 2023-10-03:
Hi, few comments:
1. Thanks for notifying on the video size warning - there is a minor bug here, and although images are publishing at 12MP this will cause publishing of invalid CameraInfo data, this will be updated in the next release.
2. While you can set the large resolutions individually, there are limitations when it comes to the device resources - memory pool is one limitation, the other one would be the input size for Spatial Location calculator - currently the limit input size for Spatial Location Calculator is FHD resolution. With NN disabled you can run both depth and RGB at 4K resolution.

Serafadam, 2023-10-03:
Yes, you can output 12MP images, stereo output gets automatically scaled up to match RGB resolution, but you can override that by setting for example:
```
stereo:
      i_width: 1280
      i_height: 720
```