luxonis/depthai-ros #423: [BUG] {Incorrect timestamps on OAK D Pro POE & OAK D Pro POE W}

Victorsoeby, 2023-10-04:
I am using the Dozkerized version of the Humble branch, where i forked the repo from Humble v. 2.7.0.
This far the driver has been working alright, with minor issues related to me using a POE camera, which i suspect this could also be about.

When i run the Ros driver on my robot i have found that every messages related to the camera has a timestamp that is about 3 hours late, making it imposibble for me to synchronize the camera with other cameras. Here you see me echo the timestamps of two topics: 

```
root@7c6b04e857cc:/workspaces/ros_ctis_camera# ros2 topic echo /oak/stereo/image_raw --field header --once
stamp:
  sec: 1696401849
  nanosec: 905180924
frame_id: oak_rgb_camera_optical_frame
---
root@7c6b04e857cc:/workspaces/ros_ctis_camera# ros2 topic echo /ctis/camera/zero_order --field header --once
stamp:
  sec: 1696413626
  nanosec: 373695670
frame_id: ''
---
```
The rest of my robot also has the ~1696413626 time, but everything related to the OAK camera has the ~1696401849 timestamp (Note that it is the sixth digit, indicating about 12000 seconds delay (3 hours). 

I have tested it both on my OAK D Pro POE camera and OAK D Pro POE W camera. 
I saw another related issue, but it was closed due to inactivity: https://github.com/luxonis/depthai-ros/issues/314

While i wait i will try to use a OAK USB camera i have laying around, to see if the issue persists. 

How can i fix this?

Victorsoeby, 2023-10-04:
Just did the same test with my OAK-d USB camera, which was 16282 seconds behind, which is ~4.5 hours.

Serafadam, 2023-10-05:
Hi, would it be possible for you to update to the latest driver version? Additionally, it would be useful if you could post more information as described in [bug report template](https://github.com/luxonis/depthai-ros/blob/humble/.github/ISSUE_TEMPLATE/bug_report.md).